thelsing/knx #2: TPUART2 compatible with NCN5120

nanosonde, 2018-08-10:
Hi!
I have just the the new tpuart data link layer in the samd branch.
Is the TPUART2 that is used in some Siemens BCUs compatible?

From here I guess that they seem to be very similar:
https://github.com/knxd/knxd/blob/master/src/backend/ncn5120.cpp
https://github.com/knxd/knxd/blob/master/src/backend/ncn5120.h

thelsing, 2018-08-11:
You are right. They are mostly compatible. Setting of PA is different and TPUART does't have a stop command, so the chip will not be disabled on knx.enabled(false). I plan on adding some ifdefs or something later. 
At the moment I'm fighting with the NCN because some bytes from the bus are lost. Maybe my hardware is broken. Maybe I need to get a Siemens BCU myself.

thelsing, 2018-08-15:
I removed the ncn specific parts. So TPUART can be tested.

closed 2018-08-15 by thelsing