thelsing/knx #5: knx.configured() returns true too early

Bernator, 2018-11-04:
During ETS configuration knx.configured() returns true even before restart of mc finished configuration procedure.

thelsing, 2018-11-05:
I just pushed some changes that should resolve this. But now I'm not sure if this is necessary. What is the harm in returning true before restart? After Loading group address table, association table, group object table and application program the device is configured. Sure ETS sends an restart command next but that should not be necessary.

Bernator, 2018-11-06:
In my case the application program gets initialized during setup() with some ETS Parameters, this is only possible if the device is configured. So if there is no configuration at startup the app is not getting initialized. 
Further if knx.configured() returns true later in loop() without restart, the uninitilaized app is getting started for a short time until the restart comes. In my case the app hang and the restart was never processed.
Sure it would be possible to fix the app and do nothing in this case and wait for the restart but i think it is much more intuitive when the stack returns true only after the restart.
Back answer, what is the benefit of returning true before restart? :)

thelsing, 2018-11-06:
Ok, seems like a valid reason for me. Can you test if it works correctly now?

Bernator, 2018-11-06:
yes it works as expected, thx

closed 2018-11-06 by thelsing