thelsing/knx #6: Separate directories for different platforms

nanosonde, 2018-11-08:
Hi!
I think it would be better to move the platform specific stuff to separate folders.

What do you think?

However, I am happy that you have removed the submodule stuff. :+1:

thelsing, 2018-11-08:
There are only 2 platform specific files per platform. (Well 3 for linux). So I don't see the need to split them up. But if you submit a pull request I will merge it ;)

nanosonde, 2018-11-08:
Ok, currently there are only those 3 platforms, but who knows. ;)

What I have in mind is something like this:
https://github.com/bluekitchen/btstack

thelsing, 2018-11-09:
Looks fine.

nanosonde, 2019-01-07:
I have closed the PR as I do not want to put any more effort into this for now as I am not sure if you really like the idea.

Just take it as an example of how I wanted to do it. However, I am not sure if it fits your VisualGDB development approach.

closed 2019-01-07 by nanosonde

thelsing, 2019-01-07:
You are a bit too impatient. You created the pull request 4 days ago and closed it 2 days ago. I do this for fun and have a life too. So it can take some time till I check a pull request in detail.

nanosonde, 2019-01-07:
Sorry for the hassle. It is not related to the time frame in anyway, but more related to my activities and priorities of those. 
I have changed them spontaneously.

So again: no offense!