thelsing/knx #10: Create support for knx-ip secure

thelsing, 2018-12-25:


nanosonde, 2020-07-05:
Crypto stuff required:

- ~~AES CBC MAC (already there due to data secure)~~
- https://github.com/monolifed/pbkdf2-hmac-sha256/
- Elliptic curve 25519

nanosonde, 2022-02-02:
Two existing implementations:

JS: https://github.com/Supergiovane/node-red-contrib-knx-ultimate/blob/master/KNXEngine/protocol/KNXSecureSessionRequest.js

JAVA: https://github.com/calimero-project/calimero-core/blob/master/src/tuwien/auto/calimero/knxnetip/SecureSessionUdp.java

nanosonde, 2022-02-25:
Another implementation in Python: https://github.com/XKNX/xknx/pull/875/files

farmio, 2022-12-20:
Hi 👋! 
Since there is no support for Tunnelling at all currently, you may be more interested in Secure Routing. You can find a Python implementation here: https://github.com/XKNX/xknx/blob/5fa7f8b8712ef3c9631fc2617fa0a0996bf9656c/xknx/io/ip_secure.py#L388
This doesn't need Elliptic curve or PBKDF2-HMAC-SHA256 iirc 😃

nanosonde, 2023-03-03:
xknx now has data secure support: https://github.com/XKNX/xknx/blob/main/xknx/secure/data_secure.py