thelsing/knx #40: Correct distinction between APCI sizes (4 bit vs. 10bit)

nanosonde, 2019-10-21:
Hi!
I have found this in the code while implementing the addtional application layer services for KNX RF (S-Mode): https://github.com/thelsing/knx/blob/3063bf8195e8491f4955e4a6e89c1533074084c1/src/knx/apdu.cpp#L15

Unfortunately, this does not work for those additional services,
Please have a look at KNX spec. 3.3.7 (Application Layer) p.9-10.

IMHO, it is not possible to decide from the APCI bits alone, if it is of 4 bit size or 10 bit size.
Instead the length field has to be taken into account for this distinction.
If it is equal to 1, APCI size is 4 bits.
If it is greater than or equal to 2, APCI is 10 bits.

See also:
https://support.knx.org/hc/en-us/articles/115003188529-Payload

thelsing, 2019-10-21:
Your algorithm doesn't work with GroupValueResponse and GroupValueWrite, because the size of the data is also included in the length field.

The line should probably be changed to:
if ((apci >> 6) < 11 && (apci >> 6) != 7)
 
Right?

nanosonde, 2019-10-23:
Yes, I have looked at the table again. 

For those fields (with 10 bit size) where (apci >> 6) < 11 && (apci >> 6) != 7 is true (e.g.  A_IndividualAddress_Write,  A_GroupValue_Read-PDU ), the lower bits are always 0. So masking them away would be ok.

thelsing, 2019-10-23:
Fixed.

closed 2019-10-23 by thelsing