thelsing/knx #44: Possible issues with interface object server

nanosonde, 2019-11-10:
Hi!

While implementing the new cEMI server feature I have analyzed the read/write access of properties.

When using Net'n'node I found out that it does the following:
1) M_PropWrite.req: cEMI Server Object, PID_COMM_MODE set 0x00 (data link layer)
2) M_PropRead.req:  cEMI Server Object, PID_COMM_MODE to verify if it is really set to 0x00.

```
cEMI USB RX len: 7 data:  data: FC 00 08 01 34 10 01
M_PropRead_req
ObjectType: 0x0008
ObjectInstance: 0x01
PropertyId: 0x34 (52) -> PID_COMM_MODE (PDT_ENUM8)
NumberOfElements: 16
StartIndex: 1
```
However, the property is only PDT_ENUM8, so a single byte.
The client (Net'n'Node) requested 16 elements (is this a bug?).
So a valid response would now contain NumberOfElements=1, StartIndex=1 and a single data byte with the comm mode. 
(See KNX spec. 3/6/3 p.100: M_PropRead.con message, positive response)

I do not see this kind of implementation anywhere in an interface object implementation.
Am I overlooking something or did I find an issue here?
Current implementation always sends back the NumberOfElements received in the request, but not the actual available MaxNoOfElements. 
Also whenever an external client (A_PropRead) wants to read the property value length, the code won't work as the current code would handle a startIndex=0 to read the length entry (first array entry) of the property value.

See KNX spec. 3/4/1 p.15 that a property value of a full interface object (what you have implemented) shall be always(!) an array. First entry is the length, followed by the data.

nanosonde, 2019-11-10:
Argh. I have read the docs again.

NumberOfElements: 4 bits (and not 8 bits!)
StartIndex: 12 bits (and not 8 bits!)

Very badly illustrated in the docs. ;)

So it does not request 16 elements, but just 1 element which is correct for PDT_ENUM8 type.

However, the thing "property value is always an array with first entry being the length" remains an issue IMHO. This won't work with the current implementation.

thelsing, 2019-11-11:
You are right. PropertyValueRead already seems to do the right thing. 
Returning the right number of elements for PropertyWrite should be easy. 
Resetting the number of elements to no elements as written in 
4.3.2.2 conflicts a bit with 4.3.1.3  (both in 3/4/1) 
If 4.3.1.3 it says that you can clear a property by writing 0 to index 0. 4.3.2.2 says that "Therefore an A_PropertyValue_Read.req with nr_of_elem = 1, start_index = 0 shall result in an A_PropertyValue_Read.res(nr_of_elem = 1, start_index = 0, data = 1) in this case."
But this sentence is probably only true, if nobody wrote a zero to index 0 of the property before.
I'm not sure if writes of values different from zero to index 0 are valid or not. Maybe the test cases in 8/*/* have some examples for this.

nanosonde, 2019-12-09:
> You are right. PropertyValueRead already seems to do the right thing.
Unfortunately, it does not. You can test it with the free version of [Net'N'Node](https://www.weinzierl.de/index.php/de/alles-knx1/software-tools/net-n-node).

Scan the device to get the list of all objects with their property descriptions. This works flawlessly.
But if you now click on a property and want to read out the value, then Net'N'Node tries to get the number of array entries first by reading array index 0. However, the current implementation returns the real value directly instead of the length resulting in error message that Net'N'Node expected 2 bytes for example, but received 6 bytes (e.g. reading PID_SERIAL_NUMBER).

PropertyWrite is also broken and does not handle requests to array index 0 properly.

> I'm not sure if writes of values different from zero to index 0 are valid or not. Maybe the test cases in 8/_/_ have some examples for this.

The test cases in 8/3/7 (AIL, A_PropValue_* tests) and 8/6/3 (EMI-IMI, M_Prop_* tests)  give some more insights. Only the latter gives concrete binary data examples.

This is also interesting: https://github.com/calimero-project/calimero-device/blob/master/src/main/java/tuwien/auto/calimero/device/ios/InterfaceObjectServer.java#L852
https://github.com/calimero-project/calimero-device/blob/master/src/main/java/tuwien/auto/calimero/device/ios/InterfaceObjectServer.java#L743

thelsing, 2019-12-09:
I guess it would be best to refactor the properties to to new class, which includes the information of the PropertyDescription class. The interface objects get an array of properties each (analog to PropertyDescription now). Most properties will keep their data themself. Some properties (IP-addresses for example) use lamdas or function-pointers to get/set their data. 
The propertyread/write methods could be complete generic in the InterfaceObject class.

We could also create a bitflag-int32 to each interfaceObject to indicate if a property is set or not, and set/check it on propertywrite/read instead.

Do you have any better, or different idea?

nanosonde, 2019-12-10:
> I guess it would be best to refactor the properties to to new class[...]

Yes, I think that would be the best way.

> We could also create a bitflag-int32 to each interfaceObject to indicate if a property is set or not, and set/check it on propertywrite/read instead.

Do you think it is necessary to have this bitflag?
You can always check if the property value is empty, i.e. index 0 is 0.

What should be implemented is a hook which gets called if a property gets modified:
https://github.com/thelsing/knx/blob/master/src/knx/cemi_server.cpp#L332
I could not implement M_PropInfo.ind properly as this there is no internal notification mechanism if a property has been modified.
For this you could use a bitmask which marks the property as "dirty" after writing to it and afterwards call the registered callbacks. On the other hand the callback could also be called directly in the call flow of the PropertyValueWrite() as all the transactions are atomic.

nanosonde, 2019-12-10:
Another thing:
We should also add the basics for FunctionProperties. This is not really a big deal.
Instead of using generic storage, a function is called with the request data and this function must provide an answer with response data in a timely manner. I read about one second in the KNX spec. Also Function Properties are only used during ETS programming and not during runtime in any way.

So next to read and write hooks for a property that are called BEFORE(read) or AFTER(write), there should be a third hook for Function Properties. This hook would be called if M_FuncPropCommand_req, M_FuncPropStateRead_req, A_FunctionPropertyCommand or A_FunctionPropertyState is called.

nanosonde, 2019-12-10:
Currently I am asking myself if is really necessary to have those full interface objects really exist as C++ objects in the code. Wouldn't it be enough to have one big table which contains all objects and properties? 
Those classes that internally need access during runtime could use the same generic PropertyValueRead/Write methods to access them instead of calling methods of instantiated interface objects. They could also register for updates if they really have to be notified about changes.

nanosonde, 2019-12-10:
However, this would probably cause to much trouble for the table objects, right?

nanosonde, 2019-12-10:
Could you make a suggestion on how the Property class shall look like (declaration)?
Maybe I find some time to work on it.

thelsing, 2019-12-10:
> > We could also create a bitflag-int32 to each interfaceObject to indicate if a property is set or not, and set/check it on propertywrite/read instead.
> 
> Do you think it is necessary to have this bitflag?
> You can always check if the property value is empty, i.e. array index 0 contains value 0.

The bitflag would be an alternative to refactoring properties. So eveything could stay as it is. Only this bitflag int would have to be added to each interface object. But I like the other solution better.

thelsing, 2019-12-10:
I also thought about one big table for all properties of all interface-objects. But I don't think that you would save memory with this, because you would need the interface object-ID for each property.
i.e.
{{ifID, PropId, propVal, ...}, ...}

With separate classes for the interface object you basicly have something like
{{ifId1, {{PropId1, Val1, ...},{Prop2, Val2, ...} }, {ifId2, {{PropId1, Val1, ...},{Prop2, Val2, ...} }, ...}

The hooks for propertychanges should only be needed for changes that happen because of PropertyWrite-telegrams, right? If so I think the bau-class should just call these hooks.

Function properties are fine with me. Where are they used? Maybe they should get an own class?

nanosonde, 2019-12-10:
> The bitflag would be an alternative to refactoring properties. So eveything could stay as it is. Only this bitflag int would have to be added to each interface object. But I like the other solution better.

If everything would stay as it is, then one of the main issues that property values are always arrays would still exist.

I would also like to have another solution.

nanosonde, 2019-12-10:
> The hooks for propertychanges should only be needed for changes that happen because of PropertyWrite-telegrams, right? If so I think the bau-class should just call these hooks.

I don't understand exactly what you mean. So a lamba or function pointer which is attached to a property would be called in the context of a bau-class method?

 
> Function properties are fine with me. Where are they used? Maybe they should get an own class?

Would be low prio, but I think we should keep it in mind. They are used for diagnosing RF link quality from within ETS for example. MDT has an ETS App for this. However, I do not know if this is vendor-specific implemenation. The KNX-RF spec. specifies some RF diag stuff using FuncProps though.

Some questions:
* How do you want to instantiate each property if you have a property class? 
Or did I misunderstand something? 
* How do you want to handle the property value which is always an array?

thelsing, 2019-12-10:
I'm not sure if the propertydescription should be separate from the Property-class or not. The descriptions are static, the propertyvalues are not. Maybe it would be best to keep them separate. That way we don't get problems if we need more instances of a  particular interface object type.

So something like:
class Property
{virtual read = 0, virtual write = 0}
class DataProperty : Property
{ uint8_t elementCount(), uint8_* data()}
for simple properties like DeviceObject:PID_ROUTING_COUNT. Maybe different classes for different (groups of?) PropertyDataTypes.
class LambdaProperty : Property
{
readcallback;
writecallback;
} for things like Loadstate or the Table properties.

The propertySize, readProperty and writeProperty methods should only be needed in the interface_object class after this. A global function to get the size of a PropertyDataType would probably needed to do this. 
The data for the properties should only be stored int the new Property classes. The getters/setters of properties, that are not used in the stack could be removed. 

As you see I didn't realy thought about it yet. So feel free to propose something different.

nanosonde, 2019-12-10:
Maybe we should write down some requirements:
* minimal memory footprint
* no runtime callback registrations, do it during compile time statically
* properties must be persistable (save/restore from NVM)
* ...

thelsing, 2019-12-10:
> I don't understand exactly what you mean. So a lamba or function pointer which is attached to a property would be called in the context of a bau-class method?

No I thought about just adding cemiserver.PropertWriteInd(...) at https://github.com/thelsing/knx/blob/aa230ee0b97b0563a53c1b38997a2d20797e8502/src/knx/bau_systemB.cpp#L207
(same for read)

The running application shouldn't change properties, should they? They only operate on the GroupObjects. So changes of properties alway happen because of user of the cemiserver or because of a telegram.

thelsing, 2019-12-10:
> * How do you want to instantiate each property if you have a property class?
>   Or did I misunderstand something?

The same way that _propertyDescriptions work now. Only non static and probably with an constructor call.d

> * How do you want to handle the property value which is always an array?
The same as a normal DataProperty. The Value-Pointer points to memory of elementCount*propertyDataSize Bytes. So you can just memcpy from/to it during propertyRead/Write.

thelsing, 2019-12-10:
Should I create a gitter chat?

nanosonde, 2019-12-10:
> 
> 
> Should I create a gitter chat?

Never used it. Maybe we should give it a try. 😃

thelsing, 2019-12-10:
https://gitter.im/thesing-knx/community

nanosonde, 2019-12-10:
> No I thought about just adding cemiserver.PropertWriteInd(...) at
> 
> https://github.com/thelsing/knx/blob/aa230ee0b97b0563a53c1b38997a2d20797e8502/src/knx/bau_systemB.cpp#L207
> 
> (same for read)

Ah, I see. Yes, good idea. 
 
> The running application shouldn't change properties, should they? They only operate on the GroupObjects. So changes of properties alway happen because of user of the cemiserver or because of a telegram.

Mhmm, probably not the application, but for example the property which indicates the bus connection status. It is changed during runtime as far as I understood. 
Maybe it is a good idea if the local user application also has access to interface objects so that it can get relevant data in a standard way. So instead of implementing some extra function which gives back bus connection status, it could query the corresponding property.

thelsing, 2020-06-24:
RfMedium object is done too. This issue should be fixed now, shouldn't it?

nanosonde, 2020-06-24:
Have you tested property reading/writing with Net'N'Node from Weinzierl?
I remember that the correct length returned on index 0 was an issue.

nanosonde, 2020-07-01:
I think I have found a bug.
Please see here: https://github.com/thelsing/knx/blob/fda8e64425ac862dc84f91f1c73615459b359f3d/src/knx/data_property.cpp#L21

I think it should be 
```
memcpy(data, _data + (start * ElementSize()), count * ElementSize()); 
```

start and count are based on the index. However, here we are operating on the raw data pointer and we have to take into account the element size.

nanosonde, 2020-07-02:
Here is another one:
https://github.com/thelsing/knx/blob/fda8e64425ac862dc84f91f1c73615459b359f3d/src/knx/data_property.cpp#L69

thelsing, 2020-07-02:
Thanks. I just changed this.

nanosonde, 2020-07-03:
https://github.com/thelsing/knx/pull/75

Now I am able to read almost everything with Net'N'Node tool while scanning the properties and querying them.
However, all table properties still seem to have issue when trying to read the PID_TABLE.

nanosonde, 2020-07-05:
When https://github.com/thelsing/knx/pull/75 gets merged, I think we can close this issue.

closed 2020-07-05 by thelsing