thelsing/knx #53: KNX-RF Device

mjm987, 2020-03-09:
According to ./doc/knx_rf_notes.md I assume it should be possible to build a KNX-RF device using a SAMD21 board but I can't find any rule or example project for this.
Could you add some HOWTO notes for dummies like me?  (preferably for linux host system)
Thx

thelsing, 2020-03-11:
Is the problem the software or the hardware part? I didn't code the knx-rf stuff. Thats from @nanosonde . But I can help to create the knx-demo for an samd21 board if this helps.

mjm987, 2020-03-11:
I would be interested in the software part. A knx-demo which supports  knx-rf on the samd21 board on cmake/Makefile base would be really great!

thelsing, 2020-03-12:
The easiest way to build the software is the arduino-ide. Just add the repository as a lib. Edit knx/config.h and set MEDIUM_TYPE to 2. Then the knx-demo example should build a knx-rf version.

Other people also use platform-io to build firmwares. It should be possible to create a cmake, but I don't know how hard it is to use the arduino-core libs with only a cmake file.

mjm987, 2020-03-13:
ok thanks a lot, this will help :-)

thelsing, 2020-03-13:
If you every create a cmake file to build a firmware, please let me know. ;)

mjm987, 2020-04-29:
Partially I had success:
After changing media type and removing USE_TP I was able to program the SAMD21 MINI by using Arduino IDE (using "Arduino Zero" board setup and adapting variant.h to use the correct SPI on Pins 10..13).

On ETS5 I installed your example knx-demo-rf.knxprod and tried to Program the device, but it seems the board does not answer as expected. According the error output of ETS5 the device is recognised as set in program mode but programming of the physical address fails.
After enabling the log output in rf_data_link_layer.cpp I get following output on the Serial1 line:
"
progmode on
RX LFN: 7 len: 10 data:  data: 29 00 10 60 FF FA 12 04 00 80 
RX LFN: 0 len: 11 data:  data: 29 00 10 60 FF FA 12 04 01 43 00 
RX LFN: 1 len: 10 data:  data: 29 00 10 60 FF FA 12 04 00 81 
RX LFN: 2 len: 16 data:  data: 29 00 00 E0 FF FA 00 00 06 01 C8 00 00 00 B0 01 
RX LFN: 3 len: 23 data:  data: 29 00 00 E0 FF FA 00 00 0D 03 EE 00 FA 00 00 00 00 00 FA 51 03 EE AF 
RX LFN: 4 len: 17 data:  data: 29 00 00 E0 FF FA 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 5 len: 17 data:  data: 29 00 00 E0 FF FA 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 6 len: 17 data:  data: 29 00 00 E0 FF FA 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 7 len: 17 data:  data: 29 00 00 E0 FF FA 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 0 len: 23 data:  data: 29 00 00 E0 FF FA 00 00 0D 03 EE 00 FA 00 00 00 00 00 FA 51 03 EE AF 
RX LFN: 1 len: 17 data:  data: 29 00 00 E0 FF FA 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 2 len: 17 data:  data: 29 00 00 E0 FF FA 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 3 len: 17 data:  data: 29 00 00 E0 FF FA 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 4 len: 17 data:  data: 29 00 00 E0 FF FA 00 00 07 03 EC 00 FA 00 00 00 00 
"

Do you have any idea what could be wrong?

nanosonde, 2020-04-29:
Hi @mjm987,

what are you using to program RF devices?
A KNX RF USB stick? A TP-RF coupler? What model?

mjm987, 2020-04-29:
I'm using a Weinzierl KNX RF/TP Coupler 672 on ETS 5.7.3

mjm987, 2020-04-29:
When enabling TX debugging as well and ETS Programm Adress only I get this:
RX LFN: 6 len: 16 data:  data: 29 00 00 E0 11 02 00 00 06 01 C8 00 00 00 B0 01 
TX LFN: 0 len: 35 data:1C 44 FF 02 00 FA 00 00 00 00 F4 37 00 00 00 00 00 E0 01 C9 00 00 00 B0 01 00 FA 00 90 0C 00 00 00 FF FF 
RX LFN: 7 len: 23 data:  data: 29 00 00 E0 11 02 00 00 0D 03 EE 00 FA 00 00 00 00 00 FA 51 03 EE AF 
RX LFN: 0 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 1 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 2 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 3 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 4 len: 23 data:  data: 29 00 00 E0 11 02 00 00 0D 03 EE 00 FA 00 00 00 00 00 FA 51 03 EE AF 
RX LFN: 5 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 6 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 7 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
RX LFN: 0 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 

so there is only a single TX package after the first RX ?!

nanosonde, 2020-04-29:
The RX packets seem to be fine:
https://www.dehof.de/eib/pdfs/EMI-FT12-Message-Format.pdf
It just shows a cEMI frame without any additonal info (add info len is 0).

In the KNX spec (03_02_05 Communication Medium RF v01.06.03 AS.pdf) you can see the TX frame format in chapters 6.1.2.4 and 6.1.2.5.
RF-info: 0x02 [OK]
Serial/Domain Address:  00 FA 00 00 00 00 [default in the code]
CRC block1:  F4 37
KNX CTRL field: 0x00 [OK]
source address: 00 00
destination address: 00 00
LPCI: 0xE0 (SN/DoA is KNX serial, dest. address is group address, RC: 0x6, LFN is 0x0) [OK]
APCI type is: 0x01C9
--- data for this APCI: 00 00 00 B0 01 00 FA 00 00 00 00
(there are two further checksums: one in between the data (90 0C) and the final checksum: FF FF)

From [here](https://github.com/thelsing/knx/blob/master/src/knx/knx_types.h#L123) you can see the APCI types. In this case, TX is the SystemNetworkParameterResponse message which is correct as the previous message in RX is SystemNetworkParameterRead.

From a first look, I would say that it looks ok so far.

nanosonde, 2020-04-29:
I would like to recommend to get free license for the KNX monitor from Weinzierl:
https://www.weinzierl.de/index.php/de/alles-knx1/software-tools/net-n-node

It is really useful while debugging as you can see the message names and the decoded payload immediately.

nanosonde, 2020-04-29:
Some other thoughts:

- How far away is your DIY KNX-RF device from the KNX RF/TP coupler?
- Do you have configured any filtering on your coupler?

nanosonde, 2020-04-29:
Concerning the filtering:
Might it be that the coupler filters out system broadcast messages and only accepts normal broadcast messages?

On KNX-RF (and only on open media) you have those two broadcast types.
systembroadcast is really a broadcast to all listing KNX-RF devices while
normal broadcast is just within your domain (=domain address). 
That's the reason why you need the additonal domain address.
It just "simulates" a closed medium like TP by adding the domain address.

mjm987, 2020-04-29:
Some more debug information - looks like the DomainAddressSerialNumberWrite does not work...

RX LFN: 2 len: 16 data:  data: 29 00 00 E0 11 02 00 00 06 01 C8 00 00 00 B0 01 
DataLinkLayer::frameRecieved()
NetworkLayer::systemBroadcastIndication()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication:SystemNetworkParameterRead
NetworkLayer::dataSystemBroadcastRequest
TX LFN: 0 len: 35 data:1C 44 FF 02 00 FA 00 00 00 00 F4 37 00 00 00 00 00 E0 01 C9 00 00 00 B0 01 00 FA 00 90 0C 00 00 00 FF FF 
NetworkLayer::systemBroadcastConfirmation()
RX LFN: 3 len: 23 data:  data: 29 00 00 E0 11 02 00 00 0D 03 EE 00 FA 00 00 00 00 00 FA 51 03 EE AF 
DataLinkLayer::frameRecieved()
NetworkLayer::systemBroadcastIndication()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication:DomainAddressSerialNumberWrite
RX LFN: 4 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
DataLinkLayer::frameRecieved()
NetworkLayer::systemBroadcastIndication()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication:DomainAddressSerialNumberRead
RX LFN: 5 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
DataLinkLayer::frameRecieved()
NetworkLayer::systemBroadcastIndication()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication:DomainAddressSerialNumberRead
RX LFN: 6 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
DataLinkLayer::frameRecieved()
NetworkLayer::systemBroadcastIndication()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication:DomainAddressSerialNumberRead
RX LFN: 7 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
DataLinkLayer::frameRecieved()
NetworkLayer::systemBroadcastIndication()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication:DomainAddressSerialNumberRead
RX LFN: 0 len: 23 data:  data: 29 00 00 E0 11 02 00 00 0D 03 EE 00 FA 00 00 00 00 00 FA 51 03 EE AF 
DataLinkLayer::frameRecieved()
NetworkLayer::systemBroadcastIndication()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication:DomainAddressSerialNumberWrite
RX LFN: 1 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
DataLinkLayer::frameRecieved()
NetworkLayer::systemBroadcastIndication()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication:DomainAddressSerialNumberRead
RX LFN: 2 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
DataLinkLayer::frameRecieved()
NetworkLayer::systemBroadcastIndication()

mjm987, 2020-04-29:
Does the device expect packages on the new address after writing the domain address??

mjm987, 2020-04-29:
> Concerning the filtering:
> Might it be that the coupler filters out system broadcast messages and only accepts normal broadcast messages? (Some coupler security setting that you should be able to change.)
> 
> On KNX-RF (and only on open media) you have those two broadcast types.
> systembroadcast is really a broadcast to all listing KNX-RF devices while
> normal broadcast is just within your domain (=domain address).
> That's the reason why you need the additonal domain address.
> It just "simulates" a closed medium like TP by adding the domain address.

At least with some other commercial RF-Devices I had no problem to "program" them

nanosonde, 2020-04-29:
Did you get it working?
I am asking, because you closed the issue.

mjm987, 2020-04-29:
no I didn't close but possibly telsing after responding the initial answer

mjm987, 2020-04-29:
what could be the reason that a ApplicationLayer::dataSystemBroadcastIndication() does not send a confirmation package?

mjm987, 2020-04-30:
The problem is that  empty functions of bau.cpp are called in ApplicationLayer::dataSystemBroadcastIndication instead those of Bau27B0.cpp

Here my logging:
ApplicationLayer::dataSystemBroadcastIndication: case DomainAddressSerialNumberWrite
 BusAccessUnit::domainAddressSerialNumberWriteIndication --> empty function

ApplicationLayer::dataSystemBroadcastIndication: case DomainAddressSerialNumberRead
BusAccessUnit::domainAddressSerialNumberReadIndication --> empty function

Unfortuantely I have no AVR_ICE adapter to debug but only the Arduino-IDE :-(

Obviously I configured something wrong

thelsing, 2020-04-30:
Can you put your sketch somewhere? You probably got the wrong Bau in knx-facade.h

mjm987, 2020-04-30:
The skech I used is the examples/knx_demo/knx-demo.ino with following changes:
- including at top of the file <FlashAsEEPROM.h> and  <FlashStorage.h> 
- changing all references of Serial to Serial1 (so I get Results on RS232 pins)

Further, according the Map file, the correct Bau (Bau23B0) is used in knx-facade is used:
grep KnxFacade  /tmp/arduino_build_885568/knx-demo.ino.map | tail -1
vtable for KnxFacade<SamdPlatform, Bau27B0>       /tmp/arduino_build_885568/libraries/telsing-knx/knx_facade.cpp.o

But for eg. domainAddressSerialNumberReadIndication both versins are linked according the Mapfile:  (but I think this is ok?)
Bau27B0::domainAddressSerialNumberReadIndication(Priority, HopCountType, unsigned char*) /tmp/arduino_build_885568/libraries/telsing-knx/knx/bau27B0.cpp.o
BusAccessUnit::domainAddressSerialNumberReadIndication(Priority, HopCountType, unsigned char const*) /tmp/arduino_build_885568/libraries/telsing-knx/knx/bau.cpp.o

The config.h file is defined as follows:
#pragma once
#ifdef ARDUINO_ARCH_SAMD
#define SPI_SS_PIN 10
#define GPIO_GDO2_PIN 9
#define GPIO_GDO0_PIN 7
#endif
#define MEDIUM_TYPE 2
#define USE_RF

And last but not least, the compiler uses following options eg. when compiling knx-facade.cpp:
/home/mm/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10900 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Zero\"" -I/home/mm/.arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/ -I/home/mm/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/home/mm/.arduino15/packages/arduino/hardware/samd/1.8.6/cores/arduino -I/home/mm/.arduino15/packages/arduino/hardware/samd/1.8.6/variants/arduino_zero -I/home/mm/Comp/Arduino/libraries/FlashStorage/src -I/home/mm/Comp/Arduino/libraries/telsing-knx/src -I/home/mm/.arduino15/packages/arduino/hardware/samd/1.8.6/libraries/SPI /home/mm/Comp/Arduino/libraries/telsing-knx/src/knx_facade.cpp -o /dev/null

Looks all ok to me, doesn't it?

mjm987, 2020-05-01:
If I declare BusAccessUnit::domainAddressSerialNumberWriteIndication() as pure abstract I get following error when compiling the project:
In file included from /home/mm/Comp/Arduino/libraries/telsing-knx/src/knx_facade.cpp:1:0:
/home/mm/Comp/Arduino/libraries/telsing-knx/src/knx_facade.h: In instantiation of 'KnxFacade<P, B>::KnxFacade() [with P = SamdPlatform; B = Bau27B0]':
/home/mm/Comp/Arduino/libraries/telsing-knx/src/knx_facade.cpp:11:46:   required from here
/home/mm/Comp/Arduino/libraries/telsing-knx/src/knx_facade.h:36:50: error: invalid new-expression of abstract class type 'Bau27B0'
     KnxFacade() : _platformPtr(new P()), _bauPtr(new B(*_platformPtr)), _bau(*_bauPtr)

                                                  ^~~~~~~~~~~~~~~~~~~~
In file included from /home/mm/Comp/Arduino/libraries/telsing-knx/src/knx_facade.h:13:0,
                 from /home/mm/Comp/Arduino/libraries/telsing-knx/src/knx_facade.cpp:1:
/home/mm/Comp/Arduino/libraries/telsing-knx/src/knx/bau27B0.h:12:7: note:   because the following virtual functions are pure within 'Bau27B0':
 class Bau27B0 : public BauSystemB
       ^~~~~~~
In file included from /home/mm/Comp/Arduino/libraries/telsing-knx/src/knx/bau_systemB.h:4:0,
                 from /home/mm/Comp/Arduino/libraries/telsing-knx/src/knx/bau07B0.h:4,
                 from /home/mm/Comp/Arduino/libraries/telsing-knx/src/knx_facade.h:12,
                 from /home/mm/Comp/Arduino/libraries/telsing-knx/src/knx_facade.cpp:1:
/home/mm/Comp/Arduino/libraries/telsing-knx/src/knx/bau.h:117:18: note: 	virtual void BusAccessUnit::domainAddressSerialNumberWriteIndication(Priority, HopCountType, const uint8_t*, const uint8_t*)
     virtual void domainAddressSerialNumberWriteIndication(Priority priority, HopCountType hopType, const uint8_t* rfDoA,
                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

mjm987, 2020-05-01:
By the way, same behaviour on the raspberry pi examples/knx-linux platform (no response to DomainAddressSerialNumberRead/Write request).

thelsing, 2020-05-01:
I suggest adding some logging to https://github.com/thelsing/knx/blob/894a54fa7f4ca66e8a8b5dd06eb8bedc1a8d4108/src/knx/bau27B0.cpp#L130

I guess the serial number send from ets doesn't match with the configured one in the sketch?

mjm987, 2020-05-01:
Thx for suggesting, but I allready checkt that and the check is passed (both addresses are same).

And the problem is that not Bau27B0::domainAddressSerialNumberReadIndication() is called but BusAccessUnit::domainAddressSerialNumberReadIndication() 
(from the base class BusAccessUnit instead of the overriden from class Bau27B0 - see log above.

mjm987, 2020-05-01:
... possibly about different method signature? Some function parameters in Bau27B0 are declared as const but in BusAccessUnit they are declared without const?

thelsing, 2020-05-02:
That was wrong indeed. I corrected this in the master branch. Can you try again please?

mjm987, 2020-05-03:
Great, one step further, Bau27B0::domainAddressSerialNumberReadIndication is sent now ;-)

But the ETS5 isn't happy with the answer of domainAddressSerialNumberReadIndication()
Could you check what could be wrong with the the response packet at bottom?

(Btw: I requested the KNX monitor from Weinzierl some days ago as @nanosonde suggested but unfortunately  Ididn't get a license file yet)

Log at SAMD21 device:
...
RX LFN: 7 len: 23 data:  data: 29 00 00 E0 11 02 00 00 0D 03 EE 00 FA 00 00 00 00 00 FA 51 03 EE AF 
DataLinkLayer::frameRecieved()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication: case DomainAddressSerialNumberWrite
Bau27B0::domainAddressSerialNumberWriteIndication
RfMediumObject::rfDomainAddress() write <-- 00 FA 51 03 EE AF 
RX LFN: 0 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
DataLinkLayer::frameRecieved()
ApplicationLayer::dataSystemBroadcastIndication
ApplicationLayer::dataSystemBroadcastIndication: case DomainAddressSerialNumberRead
Bau27B0::domainAddressSerialNumberReadIndication
RfMediumObject::rfDomainAddress() read -->00 FA 51 03 EE AF 
TX LFN: 3 len: 36 data:1D 44 FF 02 00 FA 00 00 00 00 CF A4 00 00 00 00 00 E6 03 ED 00 FA 00 00 00 00 00 FA C8 14 51 03 EE AF 64 69 
^ --- after sending this back, the ETS throws an error and stopps "programming"

nanosonde, 2020-05-03:
@mjm987 
Could also check the ETS log at: C:\Users\<your username>\AppData\Local\KNX\ETS5\Log\ETS5.log?
What does it say?

mjm987, 2020-05-03:
ok, see below ...

on TX above I guess the last few bytes are the respose of the Address:
TX LFN: ..... ED 00 FA 00 00 00 00 00 FA C8 14 51 03 EE AF 64 69
is this address correct, shouldn't it be:
TX LFN: ..... ED 00 FA 00 00 00 00 00 FA 51 03 EE AF ...
(FA 51 03 EE AF instead of FA C8 14 51 ... ??)

here the ETS Log:
...
2020-05-03 13:44:35,321 [20] INFO  Knx.Ets.Hawk.NetworkFacade - Programming IA 1.2.4
2020-05-03 13:44:35,321 [20] INFO  Knx.Falcon.Network - BeginIndividualAddressWriteBySerialNumber with address 1.2.4 and serial number 00FA:00000000
2020-05-03 13:44:35,322 [20] INFO  Knx.Falcon.Network - EndIndividualAddressWriteBySerialNumber
2020-05-03 13:44:38,332 [20] WARN  Knx.Ets.Hawk.NetworkFacade - LoadIndividualAddressInProgMode failed: Die physikalische Adresse des Geräts kann nicht programmiert werden.
2020-05-03 13:44:38,334 [20] WARN  Knx.Ets.Hawk.LoadDeviceTask - Knx.Ets.Hawk.NetworkManagementException: Die physikalische Adresse des Geräts kann nicht programmiert werden.
   bei acg.fv(Object A_0, Byte[] A_1, UInt16 A_2, IProgress`1 A_3, CancellationToken A_4)
   bei acq.a(aaa A_0, ab5 A_1, IProgress`1 A_2, CancellationToken A_3, Nullable`1& A_4, Int32& A_5, Action`2 A_6, Action`1 A_7, Func`2 A_8, Func`2 A_9, Func`2 A_10, Func`1 A_11, InternalLoadDeviceOptions A_12, Func`1 A_13)
   bei abl.a(aaa A_0, ab5 A_1, IProgress`1 A_2, CancellationToken A_3, Nullable`1& A_4, Int32& A_5, Action`2 A_6, Action`1 A_7, Func`2 A_8, Func`2 A_9, Func`2 A_10, Func`1 A_11, InternalLoadDeviceOptions A_12, Func`1 A_13)
   bei abk.a(IProgress`1 A_0, CancellationToken A_1, Nullable`1& A_2, Int32& A_3, Boolean& A_4)
   bei abk.db(IProgress`1 A_0, CancellationToken A_1)
   bei aaw.e.e()
2020-05-03 13:44:38,348 [1] ERROR Knx.Ets.ViewModel.OnlineOperations.Download.PendingOperation - Error during executing OnStateChanged method of object abk
Knx.Ets.Hawk.NetworkManagementException: Die physikalische Adresse des Geräts kann nicht programmiert werden.
   bei acg.fv(Object A_0, Byte[] A_1, UInt16 A_2, IProgress`1 A_3, CancellationToken A_4)
   bei acq.a(aaa A_0, ab5 A_1, IProgress`1 A_2, CancellationToken A_3, Nullable`1& A_4, Int32& A_5, Action`2 A_6, Action`1 A_7, Func`2 A_8, Func`2 A_9, Func`2 A_10, Func`1 A_11, InternalLoadDeviceOptions A_12, Func`1 A_13)
   bei abl.a(aaa A_0, ab5 A_1, IProgress`1 A_2, CancellationToken A_3, Nullable`1& A_4, Int32& A_5, Action`2 A_6, Action`1 A_7, Func`2 A_8, Func`2 A_9, Func`2 A_10, Func`1 A_11, InternalLoadDeviceOptions A_12, Func`1 A_13)
   bei abk.a(IProgress`1 A_0, CancellationToken A_1, Nullable`1& A_2, Int32& A_3, Boolean& A_4)
   bei abk.db(IProgress`1 A_0, CancellationToken A_1)
   bei aaw.e.e()
   bei System.Threading.Tasks.Task.InnerInvoke()
   bei System.Threading.Tasks.Task.Execute()
   bei aaw.a(IAsyncResult A_0)
   bei Knx.Ets.ViewModel.OnlineOperations.Download.PendingOperation.OnStateChanged(Object sender, StateChangedEventArgs e)

nanosonde, 2020-05-03:
> ok, see below ...
>
> on TX above I guess the last few bytes are the respose of the Address:
> TX LFN: ..... ED 00 FA 00 00 00 00 00 FA C8 14 51 03 EE AF 64 69
> is this address correct, shouldn't it be: [...]

Whole frame:
TX LFN: 3 len: 36 data:1D 44 FF 02 00 FA 00 00 00 00 CF A4 00 00 00 00 00 E6 03 ED 00 FA 00 00 00 00 00 FA C8 14 51 03 EE AF 64 69

The frame format itself is correct. 
The first block always contains 10 bytes followed by two CRC bytes: CF A4
Serial/DoA field in first block is: 00 FA 00 00 00 00
All following blocks will automatically get a CRC16 (two bytes) inserted after each 16 bytes.
Second block starts with:
KNX Ctrl field: 00
Src addr: 00 00 
Dst addr: 00 00
LPCI: 0xE6 (SN/DoA is KNX serial, dest. address is group address, RC: 0x6, LFN is 0x0) [OK]
APCI type is: 0x03ED (DomainAddressSerialNumberResponse )
--- data for this APCI: 00 FA 00 00 00 00 00 FA 51 03 EE AF (CRCs removed)
(there are two further checksums: one in between the data (C8 14) and the final checksum: 64 69)

If you look at the KNX spec. (03_03_07 Application Layer v01.06.02 AS.pdf) on page 37 (Figure 28), I can see that the response contains the serial number: 00 FA 00 00 00 00 and the domain address: 00 FA 51 03 EE AF.
So the response looks correct.

> here the ETS Log:

ok, thanks. nothing really interesting the ETS log.

The management procedure for writing the individual address is defined in 03_05_02 Management Procedures v01.09.02 AS.pdf on page 15.

Does the ETS send this error immediately or does it more look like a timeout in the ETS when you see the error?

mjm987, 2020-05-04:
After the last TX, it takes about 3s  until ETS reports the error.

When I execute two times "Program address" the behaviour is not same, so setting the domain address seems to works but ETS does not understand the 2nd TX and because ETS does not retry, I guess the result isn't lost but invalid.

1st Time "Program address":
RX domain address does not match. Skipping...
RX domain address does not match. Skipping...
RX LFN: 7 len: 16 data:  data: 29 00 00 E0 11 02 00 00 06 01 C8 00 00 00 B0 01 
TX LFN: 0 len: 35 data:1C 44 FF 02 00 FA 00 00 00 00 F4 37 00 00 00 00 00 E0 01 C9 00 00 00 B0 01 00 FA 00 90 0C 00 00 00 FF FF 
RX LFN: 0 len: 23 data:  data: 29 00 00 E0 11 02 00 00 0D 03 EE 00 FA 00 00 00 00 00 FA 51 03 EE AF 
RX LFN: 1 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
TX LFN: 1 len: 36 data:1D 44 FF 02 00 FA 00 00 00 00 CF A4 00 00 00 00 00 E2 03 ED 00 FA 00 00 00 00 00 FA 84 C6 51 03 EE AF 64 69 

2nd time "Program address": 
RX LFN: 2 len: 10 data:  data: 29 00 10 60 11 02 12 04 00 80 
RX LFN: 3 len: 11 data:  data: 29 00 10 60 11 02 12 04 01 43 00 
RX LFN: 4 len: 16 data:  data: 29 00 00 E0 11 02 00 00 06 01 C8 00 00 00 B0 01 
TX LFN: 2 len: 35 data:1C 44 FF 02 00 FA 00 00 00 00 F4 37 00 00 00 00 00 E4 01 C9 00 00 00 B0 01 00 FA 00 DC DE 00 00 00 FF FF 
RX LFN: 5 len: 16 data:  data: 29 00 00 E0 11 02 00 00 06 01 C8 00 00 00 B0 01 
TX LFN: 3 len: 35 data:1C 44 FF 02 00 FA 00 00 00 00 F4 37 00 00 00 00 00 E6 01 C9 00 00 00 B0 01 00 FA 00 FA B7 00 00 00 FF FF 
RX LFN: 6 len: 23 data:  data: 29 00 00 E0 11 02 00 00 0D 03 EE 00 FA 00 00 00 00 00 FA 51 03 EE AF 
RX LFN: 7 len: 17 data:  data: 29 00 00 E0 11 02 00 00 07 03 EC 00 FA 00 00 00 00 
TX LFN: 4 len: 36 data:1D 44 FF 02 00 FA 00 00 00 00 CF A4 00 00 00 00 00 E8 03 ED 00 FA 00 00 00 00 00 FA 3B 0B 51 03 EE AF 64 69 

So it seams the ETS does not understand the last TX package?

mjm987, 2020-05-04:
Btw: thx for explanation of the KNX package - helped me a lot to understand ;-)

On RX-frames I guess "11 02" is the the physical address of the ETS side, isn't it?
Shouldn't  it used on TX frame as destination address instead of "00 00" ?

nanosonde, 2020-05-04:
I have just revived by KNX RF HW setup that I have used during development.
I did a clean install of PlatformIO in Visual Studio Code on Windows 10 and did a fresh checkout of the current master branch.
I was able to program the individual address of my DIY KNX RF device with ETS 5.7.4 without any problem.

I will try to capture the packets...
BTW: Have you tried to get the free license from Weinzierl?
I had recieved it quite fast.
My TP/RF coupler is a MDT RF-LK001.02.

nanosonde, 2020-05-04:
Debug output while programming the individual address:
```
start
readMemory
RESTORED FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
saved memory doesn't match manufacturerId, version or hardwaretype
manufacturerId: FFFF FA
version: FFFF 3
hardwareType: FF FF FF FF FF FF
 00 00 00 00 00 00
Partnumber: 0x0
Version   : 0x14
ownaddr 0
progmode on
save saveRestores 3
-10
.
200012DC
-27
.
20001E9C
-33
.
200002C8
save tableobjs 4
-33
.
20001378
-40
.
2000132▒start
readMemory
RESTORED 00 FA 00 00 00 00 00 00 00 03 13 04 00 00 00 01 00 00 01 00 00 00 00 00 00 00 00 00 FA F9 77 F4 23 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF
manufacturerId, version and hardwareType matches
saverestores 3
-10
.
-27
.
-33
.
restored saveRestores
tableObjs 4
-33
.
-40
.
-47
.
-54
.
restored Tableobjects
Partnumber: 0x0
Version   : 0x14
ownaddr 1304
```
You can see that device received the address and the MCU was reset.
It now has the ownaddr 1304.

nanosonde, 2020-05-04:
```

Timestamp | Service | Src-Addr | Dest-Addr | Control | Prio | H-Cnt | TPCI | Sequ | APCI | AL-Data
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
04.05.2020 12:19:15,937 | L-Data.ind | 1.1.243 | Broadc | S | sys | 6 | U-Data |   | SystemNwParamRead | ObjType=0(Dev) PID=11(SerNo) TestInfo=01
04.05.2020 12:19:18,946 | L-Data.ind | 1.1.243 | Broadc | S | sys | 6 | U-Data |   | SystemNwParamRead | ObjType=0(Dev) PID=11(SerNo) TestInfo=01
04.05.2020 12:19:21,959 | L-Data.ind | 1.1.243 | Broadc | S | sys | 6 | U-Data |   | SystemNwParamRead | ObjType=0(Dev) PID=11(SerNo) TestInfo=01
04.05.2020 12:19:22,077 | L-Data.ind | 0.0.0 | Broadc | S | lo | 5 | U-Data |   | SystemNwParamResp | ObjType=0(Dev) PID=11(SerNo) TestInfoResp=01 00 FA 00 00 00 00
04.05.2020 12:19:24,997 | L-Data.ind | 1.1.243 | Broadc | S | sys | 6 | U-Data |   | DomAddrSerWrite | SerNo=00 FA 00 00 00 00 DomAddr=00 FA F9 77 F4 23
04.05.2020 12:19:25,306 | L-Data.ind | 1.1.243 | Broadc | S | sys | 6 | U-Data |   | DomAddrSerNoRead | SerNo=00 FA 00 00 00 00
04.05.2020 12:19:25,426 | L-Data.ind | 0.0.0 | Broadc | S | lo | 5 | U-Data |   | DomAddrSerResp | SerNo=00 FA 00 00 00 00 DomAddr=00 FA F9 77 F4 23
04.05.2020 12:19:25,727 | L-Data.ind | 1.1.243 | Broadc | S | sys | 6 | U-Data |   | IndAddrSerNoWrite | SerNo=00 FA 00 00 00 00 NewAddr=0x1304
04.05.2020 12:19:25,936 | L-Data.ind | 1.1.243 | Broadc | S | sys | 6 | U-Data |   | IndAddrSerNoRead | SerNo=00 FA 00 00 00 00
04.05.2020 12:19:26,052 | L-Data.ind | 1.3.4 | Broadc | S | lo | 5 | U-Data |   | IndAddrSerNoResp | SerNo=00 FA 00 00 00 00
04.05.2020 12:19:26,509 | L-Data.ind | 1.1.243 | Broadc | S | sys | 6 | U-Data |   | SystemNwParamRead | ObjType=0(Dev) PID=11(SerNo) TestInfo=01

```

nanosonde, 2020-05-04:
> So it seams the ETS does not understand the last TX package?

What is the distance between the TP/RF coupler and your KNX RF device?
What kind of antenna module are you using?

The problem might be that the the coupler gets broken packets quite often.

The whole "physical address" programming procedure does NOT use a connection-oriented approach with automatic repetitions, it just uses broadcasts to address the only device in programming mode.
Ok, it repeats the broadcasts packets up to 3(?) times, but if it does not an answer, it stops with an error.

The content of your packets looks good so far.
We really need to analyse the packets on TP side which your TP/RF coupler received to see if it received them at all.

From my experience with my own HW setup, I can tell you that I also had some issues with TX (but now RX) because of bad packets, because of the distance. I use my production KNX TP/RF coupler for my experiments and it is located on the other side of the house below a stairway(made from cement).

mjm987, 2020-05-04:
Many thx, seems quite ok compared with your answer.
Except in LPCI field where AT=1 which means LTE Group Address whereas on your devices answer it is 0.
Is this correct?

Unfortunately Weinzierl didn't send the License file, I asked now again.

mjm987, 2020-05-04:
The distance is ~5m. I guess it is ok because the behaviour on ETS side is different when the device is active or not.

mjm987, 2020-05-04:
Sorry my fault, after applying the code changes from @thelsing, I didn't check again the TP/RF coupler filter table.
Now I saw using wireshark that the response from the device didn't pass back.
Programming is woking now!
Regards and many many thx to both of you!!!!!
(and I learned a lot about C++ and KNX protocol!!!)

nanosonde, 2020-05-04:
@mjm987 Glad that you have solved it.

The filter table of the TP/RF coupler is something I do not understand completely.
In the preview table (ETS) I only see the group addresses, but not any physical address.
However, my coupler is configured to do filtering for both address types.
How exactly is filtering of physical addresses realized?

mjm987, 2020-05-04:
If I understand right, ETS should program the filter according defined groupadresses and physical adresses. 
But on my coupler, System Broadcast Telegram and Broadcast Telegram were barred on each direction but nevertheless all telegrams to the device as well as the SystemNetworkParameterRead response telegram could pass but not the DomainAddressSerialNumberRead Response telegram.

mjm987, 2020-05-04:
Yet I have another problem only on SAMD (the Raspi "Programs" fine Address as well as Appl.)

On SAMD programming the Adress works fine but during ETS-Programming the "Application" the device hangs after hunderts of "MemWrite" and "ProperyValueWrite after receiving following package:
L_Data.req 0.0.0->1.2.5 PropValueWrite OX=3 P=5 $02000000000000000000

(Which means Object Index = 3, P=5 --> PID Load State Control)

Btw: OX=4 P=5 just transmitted before works and sends a PropValueResp.

Any idea?

thelsing, 2020-05-04:
Can you post a full xml-log of the ets group monitor? The loadstate code is the same for all interface objects, so it should work the same for all off them. It seems that I have to search for my samd board to check if it still works for TP.

mjm987, 2020-05-05:
Here the "Program Application" Log of the samd until it breaks:

<CommunicationLog xmlns="http://knx.org/xml/telegrams/01">
  <Telegram Timestamp="2020-05-05T06:29:33.1028823Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA12050F63D70305100102000000000000000000" />
  <Telegram Timestamp="2020-05-05T06:29:30.0502896Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA12050F63D70305100102000000000000000000" />
  <Telegram Timestamp="2020-05-05T06:29:30.0009745Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA120500CE" />
  <Telegram Timestamp="2020-05-05T06:29:29.9800686Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA064FD60405100101" />
  <Telegram Timestamp="2020-05-05T06:29:29.9347862Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00DE" />
  <Telegram Timestamp="2020-05-05T06:29:29.8547593Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA12050F5FD70405100102000000000000000000" />
  <Telegram Timestamp="2020-05-05T06:29:29.8008387Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA120500CA" />
  <Telegram Timestamp="2020-05-05T06:29:29.7794095Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA0A4BD6040D100100FAABCE79" />
  <Telegram Timestamp="2020-05-05T06:29:29.7276841Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00DA" />
  <Telegram Timestamp="2020-05-05T06:29:29.6791584Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA12050A5BD7040D100100FAABCE79" />
  <Telegram Timestamp="2020-05-05T06:29:29.5992568Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA120500C6" />
  <Telegram Timestamp="2020-05-05T06:29:29.5770356Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA074644005800013501" />
  <Telegram Timestamp="2020-05-05T06:29:29.5326734Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00D6" />
  <Telegram Timestamp="2020-05-05T06:29:29.4710253Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA12050356040058" />
  <Telegram Timestamp="2020-05-05T06:29:29.4442883Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00D2" />
  <Telegram Timestamp="2020-05-05T06:29:29.3738653Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA1205075284005800013501" />
  <Telegram Timestamp="2020-05-05T06:29:29.3163946Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA120500C2" />
  <Telegram Timestamp="2020-05-05T06:29:29.2943919Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA0943D60107100100000058" />
  <Telegram Timestamp="2020-05-05T06:29:29.2441652Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00CE" />
  <Telegram Timestamp="2020-05-05T06:29:29.1698249Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA1205054FD501071001" />
  <Telegram Timestamp="2020-05-05T06:29:29.1359807Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA120500FE" />
  <Telegram Timestamp="2020-05-05T06:29:29.1137228Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA097E46005C000100010001" />
  <Telegram Timestamp="2020-05-05T06:29:29.0661973Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00CA" />
  <Telegram Timestamp="2020-05-05T06:29:28.9973539Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA1205034A06005C" />
  <Telegram Timestamp="2020-05-05T06:29:28.9692904Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00C6" />
  <Telegram Timestamp="2020-05-05T06:29:28.9008616Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA1205094686005C000100010001" />
  <Telegram Timestamp="2020-05-05T06:29:28.8577077Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA120500FA" />
  <Telegram Timestamp="2020-05-05T06:29:28.8356386Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA097BD6020710010000005C" />
  <Telegram Timestamp="2020-05-05T06:29:28.7656627Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00C2" />
  <Telegram Timestamp="2020-05-05T06:29:28.6916556Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA12050543D502071001" />
  <Telegram Timestamp="2020-05-05T06:29:28.6580779Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA120500F6" />
  <Telegram Timestamp="2020-05-05T06:29:28.6346626Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA0D764A004C00044F084B084B081300" />
  <Telegram Timestamp="2020-05-05T06:29:28.572486Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00FE" />
  <Telegram Timestamp="2020-05-05T06:29:28.5053536Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA1205037E0A004C" />
  <Telegram Timestamp="2020-05-05T06:29:28.4762549Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00FA" />
  <Telegram Timestamp="2020-05-05T06:29:28.3996595Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA12050D7A8A004C00044F084B084B081300" />
  <Telegram Timestamp="2020-05-05T06:29:28.3498222Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA120500F2" />
  <Telegram Timestamp="2020-05-05T06:29:28.3280548Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA0973D6030710010000004C" />
[... cut ...]
  <Telegram Timestamp="2020-05-05T06:29:16.8381295Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA03434027B0" />
  <Telegram Timestamp="2020-05-05T06:29:16.8026105Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900BC501205FFFA00C2" />
  <Telegram Timestamp="2020-05-05T06:29:16.7242182Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA1205014300" />
  <Telegram Timestamp="2020-05-05T06:29:16.7001466Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B060FFFA12050080" />
</CommunicationLog>

mjm987, 2020-05-05:
And if I restart the working Raspi and the failing SAMD I get following report of the saved config data:
Raspi:
`RESTORED 00 FA 00 00 00 00 00 00 00 03 12 04 00 00 00 01 00 00 01 00 00 00 00 00 00 00 00 00 FA 51 03 EE AF 01 00 00 00 44 00 08 01 00 00 00 58 00 04 01 00 00 00 5C 00 08 01 00 00 00 4C 00 0C 00 00 00 00 00 00 `
SAMD:
`RESTORED 00 FA 00 00 00 00 00 00 00 03 12 05 00 00 00 01 00 00 01 00 00 00 00 00 00 00 00 00 FA 51 03 EE AF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF`

mjm987, 2020-05-05:
Is there a 1:1 correspondance of the parameters in the .knxprod (XML) to the telegram values logged in the Groupmonitor (ObjectIndex? PropertyId? ...?)

thelsing, 2020-05-05:
No,
There are several interface objects. 
- Device-object for PA, serial number, version an others
- AddressTable (List of GAs)
- GroupObjectTable (List of GroupObjects (aka Kommunikationsobjekte)
- AssociationTable (Mapping between AddressTable and GroupObjectTable
- ApplicationProgramm (Parameters)
- IP-Parameters (for knx-ip)
- RF-Object (for knx-rp
Each Object has serveral properties such as ObjectType, LoadState, StartAddress (for MemoryWrite) etc.

mjm987, 2020-05-05:
Thx for explanation!
With that in mind I will try to understand the software ;-)

thelsing, 2020-05-05:
Nothing new in the log. Can you try to add some logging here https://github.com/thelsing/knx/blob/ed5dc739841d710ecba39d4cd3d4eb341f9f44a2/src/knx/bau_systemB.cpp#L212 
Then you can see if the PropertyWrite gets to the device and if the device answers.

mjm987, 2020-06-21:
solved in 
https://github.com/thelsing/knx/issues/58#issuecomment-645298241

closed 2020-06-21 by mjm987