thelsing/knx #58: Example sketch "KNX Demo" halts during programming of Application Program on Arduino Zero

Sonnengruesser, 2020-05-14:
I have set up a Zero (SAMD21) with Siemens BCU for testing and using the latest version of the library. When running the KNX Demo sketch, the programming of PA works. 
When programming the Application Program, the process does not finish. See monitor log attached. Where shall I debug to find the cause?

[ETS Monitor log.txt](https://github.com/thelsing/knx/files/4628523/ETS.Monitor.log.txt)
Note: some unrelated messages are deleted.

Sonnengruesser, 2020-05-14:
Latest working Commit is 4ef513410a7122d6892c3aa963e82d2abbbdeaf3

thelsing, 2020-05-14:
That seems to be the same problem as #53 . It would be very nice if you could debug the issue. My SAMD21 test setup is not working ATM.

Sonnengruesser, 2020-05-14:
You are right, sounds same. Will try to debug.

Sonnengruesser, 2020-05-16:
I'll continue this topic here because the linked Issue title is unrelated.

The issue is here:
https://github.com/thelsing/knx/blob/cc1fc7d5fb5050309937e99b9e0b346f2132292f/src/knx/bau_systemB.cpp#L214
tracking:
https://github.com/thelsing/knx/blob/cc1fc7d5fb5050309937e99b9e0b346f2132292f/src/knx/interface_object.cpp#L126
to this function, which does not return properly:
https://github.com/thelsing/knx/blob/cc1fc7d5fb5050309937e99b9e0b346f2132292f/src/knx/data_property.cpp#L26

Unfortunately if I add any print() in this function the sketch does compile but doesn't run on my Zero. @mjm987 any luck?

mjm987, 2020-05-16:
No unfortunately I had no luck yet, it is hard to debug without SWD Debug Probe.
May be it is a race condition between the main loop and ISRs? 
Does anybody know what is handled in ISRs? 
I guess all the knx stuff except the SPI- and UART- IO are handled in the main flow isn't it?
May be it happens when a Interrupt raises during the FlashStorage handling?
I don't think it is a stack, heap or buffer overflow during main flow itself.

Sonnengruesser, 2020-05-17:
I used a different sketch now and compared the loggings:
[Arduino Debug output.txt](https://github.com/thelsing/knx/files/4640111/Arduino.Debug.output.txt)

The interruption happens every time at the same point, so I don't thing any ISRs cause this. To me it looks like a memory allocation issue, since my project outputs "No available non volatile memory!"

However I can't debug this without debugger.

thelsing, 2020-05-26:
Maybe the problem is here https://github.com/thelsing/knx/blob/cc1fc7d5fb5050309937e99b9e0b346f2132292f/src/knx/table_object.cpp#L208

I just checked ESP8266 and it still works. I bricked my samd board so I can't check..

thelsing, 2020-05-26:
@Sonnengruesser  can you add which property is written to debug output?

thelsing, 2020-05-27:
initGroupObjects seems to hang. But I don't understand why.

mjm987, 2020-05-28:
> initGroupObjects seems to hang. But I don't understand why.

at least on my Zero (SAMD21)  it does not hang in initGroupObjects but somwhere (unknown) else

thelsing, 2020-06-06:
I tried to find the cause, but I wasn't successful. I ordered Segger Jlink Edu Mini now.

thelsing, 2020-06-08:
Today I got the debugger, but the knx power supply of my testbed died. ;(
Anyways I debugged the whole evening and still don't understand the error. The debugger goes straight to the hardfault handler at simple lines like htons(...).

I'll try the working version next.

mjm987, 2020-06-09:
may be be a alignment error?

mjm987, 2020-06-09:
I got openocd running using my nrf52 DK as debug interface and yes, it is a alignment error at the end of the code sequence below when accessing to _tableData[0] because _tableData is on a odd address!

```
bool GroupObjectTableObject::initGroupObjects()
{
println("initGroupObjects() - start");
    if (!_tableData) {
println("initGroupObjects() !tableData");
        return false;
    }
    freeGroupObjects();
    uint16_t goCount = ntohs(_tableData[0]);
```
@telsing Could you align _tableData to a even address?

mjm987, 2020-06-17:
Ok it works!
As I expected, the problem is a int16 or int32 access to a odd address.
This could be corrected by replacing htons(x) and htonl(x) in "bits.h" by:

```
#define getbyte(x,n) (*(((uint8_t*)&(x))+n))
#define htons(x)  ( (getbyte(x,0)<<8) | getbyte(x,1) ) 
#define htonl(x) ( (getbyte(x,0)<<24) | getbyte(x,1)<<16) | (getbyte(x,2)<<8) | getbyte(x,3) )
```
Of course the same might be done by using a union.
@thelsing: Could you please after verifying update bits.h accordingly?
Thx

thelsing, 2020-06-18:
@mjm987 I just committed this. I don't have time to test ATM.

Sonnengruesser, 2020-06-19:
Commit fixed the issue on the KNX Demo project for me. My other project still doesn't run and gives the "No available non volatile memory!" message. I think this is related, so I keep this issue open for the moment. trying to get rid of stuff on my sketch for further troubleshooting.

mjm987, 2020-06-19:
According the "No available non volatile memory": 
Did you try to increase EEPROM_EMULATION_SIZE form 1024 to eg. 2048 in FlashAsEEPROM.h ?

Sonnengruesser, 2020-06-21:
Changed my mind, my problem is probably unrelated. Closing this issue now.

closed 2020-06-21 by Sonnengruesser