thelsing/knx #61: Add support for STKNX and STM32F091

aibtech, 2020-06-03:
Would it be possible to add support for Bit interface (STKNX) transceiver and STM32F091?

leoujz, 2020-06-18:
Use STMdiuno, It works, I have tested it with stm32f103c8t6 and stm32f030c8t6, storing data to eeprom or internal flash, you need add a STM32Platform class similar with other already integrated platform, and do some work, not that difficult please have a try.

thelsing, 2020-06-18:
@leoujz can you share your work please?

leoujz, 2020-06-18:
I could send my code by mail, if anyone need it. But it needs sometime to tidy the code and structure, I think it's ready to create a pull request

aibtech, 2020-06-18:
> I could send my code by mail, if anyone need it. But it needs sometime to tidy the code and structure, I think it's not ready to create a pull request

That would be great. Can you please send it chaiphrom.nik (at) gmail.com?

leoujz, 2020-06-18:
Besides, 
(1) I add a knx enabled state check every 10 seconds for tpuart data link layer
(2) fix a bug when group object has enabled read_init when downloading by ETS
(3) fix a bug that invalid very large address reported by ETS, which is caused by the MemoryReference
(4) bug about some dpt, I forgot which one.

Any one need it can email me, it's not ready to create a pull request, because I add too many debugging codes and other modifications unrelated to the repository. Damn, no time to tidy them.

leoujz, 2020-06-18:
> > I could send my code by mail, if anyone need it. But it needs sometime to tidy the code and structure, I think it's not ready to create a pull request
> 
> That would be great. Can you please send it chaiphrom.nik (at) gmail.com?

already sent

thelsing, 2020-06-18:
Can you just put the repository in your github account? This way interested persons can just grab the bits they want.

thelsing, 2020-06-22:
There is a stm32 platfrom implementation here: https://github.com/etrinh/knx/blob/master/src/stm32_platform.cpp

thelsing, 2020-06-22:
I just merged the stm32 platform. Feel free to test it.

etrinh, 2020-06-22:
I have not finished to test my stm32 implementation. At least it compiles and I can communicate with my NCN5130. I am still struggling with my homemade hardware to communicate with the bus. I have not yet opportunity to program my device or check EEPROM access :(.
I will for sure make a pull request when I will have something working ;). At least, integration is very straightforward and is just couple of lines. It shouldn't be of any difficulty to anyone who have a working testbench ;)

leoujz, 2020-06-24:
> Besides,
> (1) I add a knx enabled state check every 10 seconds for tpuart data link layer
> (2) fix a bug when group object has enabled read_init when downloading by ETS
> (3) fix a bug that invalid very [...]

I already created a pull request to add the modifications to the latest master branch

etrinh, 2020-06-27:
Pull request #71 made to fix a bug in eeprom handling - forgotten variable initialization.

apiesse, 2021-02-10:
Would [this](https://github.com/mistay/stknx) be any helpful?
I'm thing of buying some stknx chips/eval, whit some indications I can try to integrate this library as a secondary datalink layer....

thelsing, 2021-02-10:
Its possible but not trivial to use this code. Mainly because there are no freertos libs available in my stack. So you need to add some abstraction for timers to platform.h/cpp. Queue-implementations are already available in the stack.

apiesse, 2021-02-12:
If I'm not wrogn, at least for the ESP32/ESP8266 we can use the idf-provided timers functions (same functions used by that library).
No idea about the arduino platform, but I imagine something similar is available also there. As well as on the STM32 platform which might used STM32 CubeMX.
What do you think about it?

If you or someone else is willing to assist in abstracting the datalink layer so that we can have both TPUART and STKNX I can try to integrate the STKNX library on esp32 platform (for starters).

caiservass, 2025-01-12:
Hi all,

I wanted to check in to see if there have been any recent updates regarding the compatibility of the STKNX chip with this library. Is there any progress, or are there any planned updates or forks addressing this?

Additionally, I’m looking to test the STKNX chip with a board like the Raspberry Pi Pico, but I’ve been unable to find any schematics or documentation on how to connect the chip to a dev board.

If anyone has experience with this setup or can provide a schematic or guidance on connecting the STKNX chip to the Pico (or any other board), it would be greatly appreciated.

Thanks in advance!