thelsing/knx #69: cant read (valid) parameter values

ajlajlajl, 2020-06-23:
Hi,
Ive tried to implement a module with ~15 channel and lots of parameters.
now it seems i cant find parameter values. what i get from knx.paramByte() function has nothing to whats i set using ETS software.

Ive 13 Group Objects right now, last ones number is 121. and around 50 parameters. and since this library seems to use only 512 byte for storage, i guss its too much?
Does implementing too many Group Objects cause this problem? Is there anyway i can expand the storage?

Thanks

thelsing, 2020-06-23:
If you run out of memory for paramters the programming via ets should fail and you should get an error on the serial console. Are you sure that you use the correct offsets? You need to count the bytes and not the parameters. If you can't figure it out, you can share the code and I can look at it.

leoujz, 2020-06-24:
> Hi,
> Ive tried to implement a module with ~15 channel and lots of parameters.
> now it seems i cant find parameter values. what i get from knx.paramByte() function has nothing to whats i set using ETS [...]

In my case, handreds of group objects and handreds of parameters seem fine, nearly 2K byte. if using EEPROM.h, simply EEPROM.begin(2048) can make it work, if your hardware supports.

About reading parameters, you can use paramByte() or paramInt() or paramData(), in condition that you know the length of each parameter. You can easily get the parameter length from the generated .xml by CreateKnxProd

ajlajlajl, 2020-06-28:
> If you run out of memory for paramters the programming via ets should fail and you should get an error on the serial console. 

I dont get any error in serial console, still values i get from paramByte() are not valid.

>Are you sure that you use the correct offsets? You need to count the bytes and not the parameters. If you can't figure it out, you can share the code and I can look at it.

yup Im pretty sure, checked several times. I'll show output below.

>In my case, handreds of group objects and handreds of parameters seem fine, nearly 2K byte. if using EEPROM.h, simply EEPROM.begin(2048) can make it work, if your hardware supports.

Well, I have several questions! in function getEepromBuffer() implemented in my platform(stm32) the only size i get as function input is always 512 as is specified in memory.cpp as constant:
https://github.com/thelsing/knx/blob/4677044f46c0e637fb669d93e62fd5c1f4cecb43/src/knx/memory.cpp#L16-L17
so it should always be 512 bytes right? how is yours 2k ???
btw, I'm using STM32F107 and Im storing data on its flash(last page)

So, heres my XML (generated with my own script, man it was hard):
https://gist.github.com/ajlajlajl/fbd2ce469b08798a27b8a6cb5ece8f61

I've added STM32 platform and a few other changes like LED & button pins. Im not gonna mention them here cuz i guss they r not important.

This is my ino file (I've simplified my program to this one file):
```C
#include "./knxStack/knx.h"

#define CHNO 12
#define LEDG PA5
#define PROGBTN PC13
#define DEBUGSERIALPINR PC11
#define DEBUGSERIALPINS PC10

HardwareSerial SerialDBG(DEBUGSERIALPINR, DEBUGSERIALPINS);

void printParameters();

void controlCallback(GroupObject& go);
GroupObject* kgoControl[CHNO];
GroupObject* kgoStatus[CHNO];

#define GroupObjectAdrs(i) (i * 2 + 2)
#define ParameterAdrs(i) (i * 4 + 2)

void setup()
{
	SerialDBG.begin(115200);
	ArduinoPlatform::SerialDebug = &SerialDBG;
	knx.readMemory();

	if (knx.configured()) {
		for (int i = 0; i < CHNO; i++) {
			kgoControl[i] = &(knx.getGroupObject(GroupObjectAdrs(i)));
			kgoControl[i]->dataPointType(Dpt(1, 1));
			kgoControl[i]->callback(controlCallback);
			kgoControl[i]->extra = i;

			kgoStatus[i] = &(knx.getGroupObject(GroupObjectAdrs(i) + 1));
			kgoStatus[i]->dataPointType(Dpt(1, 1));
		}
	}
	else {
		SerialDBG.println("Not Configured yet!");
	}
	pinMode(PROGBTN, INPUT);
	knx.buttonPin(PROGBTN);
	knx.start();
}

int ks;
unsigned long lastpressed2 = 0;
void loop()
{
	knx.loop();

	if (SerialDBG.available()) {
		byte b = SerialDBG.read();
		switch (b)
		{
		case 'R':
			printParameters();
			break;
		}
	}

}

void controlCallback(GroupObject& go) {
	ArduinoPlatform::SerialDebug->print("Set Channel ");
	ArduinoPlatform::SerialDebug->print(go.extra);
	ArduinoPlatform::SerialDebug->print(" to ");
	ArduinoPlatform::SerialDebug->print((bool)go.value());
	ArduinoPlatform::SerialDebug->println('.');
	kgoStatus[go.extra]->value((bool)go.value());
}

void printParameters() {
	if (knx.configured()) {
		SerialDBG.println("Parameters:");
		SerialDBG.print("Debug[");
		SerialDBG.print((int)0);
		SerialDBG.print("]: ");
		SerialDBG.println((int)knx.paramByte(0));
		SerialDBG.println("-- Channels --");
		for (int i = 0; i < CHNO; i++) {
			SerialDBG.print("Ch ");
			SerialDBG.print(i);
			SerialDBG.print(" : EN[");
			SerialDBG.print((int)(ParameterAdrs(i)));
			SerialDBG.print("] > ");
			SerialDBG.print(knx.paramByte(ParameterAdrs(i)));
			SerialDBG.print(" , NOC[");
			SerialDBG.print((int)(ParameterAdrs(i) + 1));
			SerialDBG.print("] > ");
			SerialDBG.print(knx.paramByte(ParameterAdrs(i) + 1));
			SerialDBG.print(" , FB[");
			SerialDBG.print((int)(ParameterAdrs(i) + 2));
			SerialDBG.print("] > ");
			SerialDBG.print(knx.paramByte(ParameterAdrs(i) + 2));
			SerialDBG.print(" , RS[");
			SerialDBG.print((int)(ParameterAdrs(i) + 3));
			SerialDBG.print("] > ");
			SerialDBG.print(knx.paramByte(ParameterAdrs(i) + 3));
			SerialDBG.println(" .");
		}
	}
	else {
		SerialDBG.println("Not Configured yet!");
	}
}
```

Group objects and everything else works fine but parameters.
This is output of printParameters() function:
`
Parameters:
Debug[0]: 0
-- Channels --
Ch 0 : EN[2] > 0 , NOC[3] > 32 , FB[4] > 197 , RS[5] > 178 .
Ch 1 : EN[6] > 0 , NOC[7] > 8 , FB[8] > 13 , RS[9] > 179 .
Ch 2 : EN[10] > 0 , NOC[11] > 8 , FB[12] > 13 , RS[13] > 179 .
Ch 3 : EN[14] > 0 , NOC[15] > 8 , FB[16] > 13 , RS[17] > 179 .
Ch 4 : EN[18] > 0 , NOC[19] > 8 , FB[20] > 13 , RS[21] > 179 .
Ch 5 : EN[22] > 0 , NOC[23] > 8 , FB[24] > 13 , RS[25] > 179 .
Ch 6 : EN[26] > 0 , NOC[27] > 8 , FB[28] > 0 , RS[29] > 0 .
Ch 7 : EN[30] > 0 , NOC[31] > 0 , FB[32] > 0 , RS[33] > 0 .
Ch 8 : EN[34] > 0 , NOC[35] > 0 , FB[36] > 0 , RS[37] > 0 .
Ch 9 : EN[38] > 0 , NOC[39] > 0 , FB[40] > 0 , RS[41] > 0 .
Ch 10 : EN[42] > 0 , NOC[43] > 0 , FB[44] > 13 , RS[45] > 179 .
Ch 11 : EN[46] > 0 , NOC[47] > 8 , FB[48] > 13 , RS[49] > 179 .
`
Ive included the offset in brackets, Ive checked them they are correct.
At least all EN values should be 1 (enable). and i have no 32, 8, 197, 178, ... values in my parameter types...

so, see if  can find whats wrong here
thanks a lot

ajlajlajl, 2020-06-30:
any idea?

thelsing, 2020-07-01:
I didn't find any obvious error. Maybe its an endianess problem or ETS changes the size of parameter?

You can try
printHex("", knx.paramData(0), 50) 
to print the raw hex data of the paramters. The stack does nothing with this block of data. So you need to test how this data changes when you change the parameters in ETS.

OutOfSync1, 2020-07-02:
Do you really need 65000 maximum entries in the address table and the association table? Perhaps this takes away a lot of memory. You can also try changing flashSize to 2048 or some other value instead of 512 in memory.cpp (line 16). I also had the problem of parameters being overwritten.

ajlajlajl, 2020-07-22:
I was busy these days so it took me a while to return to this again. Found the cause. It was my catalog XML generator. 
forgot to set size on LdCtrlRelSegment and LdCtrlWriteRelMem. So nothing was downloaded ...
seems alright now.

closed 2020-07-22 by ajlajlajl