thelsing/knx #73: STM32 EEPROM inefficiency

ajlajlajl, 2020-06-28:
Hi
I just saw STM32 has been added. and since Ive implemented one for myself and encountered this issue Ive got to mention it here too. Ive used STM32duino Arduino core and I guess this code is based on that core too.
In these lines:
https://github.com/thelsing/knx/blob/4677044f46c0e637fb669d93e62fd5c1f4cecb43/src/stm32_platform.cpp#L38-L39
EEPROM from STM32duino is used to store data in a loop. If we look in EEPROM source:
https://github.com/stm32duino/Arduino_Core_STM32/blob/d07c1db5b01d5d105f28ce369bd9658aa7785956/libraries/EEPROM/src/EEPROM.h#L56
Every write calls eeprom_write_byte() function here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/d07c1db5b01d5d105f28ce369bd9658aa7785956/libraries/SrcWrapper/src/stm32/stm32_eeprom.c#L164-L168
performs an eeprom_buffer_flush() on every call. and that means an erase on that sector of flash.

so it means for every byte we change we perform a complete sector erase-write. and beside the fact that is wastes time, considering we got only 10k write cycle on stm32 flash, its gonna stop working very soon i guess.

Im not sure how it can be fixed, since its a bad design on side of STM32duino.
Ive implemented my own flash write/read based on codes in stm32_eeprom for my device, but its not really portable and its duplication.

etrinh, 2020-06-28:
Yes I noticed too, but I wanted to stick with Arduino ecosystem. Which is by itself not the best choice for efficiency and not only for eeprom :(.

ajlajlajl, 2020-06-29:
Sure sticking with Arduino ecosystem is well and good, but, the problem is current code is wearing out STM32 way too fast. This way 10k write of STM32 will be used with 50-100 time download of ETS.
I guess best solution is modifying STM32Duino EEPROM library so that we can write without flushing. or something like that...

etrinh, 2020-06-29:
Can you make a pull request with your port?
Btw I made a pullrequest #71 without, no risk of damaging the flash as nothing was written :(
There is also condo4 port #45.

etrinh, 2020-06-29:
If we change:
	for (uint16_t i = 0; i < eepromSize; ++i) {
//		EEPROM.update(i, eepromPtr[i]);
              eeprom_buffered_write_byte(i, eepromPtr[i]);
    }
    eeprom_buffer_flush();

the loop only write to internal stm32duino buffer (which is a simple duplicate of eepromPtr... waste of memory...)
and 1 call to flush. I haven't digged into the flush process.
The best would be to copy paste bunch of code from stm32_eeprom.c or at least get the eeprom_buffer.
But as a one liner....

thelsing, 2020-06-29:
At the flash branch I started a different memory interface for flash. See https://github.com/thelsing/knx/blob/300b710740affe1fc1c24c244db4bf68b1c0b954/src/knx/platform.h#L62
I guess I'll have to finish it for SAMD21. Would this help for STM too?

etrinh, 2020-06-29:
It will require to go deeper in the hardware abstraction level and avoid simplicity of Arduino or some HAL libraries out there, with the risk to have bunches of ifdef depending of hardware in knx library (for example to get flash address or size).
But it is a very nice option, for more advanced port or platform support.
I think the best would be to keep a compatibility layer, with current interface for general/generic hardware support (the current STM32 should work whatever stm32 hardware you use) because it uses a very high level abstraction level which supports wide range of devices, but with poor performances, And the new interface to implement efficient hardware support, but with the risk to be restricted to a specific range of hardware, or with limited/unknown support for other.

commit f880114d (2020-06-30): Stm32 port (#71)

* Fix Stm32 Eeprom handler
Fix missing parenthesis
Fix some type consistencies

* Fix #73
Add ability to override config.h with define NO_KNX_CONFIG

* change tabs to space

* coding style
commit fda8e644 (2020-07-01): Add support for lambda for more platforms (#74)

* Fix Stm32 Eeprom handler
Fix missing parenthesis
Fix some type consistencies

* Fix #73
Add ability to override config.h with define NO_KNX_CONFIG

* change tabs to space

* coding style

* Allow lambda on platforms which support it, can be forced with HAS_FUNCTIONAL 0/1

* remove automatic detection of functional due to breakage with some platforms:
In file included from /home/travis/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/arm-none-eabi/include/c++/7.2.1/bits/char_traits.h:39:0,
                 from /home/travis/.arduino [... cut]
closed 2020-06-30 by thelsing