thelsing/knx #78: Wrong pointer arithmetic in TableObject

nanosonde, 2020-07-16:
Wrong variable is used here:
https://github.com/thelsing/knx/blob/96884f768e062dc3fed828133eff63d3a388d507/src/knx/table_object.cpp#L284

Instead of` propertyCount` it should be `propertiesSize`.
`propertyCount` is the array index, but we have to add `propertyCount * sizeof(Property*)` which is `propertiesSize`.

thelsing, 2020-07-18:
You're right. Fixed. Thanks.

etrinh, 2020-07-18:
After updating, I have a startup crash even with erased flash and increased flash memory (on an esp32 device); I haven't if I revert this fix.
I don't know the logic behind allProperties, but arithmetic on pointer shouldn't already take into account object size?

nanosonde, 2020-07-19:
@etrinh 
Can you try this instead of using pointer arithmetic at all?

```
    memcpy(&allProperties[0], properties, propertiesSize);
    memcpy(&allProperties[propertyCount], ownProperties, sizeof(ownProperties));
```

thelsing, 2020-07-19:
@etrinh you are right. Pointer arithmetic already handles this.
Foo[count] = *(Foo + count)

nanosonde, 2020-07-19:
Why not avoiding pointer arithmetic?
It is already an array, so let‘s just use the correct index and the address operator.

etrinh, 2020-07-19:
>     memcpy(&allProperties[0], properties, propertiesSize);
>     memcpy(&allProperties[propertyCount], ownProperties, sizeof(ownProperties));

Yes this works as well

thelsing, 2020-07-19:
Pointer arithmetic is a matter of taste. I used it here because memcpy  needs a pointer and an array is a pointer and I kind of like pointer arithmetic. Lets stick to "don't fix it if it ain't broken"

commit 1f918d69 (2020-07-18): fix #78
closed 2020-07-19 by thelsing