thelsing/knx #80: Possible bug in check for valid cemi frame

nanosonde, 2020-07-22:
While testing the coupler routing I got "invalid frame" messages in the console.

Inspecting this code:
https://github.com/thelsing/knx/blob/52d3866e41e1e9fd460f366bd9016daf82ec5e12/src/knx/cemi_frame.cpp#L369

If I remove the "+1" then everything works as expected. Why do we have the "+1" there?

nanosonde, 2020-07-22:
_data[0] is the cemi message code.
_data[1] is the additional info length which was 0 during testing
NPDU_LPDU_DIFF is 8

Looking at the cemi frame format, it seems that the "+1" is simply wrong.

nanosonde, 2020-07-22:
The NPDU_LPDU_DIFF start counting from 0 and includes the message code field.

thelsing, 2020-08-31:
You should be right. This is already included in the coupler pull request. Closing.

closed 2020-08-31 by thelsing