thelsing/knx #82: failed to build with nanoatmega328 platform.io

hhansen06, 2020-07-26:
I tried to build the knx-demo project with platform.io.
i changed the platform.ini to:

`[platformio]
; We have to keep libdeps dir out the project directory otherwise,
; library scanner seems to have issues so compilation fails
libdeps_dir = /tmp/libdeps
src_dir = .

;--- SAMD --------------------------------------------------
[env:nanoatmega328]
platform = atmelavr
board =  nanoatmega328
framework = arduino
; We consider that the this projects is opened within its project directory
; while working with VS Code.
lib_extra_dirs = ../../../

lib_deps =
  knx

build_flags =
  -DMEDIUM_TYPE=2
  -Wno-unknown-pragmas
;-----------------------------------------------------------`

Build failed with the following error:
'
Building in release mode
Compiling .pio\build\nanoatmega328\src\knx-demo.ino.cpp.o
Archiving .pio\build\nanoatmega328\libd3b\libEEPROM.a
Compiling .pio\build\nanoatmega328\lib16b\SPI\SPI.cpp.o
Compiling .pio\build\nanoatmega328\lib295\knxraumcontroller\arduino_platform.cpp.o
Compiling .pio\build\nanoatmega328\lib295\knxraumcontroller\esp32_platform.cpp.o
Compiling .pio\build\nanoatmega328\lib295\knxraumcontroller\esp_platform.cpp.o
Compiling .pio\build\nanoatmega328\lib295\knxraumcontroller\knx\address_table_object.cpp.o
In file included from C:\repo\knxraumcontroller\src/knx_facade.h:3:0,
                 from C:\repo\knxraumcontroller\src/knx.h:86,
                 from C:/repo/knxraumcontroller/examples/knx-demo/knx-demo.ino:1:
C:\repo\knxraumcontroller\src/knx/bits.h:3:19: fatal error: cstddef: No such file or directory
compilation terminated.
Compiling .pio\build\nanoatmega328\lib295\knxraumcontroller\knx\aes.c.o
Compiling .pio\build\nanoatmega328\lib295\knxraumcontroller\knx\apdu.cpp.o
*** [.pio\build\nanoatmega328\src\knx-demo.ino.cpp.o] Error 1
In file included from C:\repo\knxraumcontroller\src\arduino_platform.cpp:2:0:
C:\repo\knxraumcontroller\src/knx/bits.h:3:19: fatal error: cstddef: No such file or directory
compilation terminated.
C:\repo\knxraumcontroller\src\knx\address_table_object.cpp:1:19: fatal error: cstring: No such file or directory
compilation terminated.
In file included from C:\repo\knxraumcontroller\src\knx\apdu.cpp:3:0:
C:\repo\knxraumcontroller\src\knx\bits.h:3:19: fatal error: cstddef: No such file or directory
compilation terminated.
*** [.pio\build\nanoatmega328\lib295\knxraumcontroller\knx\address_table_object.cpp.o] Error 1
*** [.pio\build\nanoatmega328\lib295\knxraumcontroller\arduino_platform.cpp.o] Error 1
*** [.pio\build\nanoatmega328\lib295\knxraumcontroller\knx\apdu.cpp.o] Error 1
========================================================================================= [FAILED] Took 4.71 seconds =========================================================================================
The terminal process "C:\Users\henrik\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1.
'

thelsing, 2020-08-06:
You need a platform that has a working c++ compiler. Some default c++ includes seem to be missing from the atmelavr platform. Unfortunately there is nothing that I can do about this.

thelsing, 2020-08-06:
You can try to replace the #inlcude <cstddef> with #include <stddef.h>

hhansen06, 2020-09-25:
didn't work, i switched to samd21 chip.

closed 2020-09-25 by hhansen06