thelsing/knx #83: Multiple device in programming mode when my device is alone

ajlajlajl, 2020-08-26:
Hi
I've got a strange problem with my device. When ETS is looking for device address (when downloading address or in diagnostics's programming mode; if theres another KNX device on TP bus together with my device, theres no problem. But if theres only my device on TP Bus, ETS detects 2 device and fails (on downloading) saying theres more than one device in programming mode, and in Diagnostics also shows two device (*2) with same address.

This is log from Group Monitor for one request:
```
<CommunicationLog xmlns="http://knx.org/xml/telegrams/01">
  <Telegram Timestamp="2020-08-26T13:33:10.6411059Z" Service="L_Data.con" FrameFormat="CommonEmi" RawData="2E00B0E000010000010100" />
  <Telegram Timestamp="2020-08-26T13:33:10.6893658Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900B0D011020000010140" />
  <Telegram Timestamp="2020-08-26T13:33:10.7272676Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900B0D011020000010140" />
  <Telegram Timestamp="2020-08-26T13:33:10.7811246Z" Service="L_Data.ind" FrameFormat="CommonEmi" RawData="2900B0D011020000010140" />
</CommunicationLog>
```
As you see each request receives 3 response...

This is debug output of my device (per request):
```
dataBroadcastIndication
dataBroadcastRequest
dataBroadcastIndication
dataBroadcastRequest
dataBroadcastConfirm
dataBroadcastIndication
dataBroadcastRequest
dataBroadcastConfirm
dataBroadcastIndication
dataBroadcastRequest
expected L_DATA_CON not received
dataBroadcastConfirm
got U_CONFIGURE_IND: 0x11
got UNEXPECTED: 0x2
got L_ACKN_IND
got L_ACKN_IND
got UNEXPECTED: 0xE1
got U_CONFIGURE_IND: 0x1
got L_ACKN_IND
got UNEXPECTED: 0xDC
got U_STATE_IND: 0x17
got unexpected L_DATA_CON
```

any idea whats going on here??

Thanks

thelsing, 2020-08-31:
I wouldn't exclude a bug in the tpuart-code. I will check if I can reproduce this. How do you connect ETS with the bus?

ajlajlajl, 2020-09-01:
> 
> 
> I wouldn't exclude a bug in the tpuart-code. I will check if I can reproduce this. How do you connect ETS with the bus?
Thanks.
ETS is connected via cable network using a IP Router. My device is connected to KNX TP network using NCN5130.
I'm not sure if this is from software, but I couldn't find any problem in my hardware so i guess its from software. 
The whole thing is strange. Why would it fails only when its alone! and only on reading address. everything else works fine (downloading only application or controlling a group address ...).

ajlajlajl, 2020-09-16:
Hi,
I ran several more tests today to find more about this, this is my results. see if u can find out anything new?! 
I used 3 device:

1. my DIY device custom designed and using a NCN5130
2. a STM32F407 Discovery board and a wallmount back of another device thats using a Elmos chip.
3. a HDL M/R12 Actuator

obviously, device 1 & 2 is using your knx library.

device 1 & 2 behavior is similar.

- when they are not programmed( their address is 0.0.0 ). everything ok. they response once to "PhysAddrRead" ( in ETS's Diagnostics's Programming mode). alone or not.
- when they are alone in TP network and they have an address. they response 3 time to "PhysAddrRead". 
- but, if i connect device 3, they response just once then!. whether that device 3 is in programming mode or not. just its presence fixes this problem.
- also, interesting thing is, if i one of my not programmed devices too, it fixes the problem. 

Example to clarify:
- I have device 1 only. not programmed. it responses once (to "PhysAddrRead"). OK
- I programmed device 1 (set address). it responses thrice. FAIL
- I add deice 2 to network. not programmed (adress 0.0.0). it fixes the problem. now device 1 responses once. whether device 2 is in program mode or not. device 2 response once too.
- I programmed device 2 (set address). now both devices FAIL. each device responce thrice! 
- now if I add device 3 (programmed). the moment it connects, everything get alright again. anyone of device when put on programming mode, response just once.

well I can't make head or tail of this!. why its alright when its not programmed! and why it got fixed by mere presence of another device.
and since I've tried on two different hardware, I'm sure its not a hardware problem.

help me 😭

nrobadey, 2020-09-21:
I have the same problems, also with a DIY device with an NCN5130 (coupled to a SAMD21G in the MKRzero configuration).
The solution I found is either to reflash the firmware to reset the device address, or to do an ETS "full download" with the same device address the device already has stored.
If I try to change the device address while it is already programmed, I get the "multiple devices in programming mode" message, with the same address.

My device also responds multiple times to requests (actually 4 times); weird, but seems not to cause problems...

3mehdi, 2020-09-23:
I have the same problems, after set address  it responses thrice. FAIL . 
but I add device 2 to network. not programmed  ,   fixes the problem. now device 1 responses once. 
after after set address in device  2 ,  it responses thrice. and full download fail.
@thelsing  please fix it , Without solving this problem, this library is useless.

> Hi,
> I ran several more tests today to find more about this, this is my results. see if u can find out anything new?!
> I used 3 device:
>
> 1. my DIY device custom designed and using a NCN5130
> 2. a [...]

I have the same problems, after set address  it responses thrice. FAIL . 
but I add device 2 to network. not programmed  ,   fixes the problem. now device 1 responses once. 
after after set address in device  2 ,  it responses thrice. and full download fail.
@thelsing  please fix it , Without solving this problem, this library is useless.

nanosonde, 2020-09-23:
> @thelsing please fix it , Without solving this problem, this library is useless.

@3mehdi It is an open-source project. Everybody is working on in during spare time. You are welcome to help out by analyzing the problem and open a proper pull request with a proposed solution. Just stating that the library is useless does not help.

3mehdi, 2020-09-23:
### 

> @3mehdi It is an open-source project. Everybody is working on in during spare time. You are welcome to help out by analyzing the problem and open a proper pull request with a proposed solution. Just [...]

@thelsing  has spent a lot of time  And he wrote this really great library,  but something so simple should not make it useless
I'm sure the solution is easy for him , But it is very difficult for us, because we have to understand the whole library.
Many thanks to @thelsing  for taking the time to write this library
  And I know It is an open-source project. Everybody is working on in during spare time. but by fixing this bug, this library will really be useful.  
issue is for problem solving and troubleshooting, Asking questions should not upset anyone!

thelsing, 2020-09-29:
Can you try if the problem remains, if you change the default PA  here: https://github.com/thelsing/knx/blob/241e04330ded2a01449a106ab61caa970f56bad3/src/knx/device_object.h#L42 to 0xffff?

ajlajlajl, 2020-10-03:
> 
> 
> Can you try if the problem remains, if you change the default PA here:
> 
> https://github.com/thelsing/knx/blob/241e04330ded2a01449a106ab61caa970f56bad3/src/knx/device_object.h#L42
> to 0xffff?

Hi
Its alright if address is 0. when it changes using ETS or, changed manually as you instructed, it returns 3 reply.

this is when address is 0
[image]

this is when address is not 0 (0xffff here)
[image]

Sonnengruesser, 2020-10-08:
Just a guess without knowing the specifics of the knx protocol:
The debug output "expected L_DATA_CON not received" shows that the confirmation message from the other bus units is not received in your device. Then your device repeats the message usually 3 times - without receiving the confirmation.
This might be caused by the main loop blocking the KNX stack for too long causing a timeout. 
@ajlajlajl Did you try using the example sketches or just with you custom software?

nrobadey, 2020-10-14:
I'm wondering if this could be related to the nature of the transciever. The NCN5130 may have different features than a TP-UART.  AFAIK the NCN5130 is aware of the address of the device and can handle filtering and repetition of messages.
@thelsing is there a hardware reference that goes with the library? Was it designed with a specific kind of chip in mind?

ajlajlajl, 2020-10-18:
> I'm wondering if this could be related to the nature of the transciever. The NCN5130 may have different features than a TP-UART. AFAIK the NCN5130 is aware of the address of the device and can [...]

I dont think thats related to my chip. I've tested my program using a testing boeard that uses elmos E981.03. it has the same problem.

> 
> 
> Just a guess without knowing the specifics of the knx protocol:
> The debug output "expected L_DATA_CON not received" shows that the confirmation message from the other bus units is not received in your device. Then your device repeats the message usually 3 times - without receiving the confirmation.
> This might be caused by the main loop blocking the KNX stack for too long causing a timeout.
> @ajlajlajl Did you try using the example sketches or just with your custom software?

I've got to test it to be sure. but i don't think its the case. cuz my program is same whether module has address or not.

ajlajlajl, 2020-11-05:
@thelsing any idea how to solve this?

Sonnengruesser, 2020-11-05:
I'm not thelsing but why don't you try the test sketch first before blaming somebody else made a fault?

nikwest, 2020-11-07:
Actually I’m running into exactly the same problems. Also with an NCN5130 (evaluation board). In order to get it to work at all I had to change the default PA from 0 to something else (because of knxd in my case). But once programmed I’m getting these error messages too from ETS.
I also have the feeling it is some timing issue, but I’m not yet familiar enough with KNX and also this stack to figure it out. 
Wouldn’t it be in general better to use a task to read and write to the uart? At least on devices which support it (esp,  stm32 ...). I would give it a try if somebody confirms that this might make things easier.

ajlajlajl, 2020-11-12:
> 
> 
> I'm not thelsing but why don't you try the test sketch first before blaming somebody else made a mistake?

Why so toxic man, we are trying to solve a problem here. no blaming anybody.
Ive tested with one of examples and the problem still exists. I guess its something related to TP protocol.
--

Ive uncommented those 'printHex's in 'arduino_platform.cpp' so that we have serial data in debug too.
maybe these can help to solve this problem:

this is the output when this problem occurs (address is programmed, device is in programming mode and ETS is sending 'PhysAddrRead' )

```
p>B0 
p>01 
p>D1 
<p10 
p>01 
p>9E 
dataBroadcastIndication
dataBroadcastRequest
<p80 B0 
<p81 11 
<p82 0C 
<p83 00 
<p84 00 
<p85 E1 
<p86 01 
<p87 40 
<p48 F2 
p>90 
p>01 
p>D1 
<p10 
p>01 
p>BE 
dataBroadcastIndication
dataBroadcastRequest
p>B0 
p>11 
p>0C 
p>E1 
p>01 
p>40 
p>F2 
p>8B 
dataBroadcastConfirm
<p80 B0 
<p81 11 
<p82 0C 
<p83 00 
<p84 00 
<p85 E1 
<p86 01 
<p87 40 
<p48 F2 
p>90 
p>01 
p>D1 
<p10 
p>01 
p>BE 
dataBroadcastIndication
dataBroadcastRequest
p>B0 
p>11 
p>0C 
p>E1 
p>01 
p>40 
p>F2 
p>8B 
dataBroadcastConfirm
<p80 B0 
<p81 11 
<p82 0C 
<p83 00 
<p84 00 
<p85 E1 
<p86 01 
<p87 40 
<p48 F2 
p>90 
p>01 
p>D1 
<p10 
p>01 
p>BE 
dataBroadcastIndication
dataBroadcastRequest
p>B0 
p>11 
p>0C 
p>E1 
p>01 
p>40 
p>F2 
p>90 
expected L_DATA_CON not received
dataBroadcastConfirm
<p80 B0 
<p81 11 
<p82 0C 
<p83 00 
<p84 00 
<p85 E1 
<p86 01 
<p87 40 
<p48 F2 
p>11 
got U_CONFIGURE_IND: 0x11
p>0C 
got L_ACKN_IND
got L_ACKN_IND
got L_ACKN_IND
p>E1 
got UNEXPECTED: 0xE1
p>01 
got U_CONFIGURE_IND: 0x1
p>40 
got L_ACKN_IND
p>D2 
got UNEXPECTED: 0xD2
p>17 
got U_STATE_IND: 0x17
p>8B 
got unexpected L_DATA_CON
L_DATA_CON not received within expected time
```

This is output for one request. Hope it helps. Ive no idea what these means 😑
Thanks.

ajlajlajl, 2020-11-12:
btw, for reference, this is output from the same device, in same exact situation, but with another KNX device connected to TP too.

```
p>B0 
p>01 
p>D1 
<p10 
p>01 
p>9E 
dataBroadcastIndication
dataBroadcastRequest
<p80 B0 
<p81 11 
<p82 0C 
<p83 00 
<p84 00 
<p85 E1 
<p86 01 
<p87 40 
<p48 F2 
p>B0 
p>11 
p>0C 
p>E1 
p>01 
p>40 
p>F2 
p>8B 
dataBroadcastConfirm
```

thelsing, 2020-11-12:
What is the actual test setup? USB-interface <-> 1 DIY Device? Some versions of knxd cause this too: https://github.com/knxd/knxd/issues/275

ajlajlajl, 2020-11-14:
> 
> 
> What is the actual test setup? USB-interface <-> 1 DIY Device? Some versions of knxd cause this too: [knxd/knxd#275](https://github.com/knxd/knxd/issues/275)

I think thats a different issue. he couldnt program PA when he was using knxd.
I cant program PA when my device is alone.

this is my setup

ETS <-- HDL IP/TP Gateway -->My device
and i get "multiple devices in programming mode" here, If my device PA is not 0/0. when its 0/0 it works.
also if I add another device on TP line, it works whether its PA is 0/0 or not.

ajlajlajl, 2020-11-15:
OK new discoveries.
I was trying to find whats different when address is 0 or non-0. I found its Ackn byte here:
https://github.com/thelsing/knx/blob/bb818d9877ab32bf6e56b1127a466e08ba7d8d79/src/knx/tpuart_data_link_layer.cpp#L269-L283
If this byte (U_Ackn.req) is send as 0x11, thins go as it should be.
but (in response to ETS's PhysAddrRead ) if we send 0x10 ( bit 'addressed' set to 0), It seems NCN5130 send that request 3 more times, here:
```
p> B0 
p> 01 
p> D1 
<p 10                                  // <-- this byte should be 0x11
p> 01 
p> 9E 
dataBroadcastIndication
p> 90 
p> 01 
p> D1 
<p 10 
p> 01 
p> BE 
dataBroadcastIndication
p> 90 
p> 01 
p> D1 
<p 10 
p> 01 
p> BE 
dataBroadcastIndication
p> 90 
p> 01 
p> D1 
<p 10 
p> 01 
p> BE 
dataBroadcastIndication
```

so I think that's source of our problem, right?
then the question is why Ackn is 0x10? 
my guess: when PA is 0, isAckRequired() always returns true, but when device has a non-zero PA, and this request has no destination address as its a broadcast kinda??! request, isAckRequired() fails to recognize this request as a request addressed to this device. and despite that, device responses to it every time. and that's why we receive 3 response each time (last one fails i guess cuz collision or something as seen in log).
and thus, why isAckRequired() returns false on this request?

(well i really tried to find the solution but after reading a few references and datasheets, my only achievement is a headache. I still cant undrestand this KNX protocol ...)

a quick fix is if I set this:
https://github.com/thelsing/knx/blob/bb818d9877ab32bf6e56b1127a466e08ba7d8d79/src/knx/tpuart_data_link_layer.cpp#L269
manually to 0x11 on this line. tested and worked. but, well, that's not the answer. something in isAckRequired() should be fixed. (PLS 😭)

nikwest, 2020-11-15:
I think that the reason might be that the actual PA isn't communicated to the NCN5130 using U_SET_ADDRESS_REQ? At least I couldn't find it in code. That might leave the address in the NCN5130 at 0, which it will auto acknowledge? 
Just assumptions from my side, I have not verified them as I am still trying to get familiar with the KNX spec and the actual code.

nanosonde, 2020-11-15:
@nikwest 
According to the datasheet the NCN5130 does NOT activate the auto-Ack if U_SET_ADDRESS_REQ was not send. So ot should be ok. The KONNEKTING stack does not use U_SET_ADDRESS_REQ either.

I have just verified that I did not break things with the coupler pull request. The logic is still the same.

Broadcasts are send as group communication mode with a destination address of 0 according to the docs.
AFAIK 0 as destination address is forbidden in all other comm modes.

ajlajlajl, 2020-11-16:
> I think that the reason might be that the actual PA isn't communicated to the NCN5130 using U_SET_ADDRESS_REQ? At least I couldn't find it in code. That might leave the address in the NCN5130 at 0, [...]
Actually I've activated that. I was reading NCN5130 datasheet and saw U_SET_ADDRESS_REQ and thought it would fix my problem. I added it to code, and it activated. but AFAIK nothing changed.

thelsing, 2020-11-16:
Maybe the problem is really only the ACK? Maybe devices should ACK broadcasts? The stack doesn't seem to do this. If you have an other (offical) device on the line, it should  do the ACK. But this still doesn't  explain why it works with only two DIY-devices. Who sends the ACK then?
I have to read the spec on ACKs and Broadcast.

Are ACKs visible with Net&Node?

thelsing, 2020-11-16:
3/2/2 p. 38:
The remote Data Link Layer shall check whether it is addressed or not. The remote Data Link Layer is addressed in any of the following cases.
1. The Destination Address is an Individual Address
1.1 that corresponds to the own Individual Address of the remote Data Link Layer, or
1.2 that is topologically located on the other side of a Coupler, or
1.3 that is one of the Tunnelling Individual Addresses of a KNXnet/IP Tunnelling Server
2. The Destination Address is
• a Standard Mode Group Address or
• an LTE-HEE Group Address,
and the remote Data Link Layer supports the non-selective Data Link Layer acknowledge.
3. The Destination Address
3.1 is a Standard Mode Group Address that is contained in the Group Address Table of the receiver or
3.2 is an LTE-HEE Group Address addressing an Area that is assigned to the remote Data Link Layer.
4. The Destination Address is the broadcast address.
If the remote Data Link Layer is addressed then it shall confirm the Frame with an acknowledge-Frame.

So broadcast destination (Dest PA = 0.0.0) should be acked.

thelsing, 2020-11-16:
@ajlajlajl  I just committed 45158bb (and build fix). Can you check if this solves the problem?

thelsing, 2020-11-16:
Probably not. It seems I need to build a testbed for this.

ajlajlajl, 2020-11-17:
well it solved my problem.
https://github.com/thelsing/knx/blob/4f1b7502f644bdde2a662fefd002104d71c35b3a/src/knx/bau07B0.cpp#L142
"address" is always 0 on "PhysAddrRead" request

nanosonde, 2020-11-17:
@thelsing 

I think that this is wrong:
https://github.com/thelsing/knx/blob/4f1b7502f644bdde2a662fefd002104d71c35b3a/src/knx/tpuart_data_link_layer.cpp#L275

At this point we are expecting that buffer[] contains an L_DATA Extended frame. Which means that the address type is in extended control field (CTRLE) which would be buffer[1] and not buffer[6]!
Just before this line the conversion to extended frame took place:
https://github.com/thelsing/knx/blob/4f1b7502f644bdde2a662fefd002104d71c35b3a/src/knx/tpuart_data_link_layer.cpp#L259-L265

That is the reason why we see a broadcast with (destinaton address == 0) and address type of "individual address" which is wrong! Broadcasts are sent with address type of "group address" and  (destinaton address == 0).

nanosonde, 2020-11-17:
We should put an error message here instead:
https://github.com/thelsing/knx/blob/4f1b7502f644bdde2a662fefd002104d71c35b3a/src/knx/bau07B0.cpp#L142

Adress type "individual address" and `destination address == 0` is forbidden.

nanosonde, 2020-11-17:
With this bug, a DIY device with this stack would never currently ACK a group address.
If you have other (real) KNX devices on the same line which are using the same group address as our DIY device, the other device will take over the ACK for the group address.

Broadcasts are also never ACK'ed by our DIY device in this case. Other devices on the same line which receive the broadcast will handle this and send an ACK.

thelsing, 2020-11-17:
So I reverted the the last change in an new attempt to fix this issue: d962da4
@ajlajlajl Can you try if this fixes the issue? Maybe this is also the problem for #89.

thelsing, 2020-11-17:
Merging the pull request closed this issue. (Probably because of the message)

nrobadey, 2020-11-17:
Problem fixed for me (NCN5130 with SAMD21).

ajlajlajl, 2020-11-18:
Fixed for me too. Thanks. (NCN5130 & STM32)

nanosonde, 2020-11-18:
Can we close this issue?

PR #104 (2020-11-17): Add warning message in TP-UART data link layer
~~This is supposed to fix #83~~

Only add warning message.
closed 2020-11-18 by thelsing