thelsing/knx #85: TPUART-2 and samd21 on TP wiring ?

zelogik, 2020-08-31:
Hi,

My question should be stupid, but I'm unable to make pairing in ETS. (even when changing knx.buttonPin() value)

I have an GY-SAMD21 (a clone from the sparkfun mini) connected to a siemens 5WG1117-2AB12 on the tx,rx line (even crossed the rx/tx line without any luck). I have an old config with that 5WG1117-2AB12 and a arduino micro pro that work well but is not configurable... anyway...

And when i see demo/example code everything have IP connectivity, nothing with TP, where can I change the pin for TX,RX, or view the default value, nothing in knx_facade.h?

So much critics, but your software createKnxProd and that knx stack look a perfect combo for an clean ETS integration, for DIY knx device :+1: 

Regards

etrinh, 2020-09-01:
Don't know if it will help, but I had similar issue, it turned out that my individual address was still the default (0) which is not considered as an authorized address ([so](https://support.knx.org/hc/fr/articles/115003185789-Adresse-individuelle) link in french), ETS didn't recognize my device in programming mode. I changed it  with `knx.bau().deviceObject().induvidualAddress(1);` and it worked.

zelogik, 2020-09-01:
Hi,

Thanks for the advice, i will test it this evening. As french too, I'm just complaining because this project seem just wonderful, just lack some simple example/wiring/how-to/rtfm etc...

@etrinh: I have checked your github accound, your project teleinfoKNX seem like good too, have you more description ?

Regards

etrinh, 2020-09-01:
About my repo, if you want, you can directly interact with it.

thelsing, 2020-09-01:
Just a dumb question: Do you have the Gnd of Samd and the Siemens BCU connected?

zelogik, 2020-09-01:
Gnd bcu <-> gnd samd
5v bcu <-> vin samd
Rx bcu <-> TX samd ( and inverted it )
TX bcu <-> rx samd ( same)

So that config must work for you, i will try etrinh advice in 1h

zelogik, 2020-09-01:
So, the problem seem to be a pin definition.

A simple blink program work with the buildin_led, but when I include the knx.h lib that same buildin_led blinking fast but very little light.

etrinh, 2020-09-02:
the blinking led looks like a fatalError. I also get similar issue in my PR #77 :p.
By default the library tries to access 8kB of memory but it seems you have only 1. You can add in you project the define: KNX_FLASH_SIZE=EEPROM_EMULATION_SIZE or KNX_FLASH_SIZE=1024 or lesser number.

zelogik, 2020-09-02:
So the KNX_FLASH_SIZE was incorrect for my board, 2048 does the trick, now the BUILDIN_LED is LOW as wanted, but the samd21 seem to freeze just after the first knx.readmemory. ie: a simple digitalWrite before work but not after....

Thanks @etrinh for your effort.

zelogik, 2020-09-02:
Hi,

Yaypee! got some debug log from the serialUSB, but what mean:

`
readMemory
RESTORED FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 01 00 D0 91 01 00 00 00 00 00 E8 03 00 00 00 00 00 00 B8 07 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1F 1E 03 01 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF 00 00 00 D0 91 01 00 00 00 00 00 E8 03 00 00 00 00 00 00 A8 07 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 01 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF 00 00 00 00 14 00 42 00 08 00 42 00 0C 00 42 00 10 00 42 00 18 00 42 00 1C 00 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
saved memory doesn't match manufacturerId, version or hardwaretype
manufacturerId: FFFF FA
version: FFFF 3
hardwareType: FF FF FF FF FF FF 
 00 00 00 00 00 00 
ERROR, TPUART not responding
got UNEXPECTED: 0xE3
got L_ACKN_IND
got L_ACKN_IND
Timeout during RX_L_DATA
dataBroadcastIndication
dataBroadcastIndication
dataBroadcastIndication
dataBroadcastIndication
dataBroadcastIndication
dataBroadcastIndication
progmode on
dataBroadcastIndication
dataBroadcastRequest
L_DATA_CON not received within expected time
progmode off
progmode on
progmode off
dataBroadcastIndication
progmode on
dataBroadcastIndication
dataBroadcastRequest
L_DATA_CON not received within expected time
dataBroadcastIndication
dataBroadcastRequest
L_DATA_CON not received within expected time
dataBroadcastIndication
dataBroadcastRequest
L_DATA_CON not received within expected time
dataBroadcastIndication
dataBroadcastRequest
L_DATA_CON not received within expected time
dataBroadcastIndication
dataBroadcastRequest
L_DATA_CON not received within expected time
Timeout during RX_L_DATA
Timeout during RX_L_DATA
got L_ACKN_IND
`

Wiring Tx/Rx look good no ? but why L_DATA_CON not received within expected time and Timeout during RX_L_DATA

etrinh, 2020-09-03:
the " ERROR, TPUART not responding" looks not good from your RX/TX, it seems that the tpuart device is not inited properly...
Can you check what happen at your side inside the `TpUartDataLinkLayer::resetChip()` code? Maybe the timeout is too small, or there is an issue with your uart. (Again I had the same problem until I discovered I mixed my TX and RX line...)

thelsing, 2020-09-04:
Can you reproduce this? If yes can you add a xml export from the group monitor.

zelogik, 2020-09-04:
Yes I can reproduce this easily.

Step: Put the device in prog mode, before or after start the full upload.
The device address is 1.1.16
Do you need the same with some filtering?

[
[samd21-program-half_failed.xml.zip](https://github.com/thelsing/knx/files/5176431/samd21-program-half_failed.xml.zip)

](url)

Space-Farmer, 2020-09-08:
The bug with the individual Address is also valid with an ON-Semi HW. (micro BCU)

Solved with:     
  // start the framework.
      if (knx.induvidualAddress() == 0)
          knx.bau().deviceObject().induvidualAddress(1); //65535
    knx.start();

(this line costs me 4 month :-( )

With the Siemens BCU you have to realy take care about the voltage levels, because it the BCU has a low impedence load.

Regarding error messages, please find attached my programming of Group addresses (micro BCU and Ardu Zero clone)
[prog_GAs.txt](https://github.com/thelsing/knx/files/5191128/prog_GAs.txt)

nanosonde, 2020-11-18:
@Space-Farmer @zelogik 
Could you please re-test if the current master branch fixes your issues?

zelogik, 2020-11-24:
@nanosonde, @thelsing 
I have tested with the same result on my side: (last knx branch)
at around 60% on ETS i got error (already posted the log file of ets), and on the serialUSB debug log i have:

> progmode on
dataConnectedIndication
dataConnectedRequest
dataConnectedIndication
dataConnectedRequest
dataConnectedIndication
Basic restart requested
save saveRestores 3
-10
.
-26
.
-68
.
save tableobjs 4
-68
.
-75
.
-82
.
-89
.
SAVED 00 FA 00 00 00 00 00 00 00 03 11 18 00 01 00 00 01 60 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 00 01 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 01 01 00 D0 98 01 00 00 00 00 00 E8 03 00 00 00 00 00 00 B8 07 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 1F 1E 03 01 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF 00 00 00 D0 98 01 00 00 00 00 00 E8 03 00 00 00 00 00 00 A8 07 00 20 11 18 11 FF 60 C6 0E 8B B0 11 18 11 FF 6B 47 D6 00 0B 10 01 00 FA 01 02 03 04 B6 8B B0 11 FF 11 18 50 C6 3E B0 11 FF 11 18 51 4B 80 32 B0 11 FF 11 18 50 81 79 11 18 55 47 D5 00 0B 10 01 75 B0 35 00 00 00 2D 00 00 00 81 11 82 18 83 11 84 FF 85 6B 86 47 87 D6 88 00 89 0B 8A 10 8B 01 8C 00 8D FA 8E 01 8F 02 90 03 91 04 52 B6 84 FF 85 63 86 43 87 40 88 07 89 B0 4A 7F 81 11 82 18 83 11 84 FF 85 60 86 C6 47 0E 24 00 00 00 24 00 00 00 00 01 03 01 FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF 00 00 00 00 14 00 42 00 08 00 42 00 0C 00 42 00 10 00 42 00 18 00 42 00 1C 00 42 85 27 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
restart
dataConnectedIndication
dataConnectedRequest
dataConnectedIndication
dataConnectedRequest
...
dataConnectedIndication
dataConnectedRequest
... 

As the Samd21 is limited on flash, I have tested between 256 to 1024 with the same result.

Any advice?

thelsing, 2020-12-05:
Can you check if this comment helps: https://github.com/thelsing/knx/issues/110#issuecomment-735425329

zelogik, 2020-12-05:
Yes! thanks @thelsing for pointing me the comment, and thank @rueckix. 

Close the issue.

closed 2020-12-05 by zelogik