thelsing/knx #86: [ESP8266, build] Compiling latest ESP8266Wifi fails

nanosonde, 2020-09-04:
Compiling knx-demo for ESP8266 platform breaks with latest ESP8266 platform core:
https://travis-ci.org/github/thelsing/knx/jobs/724208015#L984

I do not know how to fix this as I am not familiar with the ESP8266 and dev ecosystem.
@thelsing Could you please have a look?

Using a fixed old version does not seem to be an option as other libs require a recent ESP8266 platform core.

str0bs, 2020-09-06:
@nanosonde ran into the same problem with sonoff demo.   
Created an new project with just src/main.cpp and let platformio do all the lib retrieval.
Here's my platform.ini  
```
[env:d1_mini_pro]
platform = espressif8266
board = d1_mini_pro
framework = arduino
build_flags = -Wno-unknown-pragmas
lib_deps = 
	WifiManager@0.15.0
	knx
```

nanosonde, 2020-11-18:
@thelsing 
I think it is still failing right?
What do we have to do to get the ESP8266 fixed again?

ajlajlajl, 2020-11-18:
seems "WiFiManager" library is outdated or something.
I compiled it successfully after commenting out WifiManager.
Just use arduino core's own functions to connect to wifi, i guss.
I can change and compile it but i cant test it on a device.

thelsing, 2020-11-18:
No it's a conflict between enum values. DataSecurity::none. I renamed to None and it compiles again. We should probably use namespaces. After some more copy&paste errors the build will work again.

closed 2020-11-18 by thelsing