thelsing/knx #89: Error Messages while programming Application with ETS

Space-Farmer, 2020-09-11:
Hello,

my Setup:
IDE: Platform IO Core 5.0.1 Home 3.3.0
MCU: SAMD21 developement Board (Arduino Zero clone without debug circuit, wemo samd21)
Interface: MicroBCU (OnSemi 5120)
SW: ETS 5.7.2
KNX interface: knxd on BBB (ingBudde shield)

Additional Equipment:
Several Isolated Serial to USB-Sticks
Oscilloscope
J-link debugger

Version from GIT: TAG: 1.2.0 Application knx-Demo

Issue: Programming physical Address working normal, but during programming Application severel errors are shown in serial console.

`got unexpected L_DATA_CON<\r><\n>
got U_STATE_IND: 0x17<\r><\n>
got U_STATE_IND: 0x17<\r><\n>
got U_STATE_IND: 0x17<\r><\n>
got U_STATE_IND: 0x17<\r><\n>
got U_STATE_IND: 0x17<\r><\n>
L_DATA_CON not received within expected time<\r><\n>
...`
My current status of investigation:
HW: 
  Pinning fits.
  Signals RX/TX checked. Levels Ok, rise time Very fast, timing perfect to 19,2kHz.
  Voltage supply. No Votage drops on SAMD21 or NCN5120 side.
SW:
  Tried several "Board" Options (adafruit_feather_m0 [here my debug serial is not working],sparkfun_samd21_dev_usb,zeroUSB) 
  Update to the latest "MASTER" leads to not getting PA programmed, on ESP32 Board-> cyclic resets. -> Back to 1.2.0
  removed comments inside ReadUART
  Logged this data
  Additional I logged this data with newline every 10ms to get an idea what is sent within which time. Please find attached these  files

Do you have an idea where to look? 

Please help me.

thelsing, 2020-09-14:
I'm quite clueless on the electronic part myself. You can try a simple sketch like https://knx-user-forum.de/forum/öffentlicher-bereich/knx-eib-forum/diy-do-it-yourself/33016-arduino-am-knx?p=1141860#post1141860 to see if your wiring is correct.

Space-Farmer, 2020-09-23:
Hi, wireing and electronics seems correct, also tested with konnekting.

Here are the results of my bus traffic:

PA: 1.2.15 GA: 1/7/15 Data length (bytes): 4 Data (HEX): 00 00 00 00
BC 12 0F 0F 20 D5 00 80 00 00 00 00 24
PA: 1.2.15 GA: 1/7/32 Data length (bytes): 4 Data (HEX): 00 00 00 00
BC 12 0F 0F 21 D5 00 80 3F 3C ED 91 5A
PA: 1.2.15 GA: 1/7/33 Data length (bytes): 4 Data (HEX): 3F 3C ED 91
BC 12 0F 0F 22 D5 00 80 00 00 00 00 26
PA: 1.2.15 GA: 1/7/34 Data length (bytes): 4 Data (HEX): 00 00 00 00
BC 12 0F 0F 23 D5 00 80 3E 4A C0 83 10
PA: 1.2.15 GA: 1/7/35 Data length (bytes): 4 Data (HEX): 3E 4A C0 83
BC 12 0F 0F 24 D5 00 80 00 00 00 00 20
PA: 1.2.15 GA: 1/7/36 Data length (bytes): 4 Data (HEX): 00 00 00 00
BC 12 0F 0F 21 D5 00 80 3F 3C ED 91 5A
PA: 1.2.15 GA: 1/7/33 Data length (bytes): 4 Data (HEX): 3F 3C ED 91
BC 12 0F 0F 2E D5 00 80 3C E5 60 42 D1
PA: 1.2.15 GA: 1/7/46 Data length (bytes): 4 Data (HEX): 3C E5 60 42
BC 12 0F 0F 2F D5 00 80 00 00 00 00 2B
PA: 1.2.15 GA: 1/7/47 Data length (bytes): 4 Data (HEX): 00 00 00 00
BC 12 0F 0F 2E D5 00 80 3C E5 60 42 D1
PA: 1.2.15 GA: 1/7/46 Data length (bytes): 4 Data (HEX): 3C E5 60 42

Look fine for me.

But if I start to programm with ETS, I get some stuff like this:

PA: 1.2.238 GA: 2/2/148 Data length (bytes): -1 Data (HEX):
B0 12 EE 12 94 51 43 00 27
PA: 1.2.238 GA: 2/2/148 Data length (bytes): 1 Data (HEX): 00
B0 12 EE 12 94 51 43 00 27
PA: 1.2.238 GA: 2/2/148 Data length (bytes): 1 Data (HEX): 00

Is the Data lenght of -1 correct? ( I dont know the KNX spec that good)

Additionaly I added some more comments inside the tpuart_data_link_layer.cpp

{
                //should not happen
                print("expected L_DATA_CON not received data 0x");
				println(rxByte, HEX);
                print("current counter 0x");
                println(_RxByteCnt, HEX);
                print("Buffer 6  0x");
                println(buffer[6], HEX);
                dataConBytesReceived(_receiveBuffer, _RxByteCnt + 2, false);
                _waitConfirm = false;
                delete[] _sendBuffer;
                _sendBuffer = 0;
                _sendBufferLength = 0;
                _loopState = IDLE;
            }

This results in error messages:

expected L_DATA_CON not received data 0xB0
current counter 0xF
Buffer 6  0x6
got UNEXPECTED: 0x12
got UNEXPECTED: 0xEA
got UNEXPECTED: 0x12
got unexpected L_DATA_CON
expected L_DATA_CON not received data 0x90
current counter 0x9
Buffer 6  0x0
got UNEXPECTED: 0x12
Timeout during RX_L_DATA
got unexpected L_DATA_CON
L_DATA_CON not received within expected time

Only Frames with Size ( I think that is whats behind buffer[6] ) 0 or 6 are effected. 

Do you have an idea why the  L_DATA_CON (0x12) is expected in this cases 1 Byte earlier?

nanosonde, 2020-11-17:
@Space-Farmer 
Please try again with the recent fix.

thelsing, 2020-12-05:
No response after two weeks. Closing as it was most likely resolved. Feel free to open if the problem persists.

closed 2020-12-05 by thelsing