thelsing/knx #93: ETS Failed to Programm Partly

hhansen06, 2020-10-03:
I have a working knx Stack. Using a STM21. 
Everything is working fine, except:

- When i try to do a partly write, ETS Cancel the Transaction with the Error: 
"Es wurde versucht einen geschützten oder nicht vorhandenen Speicherbereich zu lesen."

Any hints what is wrong here?

Sonnengruesser, 2020-10-08:
I think partly write is not supported yet by the knx stack. Just do a full application upload.

thelsing, 2020-11-24:
The flash memory usage is managed dynamically. So if the flash starts at address 1000 and you have an address table of 100 Byte, the association table starts at address 1100. I you now increase the address table by adding some more group addresses in ETS you need more space. So you have to rewrite everything, or move the address table to an address with more free space. This should work already, but you get flash memory fragmentation.

One would have to define maximum sizes for the address table, application program, etc to support partial programming.

Sonnengruesser, 2020-11-25:
In the meanwhile is there an option to disable partial programming in the knxprod xml file?

mumpf, 2021-02-14:
Hi @thelsing,

there are these entries in knxprod:

            <AddressTable MaxEntries="65535" />
            <AssociationTable MaxEntries="65535" />

Why not use these for the sizes of the according tables? Of course this would mean, they have to be set correctly in knxprod.
A further option for compability: As long as they are 65535, the flash memory is managed dynamically without partly programming (as it is now), but as soon as they are set to any other value, the required memory is used.

Regards, Waldemar

thelsing, 2021-02-18:
You would have to keep those values in sync (knxpod <-> code).  But otherwise its doable. One could use dynamic memory management if no sizes are set, and fix the sizes of the tables if the limits are set.

OutOfSync1, 2021-04-12:
I created a pull request (#132 ) to enable partial programming. This seems to work at least on ESP32 and SAMD. Thanks to mumpf and proggerKA at KNX-UF for the idea and testing.

closed 2021-04-16 by thelsing