thelsing/knx #96: Device not responding on last commit

ajlajlajl, 2020-10-26:
Hi
I've got a project that was on this commit 
https://github.com/thelsing/knx/commit/52d3866e41e1e9fd460f366bd9016daf82ec5e12

I pulled yesterday and updated my knx stack. since then my device is not responding to requests!
or if i have to be more specific, ETS cant find module (PhysAddrRead receives no answer).
But in debug serial I get this:
```
dataBroadcastIndication
dataBroadcastRequest
dataBroadcastConfirm
```
so it seems device is receiving and answering the request (? not sure).
btw. Im usinf STM32 and ncn5130.

and, if i rollback to the commit Ive mentioned above, everyting works fine. Ive tested this with a new fresh project and one of samples and it had the same problem too.

thelsing, 2020-11-11:
KNX-IP works now. The default routing count was wrong. I also disabled Data-Secure but this is probably unrelated.

Can you test with current head please?

ajlajlajl, 2020-11-12:
yup, its working now.
Thanks.

closed 2020-11-12 by ajlajlajl