thelsing/knx #97: How to turn off the function can reduce the volume

aniceking, 2020-10-31:
hello, I used the example to compile, but the generated bin file is too large to download on stm32f103c8t6. How to turn off some unnecessary functions and make the generated file smaller? Here are the compilation parameters and results:

build_flags =
  -DNCN5120
  -DMASK_VERSION=0x07B0
  -Wno-unknown-pragmas
RAM:   [=         ]   5.8% (used 3812 bytes from 65536 bytes)
Flash: [==        ]  18.0% (used 94364 bytes from 524288 bytes)

ajlajlajl, 2020-10-31:
I dont you can reduce size of this library. in my project knx lib takes ~49KB alone. (im at 129K right now). go for stm32f103CB 😑

thelsing, 2020-10-31:
You can disable some knx-data-secure in https://github.com/thelsing/knx/blob/master/src/knx/config.h.

An other way would be to disable some datapoint type in https://github.com/thelsing/knx/blob/master/src/knx/dptconvert.cpp

closed 2020-12-05 by thelsing