thelsing/knx #99: problem with DPT 3.7

ajlajlajl, 2020-11-02:
Hi
I've got a problem with using dimming control (3.7) datapoint. its a 4bit value.
But in function 'valueToBusValueStepControl' here:
https://github.com/thelsing/knx/blob/ffa2eea65bf43be546871a5b03bd9922c327d65c/src/knx/dptconvert.cpp#L973-L992
when index=0 
https://github.com/thelsing/knx/blob/ffa2eea65bf43be546871a5b03bd9922c327d65c/src/knx/dptconvert.cpp#L978
it sets only one bit. so no matter what i send, it actually only sends 0 or 8. (it should be 0-15).
its a bug right?

btw 1: Same goes for 'busValueToStepControl' & 'busValueToBinaryControl' & 'valueToBusValueBinaryControl'
btw 2: whats index 1? i couldn't find anything in documents(chapter 3.7.2, section 3.3). Should I look somewhere else?

ajlajlajl, 2020-11-04:
same goes for DPT 18.1 and valueToBusValueSceneControl().

thelsing, 2020-11-09:
The logic of the index is as follows:
DPT 3.7 consists of two parts:
- c 1 bit Increase/Decrease (index 0)
- bbb 3 bit Stepcode (000b - 111b) (index 1)

The whole DPT-Encoding stuff was taken from a knxd branch. It would probably b easier to understand if every complex dpt would get a struct.
In case of DPT 3.7 a signed char would do the job just fine.

Feel free this ;)

ajlajlajl, 2020-11-10:
oh, so I've got to create 2 group object to handle a, for example "scene control" DTP? something like this?
```
	GroupObject& sceneLearn = knx.getGroupObject(2);
	sceneLearn.dataPointType(Dpt(18, 1, 0));
	GroupObject& sceneNumber = knx.getGroupObject(2);
	sceneLearn.dataPointType(Dpt(18, 1, 1));
	sceneLearn.value(1);
	sceneNumber.value(5);
```
right?

and, if i want to manage it myself i can just use another DTP? like 5.10 ?
```
	GroupObject& scene = knx.getGroupObject(2);
	scene.dataPointType(Dpt(5,10));
	scene.value(0x85);
```
would it be alright? (I donno much about KNX protocol so im not sure if its okey to use another DPT there)

thelsing, 2020-11-10:
No the group object is the same.
Something like:
```
GroupObject& scene = knx.getGroupObject(2);
learn = scene.value(Dpt(18,1,0));
number = scene.value(Dpt(18,1,1));
```
When you write a group object you have to set the first components with `GroupObject::valueNoSend` and only the last component with `GroupObject::value`.

If you want to decode the data yourself,, you can get a raw pointer with `GroupObject::valueRef` and the size of the data with `GroupObject::valueSize`.

The better way would be custom structs for complex datapoints.

closed 2020-11-11 by ajlajlajl