thelsing/knx #106: addNewUsedBlock: found block can't contain new block

aniceking, 2020-11-20:
The main board I used is cbt6 128K, which was successfully compiled and uploaded with the example. However, there are some problems when using ETS configuration parameters. What is the problem that the program can not run normally? Is it not configured well?

platformio.ini
[env:genericSTM32F103CB]
platform = ststm32
board = genericSTM32F103CB
framework = arduino

build_flags = 
  -DMASK_VERSION=0x07B0
  ; -DNO_KNX_CONFIG
  -DNCN5120
  -DHAVE_HWSERIAL3
  -DKNX_FLASH_SIZE=128
  -Wno-unknown-pragmas

manufacturerId, version and hardwareType matches
saverestores 3
-10
.
-26
.
-68
.
restored saveRestores
tableObjs 4
-68
.
addNewUsedBlock: found block can't contain new block

thelsing, 2020-12-05:
I just tested it on esp8266. Maybe this is an endianess issue? can you check with the printHex function if the Memory changed?

thelsing, 2021-01-08:
Closing because no response after a month. Feel free to reopen if the problem persists.

closed 2021-01-08 by thelsing