thelsing/knx #109: std::function support for SAMD

rueckix, 2020-11-29:
I found that `ARDUINO_ARCH_SAMD` does not imply `HAS_FUNCTIONAL` in `group_object.h` even though SAMD supports this c++11 template. I believe support for SAMD should be added here unless there is a specific reason not to. 

_Why would one need this?_
This is quite useful when using callbacks on group objects. When modelling a device as a class, the callback is typically a non-static member function that cannot be passed as a regular function pointer. 
An elegant solution is to combined std::bind with std::function to register a member function of a specific class object.

_Implications_
The change is simple but including STL headers conflicts with the definition of `max` and `min` macros in the arduino framework for SAMD. This should be fixed in https://github.com/arduino/ArduinoCore-samd/pull/399  but not released. 
Per interim, a hacky 
```
#undef max
#undef min
```
is required before including STL headers.

Opinions?

thelsing, 2020-12-05:
you could add these undefs to bits.h as (with #ifdef arduiono_samd) . We can remove them if the arduino core gets fixed. Feel free to submit a pull request.

rueckix, 2020-12-06:
PR open

closed 2020-12-06 by rueckix