thelsing/knx #110: addNewUsedBlock: found block can't contain new block on SAMD21

schroepf, 2020-11-29:
First of all: Thanks a lot for all the work and effort you spent developing and maintaining this library! Really Awesome!

I came across some issue when I tried to use that lib on a SAMD21 board.

- knx library version: current master branch (commit: f3751b6ccd178a0a6a72bfb4e37e171506b11027)
- Hardware: same device as documented [here](https://github.com/schroepf/SAMD21-M0-Mini)
- platform.io `build_flags`:
  ```
  -DMASK_VERSION=0x07B0
  -DKNX_FLASH_SIZE=1024
  ```
  (leaving out `KNX_FLASH_SIZE=1024` uses the default value of 8192 which causes a fatal error in `SamdPlatform::getEepromBuffer()` to be thrown)

But using the `KNX_FLASH_SIZE` of 1024 I'm able to program the device through ETS just fine. But after restarting the device after programming I get:

```
addNewUsedBlock: found block can't contain new block
```

Using the **exact same setups** but version 1.2 of the knx library (i.e. the version which has been published to platform.io) instead of current master **it works just fine!!** (i.e. I can program with ETS successfully, and the values sent from the `knx-demo` sample project are successfully sent to KNX and can be observed on the ETS busmonitor.

# Full log

The full log of the ETS programming session up to the moment when the issue occurs (right after ETS has - successfully - finished programming)
```
readMemory
RESTORED FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 00 01 00 40 00 00 00 00 10 00 00 00 00 04 00 00 01 00 00 00 97 01 00 01 04 00 00 C8 1C 00 20 C4 0F 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 1F 00 20 00 00 00 00 14 00 00 00 E4 8A 01 00 00 27 48 5B 08 04 00 20 FC FF FF FF 50 05 00 00 FC FF FF FF 5C 8E 01 00 C8 0F 00 20 24 10 00 20 24 10 00 20 28 11 00 20 20 03 00 20 00 00 00 00 00 00 00 00 30 10 00 20 50 10 00 20 68 10 00 20 80 10 00 20 03 04 00 20 71 0B 00 20 00 00 00 00 00 00 00 00 15 06 00 20 7C 7F 01 00 20 17 00 20 0E 00 00 00 18 7A 01 00 30 18 00 20 06 00 00 20 E4 0F 00 20 00 00 00 00 C8 0F 00 20 00 01 EC E9 00 00 00 00 58 8B 01 00 E0 18 00 20 04 00 01 00 E4 0F 00 20 00 00 00 00 00 00 00 00 C8 8D 01 00 A0 19 00 20 05 00 00 20 E4 0F 00 20 00 00 00 00 00 00 00 00 EC 7F 01 00 50 1A 00 20 04 00 00 20 E4 0F 00 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 C8 88 01 00 34 11 00 20 0C 6C 29 8D 00 00 10 00 68 10 00 20 E0 0F 00 20 FF FF FF FF 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 32 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 28 11 00 20 24 10 00 20 50 10 00 20 00 00 00 00 E8 89 01 00 78 1C 00 20 10 00 00 00 0C 00 00 00 00 00 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
saved memory doesn't match manufacturerId, version or hardwaretype
manufacturerId: FFFF FA
version: FFFF 3
hardwareType: FF FF FF FF FF FF 
 00 00 00 00 00 00 
ownaddr 0
dataBroadcastIndication
dataBroadcastIndication
progmode on
progmode off
dataBroadcastIndication
progmode on
dataBroadcastIndication
dataBroadcastRequest
dataBroadcastConfirm
dataBroadcastIndication
dataBroadcastIndication
dataBroadcastRequest
dataBroadcastConfirm
dataConnectedIndication
dataConnectedRequest
dataConnectedIndication
dataConnectedRequest
dataConnectedIndication
Basic restart requested
save saveRestores 3
-10
.
-26
.
-68
.
save tableobjs 4
-68
.
-75
.
-82
.
-89
.
SAVED 00 FA 00 00 00 00 00 00 00 03 11 2A 00 01 00 00 01 60 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00 01 02 03 04 05 06 07 08 09 0A 0B 0C 0D 0E 0F 00 01 00 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  [... cut ...] 20 24 10 00 20 50 10 00 20 00 00 00 00 E8 89 01 00 78 1C 00 20 10 00 00 00 0C 00 00 00 00 00 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
manufacturerId, version and hardwareType matches
saverestores 3
-10
.
-26
.
-68
.
restored saveRestores
tableObjs 4
-68
.
addNewUsedBlock: found block can't contain new block
```

rueckix, 2020-11-29:
Using the same board since a few days. I am configuring the EEPROM emulation size instead. 

> -DEEPROM_EMULATION_SIZE=8192

No need to size it down in my case. No need to fiddle with KNX_FLASH_SIZE for this board.

Can you try that?

schroepf, 2020-11-30:
Thanks @rueckix for the hint, yes, it seems like passing `-DEEPROM_EMULATION_SIZE=8192` solves the original error. 

However with this fix when I try to run the knx-demo project it seems like the [knx.configured()](https://github.com/thelsing/knx/blob/f3751b6ccd178a0a6a72bfb4e37e171506b11027/examples/knx-demo/knx-demo.ino#L109) check in the `loop()` function still returns false (even after successfully programming the controller from ETS)...

Do you have an idea what else could be wrong?
Glad to hear that you're using the same hardware as I do - do you have it up and running already (with the master branch)?

rueckix, 2020-11-30:
See https://github.com/thelsing/knx/pull/108, you probably do not have KNX Secure (encryption on the wire) support in your KNX installation. Until this is addressed, you need to remove the "USE_DATASECURE" flag in config.h.

schroepf, 2020-11-30:
Thanks a lot! This fixed the issue! Awesome... 👍

closed 2020-11-30 by schroepf