thelsing/knx #111: ESP8266 compiling issues

ouranos6, 2020-12-02:
Hello,

I try to compile the demo or the sonoff on my esp8622 but no luck, I have always this error

```
         #error "Mask version not supported on ARDUINO_ARCH_ESP8266"
          ^
exit status 1
Error compiling for board LOLIN(WEMOS) D1 R2 & mini.
```

can you help me?

ouranos6, 2020-12-02:
I remove all the libraries and the arduino IDE then 
1. reinstall latest Arduino 
2. add `https://arduino.esp8266.com/stable/package_esp8266com_index.json`
3. add zip library `knx-master`
4. add tzapu esp8266 WiFiManager via the library manager
5. open knx-sonoffS20 example
6. upload
7. Enjoy!!

ouranos6, 2020-12-02:
ok I have another kind of issue, the serial monitor gone crazy and shows all the time 

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
ISR not in IRAM!

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

Of course I search for this ISR not in IRAM but don't find anythink that could help me.

thelsing, 2020-12-02:
use the knx-demo example first. I didn't look at the sonoff example for ages. You need to configure the correct mask-version in in config.h (if you use the arduino IDE)  or with a -D compiler switch. Disable the USE_DATA_SECURE there too, because the example knxj-proj don't have it enabled.

The ISR is probably related to button_up in knx_facade you could uncomment the attach_isr and activate the program mode automatically if !knx.configured

ouranos6, 2020-12-02:
😳 whaou give me some time to figure out what you meant.  

If I only use my weemos D1 mini it has to be KNX-ip right?

Now I have to find how to configure that. 

I try in a couple of hours

ouranos6, 2020-12-02:
OK!!!

So now it's better! but I can send the prog via ETS. I've imported the DB and put in prog mode as you can see but in ETS there is no devices in programming mode

```
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: Home
*WM: [3] Using Password: !secret
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.1.164
readMemory
RESTORED FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
saved memory doesn't match manufacturerId, version or hardwaretype
manufacturerId: FFFF FA
version: FFFF 3
hardwareType: FF FF FF FF FF FF 
 00 00 00 00 00 00 
setup multicast addr: 224.0.23.12 port: 3671 ip: 192.168.1.164
result 1
*WM: [3] unloading 
progmode on
progmode off
progmode on
progmode off
progmode on
progmode off
progmode on
progmode off
progmode on
progmode off
progmode on
progmode off
progmode on

```

ouranos6, 2020-12-02:
but something appears when I "test" my IP router

```
Unhandled service identifier: 20B
Unhandled service identifier: 20B
Unhandled service identifier: 20B
Unhandled service identifier: 20B
```

ouranos6, 2020-12-04:
Hello it's me again,

I've done almost everything you mentiond like :
- use the knx-demo example
- configure the correct mask-version in in config.
`#define MASK_VERSION 0x57B0`
- Disable the USE_DATA_SECURE there too
`//#define USE_DATASECURE`

This part it's not very clear for me:

> The ISR is probably related to button_up in knx_facade you could uncomment the attach_isr and activate the program mode automatically if !knx.configured

I have uncommented theses lines and it work better or at least it's not rebooting all the time.
If I touch the PIN0 I have the message on serial monitor `progmode on` and `progmode off`

```
    // pin or GPIO the programming led is connected to. Default is LED_BUILTIN
     knx.ledPin(LED_BUILTIN);
    // is the led active on HIGH or low? Default is LOW
     knx.ledPinActiveOn(LOW);
    // pin or GPIO programming button is connected to. Default is 0
     knx.buttonPin(0);
```
but as I mentionned before I can't program it via ETS. I've unplug and replug the power after upload the sketch but no luck.

I'm impressed with all the work you have done on this repo and I'am KNX tutor and arduino enthousiast and I want to build some stuffs for my home. And I was very excited when I read the README.md

And I still think that I can make it but with a little help :)

thelsing, 2020-12-05:
I just tried it with a fresh checkout:
- edit config.h to enable:
#define MASK_VERSION 0x57B0 
and disable:
//#define USE_DATASECURE

- upload sketch, check if progmode is activated with button (or pin)
- remove power from board to hard reset it, replug it
- set Wifi via Wifimanager
- knxprod is already imported to ETS
- program PA+Apllication via ETS
- Done.

Can you export the log of ETS groupmonitor during programming? You can also try to only program the PA first.

And check if you can reach the MCU via IP.
You can enable the lines https://github.com/thelsing/knx/blob/f3751b6ccd178a0a6a72bfb4e37e171506b11027/src/esp_platform.cpp#L63 and https://github.com/thelsing/knx/blob/f3751b6ccd178a0a6a72bfb4e37e171506b11027/src/esp_platform.cpp#L84 to check if the MCU gets any data at all.

ouranos6, 2020-12-07:
Hello!

This is the Arduino IDE upload output
```
Executable segment sizes:
IROM   : 359164          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28072   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1328  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 4360  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25608 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 392924 bytes (37%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31296 bytes (38%) of dynamic memory, leaving 50624 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM4
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 5c:cf:7f:0b:66:f8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 397072 bytes to 276617...
Wrote 397072 bytes (276617 compressed) at 0x00000000 in 6.3 seconds (effective 504.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
I can ping the MCU
```
Envoi d’une requête 'Ping'  192.168.10.115 avec 32 octets de données :
Réponse de 192.168.10.115 : octets=32 temps=2 ms TTL=255
Réponse de 192.168.10.115 : octets=32 temps=2 ms TTL=255
Réponse de 192.168.10.115 : octets=32 temps=6 ms TTL=255
Réponse de 192.168.10.115 : octets=32 temps=3 ms TTL=255
```

But I have no other kind of output. Impossible to see it on the ETS and I change the device for another esp8266 same wemos D1.
I've tried changing the wifimanager version but no luck

[image]

ouranos6, 2020-12-07:
After some try to put on prog mode etc.. I finally have some good news.

But after an succesfull download from the ETS I have no values on the ETS group monitor and even if I make a read request I have no ansewer

```
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: SSID
*WM: [3] Using Password: !Secret
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 192.168.10.115
readMemory
RESTORED FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF 
saved memory doesn't match manufacturerId, version or hardwaretype
manufacturerId: FFFF FA
version: FFFF 3
hardwareType: FF FF FF FF FF FF 
 00 00 00 00 00 00 
setup multicast addr: 224.0.23.12 port: 3671 ip: 192.168.10.115
result 1
*WM: [3] unloading 
progmode on
progmode off
progmode on
Unhandled service identifier: 20B
Basic restart requested
save saveRestores 3
-10
.
-26
.
-82
.
save tableobjs 4
-82
.
-89
.
-96
.
-103
.
SAVED 00 FA 00 00 00 00 00 00 00 03 20 01 00 01 00 00 01 60 00 01 00 00 00 00 00 00 00 01 00 00 00 01 04 00 00 00 00 00 00 00 00 00 01 E0 00 17 0C 00 01 10 00 1E 54 65 6D 70 65 72 61 74 75 72 73 65 6E 73 6F 72 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 FF FF FF FF FF FF FF FF FF FF 
restart
⸮
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
*WM: [1] AutoConnect 
*WM: [2] Connecting as wifi client... 
*WM: [3] STA static IP:
*WM: [2] setSTAConfig static ip not set, skipping 
*WM: [1] Connecting to SAVED AP: SSID
*WM: [3] Using Password: !Secret
*WM: [3] WiFi station enable 
*WM: [3] enableSTA PERSISTENT ON 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
[... 219 lines cut ...]
1421
1185
1652
Unhandled service identifier: 20B
1598
1825
1459
1652
1384
1261
1276
1317
1327
1790
1349
1559
1167
1928
1762
1537
1400
1061
1732
1552
```

thelsing, 2020-12-09:
This seems like a network issue. You can check if the esp sends something with the printHex as I wrote above. Or you can check with wireshark. Or maybe the IP-router blocks the group adresses?

betamoojw, 2020-12-23:
> After some try to put on prog mode etc.. I finally have some good news.
>
> But after an succesfull download from the ETS I have no values on the ETS group monitor and even if I make a read request I [...]

Is there any updates? Several months back I tried to program the NodeMCU ESP8266 board with board LED (as program status indicator) and Flash button (as program button) via ETS however it stuck in downloading Physical Address as below 
[image] and still failed after some attempts. I was wondering if you could share your experience on how to make a successful download.

ouranos6, 2020-12-24:
Hello,

Merry Christmas to all of you. 

No, I spent so many hours trying to find the right way but I didn’t find something viable. 

For now I paused because my free time is limited. but if it’s possible to have some tips and tricks or some help I restart to try because I love the idea to make my own sensor. 

Cheers and have a safe holiday. 

Ivan

thelsing, 2020-12-25:
As I wrote before: You need to hard reset the MCU after flashing via reset-button or power disconnect. Otherwise it will hang while rebooting. 

For group objects not showing in group monitor: Enable the lines as I wrote above and check if the MCU sends something. If not check the flags on the Com-Objects. If so, check your network traffic with wireshark.

ouranos6, 2020-12-26:
Hello,

I’ve tried with and without hard reset. 

I’ve add all the modifications and I have no information on the serial monitor. 

I’ve also put all the setup on an simple 4g wifi router to avoid external issues and a Weinzierl 751 IP router. 

But no luck

Boolkinator, 2021-04-25:
I have always loaded only when using a multicast connection.  Also, to continue downloading, after the first reboot in ETS, you must first specify the gateway (not multicast) in the line properties.  When changing the parameters of the device in the ETS, you also need to download via multicast.  I will try to add screenshots later.  

It would be very cool to make it possible to download over TCP.

closed 2021-09-08 by thelsing