thelsing/knx #114: max delay() between knx.loop()

zelogik, 2020-12-23:
Hi, 

It's not really an issue, but do you have already tested the maximum interval between each knx.loop() before got problems? Normally i don't like using delay but it's easier to block the loop for around 35ms for my use case and get precise timing.( 'KNXify' a Steam generator, removing controller by a samd21)

@thelsing i have a working pzem004t-v30 setup sending data over my KNX bus. Do you want pull request as demo exemple?

Regards

thelsing, 2020-12-25:
No idea. It depends on the size of the buffer for udp-packets/serial-port and the traffic on the bus.

thelsing, 2020-12-25:
More examples, especially ones that can be directly used are always welcome.

closed 2021-01-11 by zelogik