thelsing/knx #115: datetime example.

zelogik, 2020-12-23:
Hello,

I want to use the group adress datetime from my KNX bus. And before reinventing the wheel, i'm looking for a working class/fonction, as example.

Without simple example seem like i will need to translate the xknx python class of datetime...and learn with bitwise stuff.

Thanks

thelsing, 2020-12-25:
The whole code for datapoint conversion is in https://github.com/thelsing/knx/blob/master/src/knx/dptconvert.cpp.

DPT 13.* and 19.* are supported too.  Just set the group object to the DPT you want and use the value method to read/write struct tm from it.

zelogik, 2021-01-11:
Hi,

Maybe simple for you.... but I don't get how to make busValueToDateTime works ...

What to do with that knx.groupObjects(X).value(), when I read the busValueToDateTime  and KNX_Decode_Value, I don't know where functions put that struct ... (reference?pointer?) 

Can you point me how do you make a simple Serial.println() ?

Thanks

zelogik, 2021-01-13:
struct tm myTime;
  myTime = goDateTime.value();
  unsigned short tmp_sec = myTime.tm_sec;

So simple ... sorry

closed 2021-01-13 by zelogik