thelsing/knx #117: Requesting a GO from the bus

nrobadey, 2021-01-05:
Hi,
I try to request a GO value from the bus.

Here is the simplified arduino loop:

```
#include <Time.h>
void loop() {
  knx.loop();
  [some code]
  if (!knx.configured())
    return;
  if ((year()<2000) goDateTime.requestObjectRead();
}
```

Monitoring the bus, I see the request, and then the answer from the KNX clock device (actually an Ethernet bridge).
But my device doesn't register this answer, and keeps asking since its internal clock hasn't been updated.

The KNX clock device sends the time GO on the bus every 5 minutes, and these telegrams get registered by my device. The callback updates the internal clock, and my device stops asking for the GO.

Any idea why my device doesn't react to the GO it requests, but works fine with the GO updates sent automatically on the bus?
Am I using the .requestObjectRead() wrongly?

The only difference I see is that the Telegrams sent by the clock in response to my queries have a "GroupValueResponse" type, and the ones received from an automatic update have the "GroupValueWrite" type.

thelsing, 2021-01-06:
This sounds like a flags problem:
https://support.knx.org/hc/en-us/articles/115003188089-Flags

nrobadey, 2021-01-06:
You are right, the "U" flag was missing in my ETS config.
Thanks for the advice!

closed 2021-01-06 by nrobadey