thelsing/knx #130: Zephyr Integration

scholzi100, 2021-04-09:
Zephyr is a RTOS, targeting some what IoT systems, it supports a lot of boards and is included in the Linux foundation.

Website: https://www.zephyrproject.org/

Code Examples:
https://docs.zephyrproject.org/latest/samples/index.html#samples-and-demos

Are there any plans to integrate with Zephyr or any other RTOS for better hardware support?

nikwest, 2021-04-09:
I started a port. Initial port is not difficult, but in order to take advantage of zephyr features you'd need to make significant changes. Also having to use C++ makes the binary pretty large. I had to disable some features to fit it into 128kB for a stm32 g0. Didn't proceed further as I also wanted to wait for the 256kB version of the g0.

nanosonde, 2021-04-10:
Concerning the large binary:
Have you disabled RTTI, exception handling and the stdlib?

C++ in this project is not using the STL and does not make use of RTTI and exceptions.
Dynamic memory allocations are used with new, but in a very "controlled" manner.

I can recommend reading this: https://arobenko.github.io/bare_metal_cpp/#_know_your_compiler_output

Basically it boils down to the following compiler flags for C++:
`-fno-exceptions -fno-unwind-tables -fno-rtti`

In the CC1310 port I am additionally using; `-std=c++11 -fno-threadsafe-statics `.

nikwest, 2021-04-10:
Ah, didn’t know this. C++ is the one language I always tried to avoid :)
Will check. Thank you that’s very helpful.

thelsing, 2021-04-11:
If somebody wants to further reduze the footprint of the lib with further tweaks from the article. Feel free to submit a pull request.

Regarding the topic: I have no plans to port the lib to other OSs. But pull requests are welcome.

scholzi100, 2021-04-12:
> I started a port. Initial port is not difficult, but in order to take advantage of zephyr features you'd need to make significant changes. Also having to use C++ makes the binary pretty large. I had [...]

Well, lead times are endless long because of the "chip shortage". I am also waiting on a ST G0B1RE board for testing with NCN5130. Thankfully [async uart for stm32](https://github.com/zephyrproject-rtos/zephyr/pull/30917) got merged into zephyr back in January. One less thing to worried about.

thelsing, 2023-12-06:
Closing this as it's out of scope the the project. This lib can be used by any RTOS. You just have to call the knx.loop() task often enough.  Feel free to submit a PR if there's a need for finer control of the background tasks.

m-byte, 2026-10-23:
Potentially relevant as well: https://github.com/zephyrproject-rtos/zephyr/pull/119619/