thelsing/knx #144: Unload application is not permanent

croghostrider, 2021-06-14:
Hi @thelsing  

your stack is amazing.

I found a bug if I unload the application over the ETS.
The address table, association table, and the application will be unloaded only until the restart of the board.

After the restart, it shows everything loaded and the device is sending telegrams.

I am not sure how to fix this.

thelsing, 2021-06-14:
I would add a method the the Memory class that writes some bytes to _data and calls _platform.commitToEeprom(); in such a way that readMemory failse [here. ](https://github.com/thelsing/knx/blob/6254fc9b6789adf13058c3f243f3becde8a69b3c/src/knx/memory.cpp#L39) Just writing 0x00 should be enough for that.

SirSydom, 2022-03-04:
unload should only unload the application but not the PA, right?

croghostrider, 2022-03-04:
it should unload everything.
See here the support article:
https://support.knx.org/hc/en-us/articles/4402998506386-Unload-Device

SirSydom, 2022-03-04:
the article differs between unloading application and unloading application and address,
so both should be possible.

thelsing, 2022-03-04:
Yes. See https://github.com/thelsing/knx/blob/22af8267c3d059fb9e63a5d8ab351e74605b2540/src/knx/bau_systemB.cpp#L150 and 3.7.1.2 in 3/5/2 of the spec.

Phil1pp, 2025-05-07:
It seems that unloading is only permanent if you do a writeMemory() before the next restart.

After loading an application, ETS will initiate a restart of the device which will call
BauSystemB::restartRequestIndication, where a writeMemory() will happen.

However for unloading an application the behaviour is different.
ETS will not initiate a restart of the device and therefore writeMemory() won't be executed.
The Unload - all procedure in Kaenx-Creator is missing the <LdCtrlRestart />
Is this by design?
Should the writeMemory() and restart be handled by the application?

thelsing, 2025-05-09:
It think calling writeMemory() on unload would be ok. It would be best if we only call it only once event though unload adress table, association table etc.  

Flash memory could probably tolerate an additional 4 write on every reconfiguration from ETS, but it would be much better if we only call writeMemory() wenn all table objects are unloaded or something similar.

Phil1pp, 2025-05-09:
I can confirm that calling `writeMemory()` in line 257 does indeed work.
By adding additional debug output I can also confirm that it is called 4 times.

However due to my limited knowledge of the stack, I don't know how to implement that it would only be called once.

thewhobox, 2025-05-09:
You simply cant do that.
This function gets called everytime the ets unloads a table.
There are 4 tables, so it gets called 4 times.

But it will also be called 4 times, when the ets transfers the parameters, because the ets first unloads every table before loading it again.

thewhobox, 2025-05-09:
How about something like a timeout?
If there was no telegram with LoadControl or MemoryWrite in the last few seconds call writeMemory?

For transfer parameters would have no effect, because the LoadControlRestart comes first.
And when unloading it will write it a few seconds after only once.

Phil1pp, 2025-05-12:
Yes, I also thought about a timer based solution.
Will you implement this in the code?
I don't know how to do this properly so the timeout is evaluated in the main loop.

Btw. what's the best way from within my arduino-application code to check if a application is currently loaded in the knx stack or not?

thewhobox, 2025-05-12:
I will have a look at it.

You can use knx.configured().

thewhobox, 2025-05-12:
:D yes you are completely right!
I fixed it.