thelsing/knx #146: STM32G0 sample code

ZeR0build, 2021-07-14:
Greetings,

I've issue with using the stack with STM32G0 family, more specific G071RB. May I know if someone here who has successfully made this working on G0?

Actual issue is after application download via ETS. Application download doesn't give any error, but when testing and checking bus monitor the device doesn't sends any telegrams. I think is related to EEPROM as the memory register is different to generic series.

Anyone here who please can guide me?

nikwest, 2021-07-14:
I also had similar issues when I tried it a couple of months ago with an stm32g0. I just couldn't get it to work. I didn't think it might be due to the stm32g0 though. 
I will try again when I find some time.

thelsing, 2021-07-15:
Maybe its just an issue to set the KNX_FLASH_SIZE define to a valid size.

ZeR0build, 2021-07-15:
> Maybe its just an issue to set the KNX_FLASH_SIZE define to a valid size.

I tried -DKNX_FLASH_SIZE=1024 and -DEEPROM_EMULATION_SIZE=8192 also, but no luck.

thelsing, 2024-09-13:
You can try the devel branch and enable the various loggers.