thelsing/knx #147: Is it possible to configure KNX device programmatically?

wladwnt, 2021-09-04:
The lib supports configuring via ETS, but what if I want to configure the device programmatically? I mean how to set device address and group addresses etc. Is it possible? Maybe some (at least basic) examples or hints where to look. Thank you.

thelsing, 2021-09-04:
Its currently not possible. You need to do the same stuff that ets does:
1. assign indivisual address via DeviceObject (possible)
2. allocate memory for address table, association table and group object table (not possible because method is not public) 
3. set the data of above tables correctly (possible but a bit hard)
4. save memory
5. reinit stack.

I think if you need this it's probably better to use one of the not ets knx libs. During ETS programming you device will restart at least once. On linux this restarts the application itself.

What do you want to do?

wladwnt, 2021-09-04:
Thank you for the hints!
1. and 2. ok, basically possible (I think I will manage to change some methods to public)
3. have to get better understanding of those tables ...
4. , 5. ok
5. 
I want to implement KNX IP (and later maybe KNX TP) as an additional interface (additionally to what I have already: MQTT, REST-API, Modbus TCP) to my special Gateway (based on ESP32).
The Gateway is developed to have access to many hundreds possible parameters on a proprietary CAN-Bus (used by Hoval AG in their heat pump/climate/ventilation etc. devices ) via some open interface (currently MQTT, REST-API, Modbus-TCP). The user has to configure what parameters he wants to see (maximum 40 simultaneously because of RAM/performance limitations). He does it via WEB UI of the Gateway. For KNX configuration that would be also most convenient for the user to configure via WEB UI rather than generate .knxprod etc. and use ETS (I am not sure, but I think I cannot generate “general” .knxprod for Gateway since I cannot know what 40 of many hundreds possible parameters the user will select to forward via KNX IP (and I cannot distribute it anyway since it is probably not completely legal, the signing issue). If some advanced users want to do, it also would be possible for them with your lib. Otherwise I could use https://github.com/envy/esp-knx-ip, that is easy to understand and configure via WEB UI, but it looks like it is stopped in development and definitely does not support KNX TP.

thelsing, 2021-09-05:
I think that esp-knx-ip is more what you need. My lib only makes sense if you want to configure with ETS. 

IMO thats the core of knx: configure everything in a central place. If you have to configure group adresses and parameters of a gateway it's not really a knx integration but more a knx-addon for the gateway. 

I wouldn't care too much about the the knx-prod issue. You could just distribute the xml file and let users sign it.

wladwnt, 2021-09-06:
"IMO thats the core of knx: configure everything in a central place."
- Yes, for sure, and probably people who deep in KNX (not like me) think the same.

"You could just distribute the xml file and let users sign it."
- This is idea ... and one more reason to use your Lib :)  However, I believe I have to read significant part of KNX Standart to understand how to configure things correctly (and how to apply your Lib correctly). That is why for the beginning I probably will start with esp-knx-ip (even if I also have to change it in some parts, but is simple...). 

Anyway, thank you for the discussion (and the lib, of course).

closed 2021-09-08 by thelsing

PaoloTK, 2024-03-22:
Has anything changed on the topic in recent time? This library seems to be the most up to date one and others are not as feature complete. While I see the value in allowing users to configure the device with ETS, if one wants to implement KNX  in other software allowing the user to configure KNX through the software GUI is more convenient.

thelsing, 2024-03-25:
No nothing has change and no such change is planned.