thelsing/knx #148: Demo Application on ItsyBitsy M0

Rolf-M, 2021-09-19:
Hello all,
fist of all, thanks for this great project, looks like a lot of work, and an impressive platform, if I ever find out how to use it :-)

Well, I used "SirSidom"s Micro BCU, with the "ItsyBitsy" Board and Connector, from the Konnecting Project.
This gave me the chance, to not having to deal with RX-TX and Power issues.
I've used the Demo Provided with the project, and added a config in PlatformIO for the itsybitsy.
[env:itsybitsy_m0_tp]
board = adafruit_itsybitsy_m0
platform = atmelsam
framework = arduino
lib_extra_dirs = ../../../
lib_deps = 
	SPI
	https://github.com/thelsing/FlashStorage.git
	knx
build_flags = 
	-DMASK_VERSION=0x07B0
	-Wno-unknown-pragmas

I've set the Pin for Progmode to pin 2 and the LED to high active.
After I've uploaded the Program, I can see the following Output, when I try to Program the Board from ETS.

got U_RESET_IND
progmode on
expected L_DATA_CON not received
got UNEXPECTED: 0xF1
got UNEXPECTED: 0x32
got UNEXPECTED: 0xF1
got U_STATE_IND: 0xFF
got UNEXPECTED: 0x60
got UNEXPECTED: 0xC2
got L_ACKN_IND
got U_STATE_IND: 0x17
got U_STATE_IND: 0x17
got unexpected L_DATA_CON
L_DATA_CON not received within expected time

After A while I can see an error in ETS, that the device didn't respond in time.

What did I miss, is there anybody able to help me.
Thanks!

Rolf-M, 2021-09-23:
I can't find any problem, and I'm quite sure now, that the application is running well on the samd.
The only thing, I didn't try so far, is a different KNX-Interface for the ETS. Maybe that there is a timing problem through my knx-router. I'll grab a usb-interface and check if this makes any difference.

thelsing, 2021-09-23:
If you have a test environment with no additional devices you could try if that works. You can also add printHex to the *uart* methods of the platform and compare between samd and your board. 

But first make sure that GND is the same for the tpuart and the MCU.

wladwnt, 2021-10-19:
I did some investigations. The problem happens apparently because when some data are sent, then L_Data.con is not received immediately but after some other telegrams that are recieved inbetween. Therefore, the typical log have the following sequence:
...
**expected L_DATA_CON not received**
LOOP STATE: 0
LOOP STATE: 1
p>FF 
got U_STATE_IND: 0xFF
got UNEXPECTED: 0x60 <-- LOTS OF THAT
.....
p>8A 
LOOP STATE: 0
LOOP STATE: 1
p>0B 
**got unexpected L_DATA_CON**

The quick and very verdy dirty solution was to emulate immediate L_DATA_CON changing state loop in tpuart_data_link_layer.cpp:

```c++
...
//complete Frame received, payloadLength+1 for TCPI +1 for CRC
                if (rxByte == (uint8_t)(~_xorSum))
                {
                    //check if crc is correct
                    if (_isEcho && _sendBuffer != NULL)
                    {
                        //check if it is realy an echo, rx_crc = tx_crc
                        if (rxByte == _sendBuffer[_sendBufferLength - 1])
                            _isEcho = true;
                        else
                            _isEcho = false;
                    }
                    if (_isEcho)
                    {
                        _loopState = RX_WAIT_DATA_CON;
                        ////////////////// ADD THIS BLOCK
                        println("L_DATA_CON received");
                        _receiveBuffer[0]=0x29;
                        _receiveBuffer[1]=0x00;
                        _receiveBuffer[2]=0x0B;
                        dataConBytesReceived(_receiveBuffer, 3, true);
                        _waitConfirm = false;
                        delete[] _sendBuffer;
                        _sendBuffer = 0;
                        _sendBufferLength = 0;
                        _loopState = IDLE;
                        //////////////
                    }
                    else
                    {
                        frameBytesReceived(_receiveBuffer, _RxByteCnt + 2);
                        _loopState = IDLE;
                    }
                }
...
if (_waitConfirm)
    {
        if (millis() - _waitConfirmStartTime > CONFIRM_TIMEOUT)
        {
            ///// COMMENT OUT THIS
            /*println("L_DATA_CON not received within expected time");
            uint8_t cemiBuffer[MAX_KNX_TELEGRAM_SIZE];
            cemiBuffer[0] = 0x29;
            cemiBuffer[1] = 0;
            memcpy((cemiBuffer + 2), _sendBuffer, _sendBufferLength);
            dataConBytesReceived(cemiBuffer, _sendBufferLength + 2, false);
            */
            _waitConfirm = false;
            delete[] _sendBuffer;
            _sendBuffer = 0;
            _sendBufferLength = 0;
            ///// AND THIS
            /*if (_loopState == RX_WAIT_DATA_CON)
                _loopState = IDLE;
            */
        }
    }
```

After this change I was able to program the ItsyBitsy with address, get device Info etc.

I try to use KNX TP only since last week, so I dont know yet what exactely the problem is:
 - Is that within of KNX/TPUART specification that L_Data.con does not receieved immediately, but after some telegrams inbetween?
 - If yes, the lib must be changed to handle that
 - If not, what is the reason? Arduino layer inbetween, knxd (delays because of IP to TP translation), NCN5120 or something else?
 - if knxd, why my Hager KNX TP device can be programmed via knxd? Is the knx implementation in Hager device more robust?

ChriD, 2022-01-13:
@wladwnt & @Rolf-M 
 My Problem was solved 16 days ago with #161 
I think this may solve this Problem here too?

Rolf-M, 2022-01-14:
@ChriD I'll be checking this on the weekend, thanks for the update and keeping this in mind!
Will be giving some feedback, as soon as I know better...

SirSydom, 2022-02-11:
I had exactly the same problem and it was finally solved with #161

closed 2022-02-12 by thelsing