thelsing/knx #150: Difficulty compiling the .ino file with #include <knx.h>

richardpub4136, 2021-09-29:
I have been working on this superb project for several days.
To simplify the understanding I used from the example files of the bme680.
  When I try to compile the bme680.ino file on the Arduino IDE but I get stuck at #include <knx.h>
I don't understand why I'm blocking at this level.
I couldn't find the corresponding library.

I also tried with VSCode but also failed.

My final goal is to flash a Sonoff R2 basic module so that I can then program it from ETS5.

Thanks for your help figuring out why I can't even compile the .ino file

Regards

richardpub4136, 2021-10-11:
Thank you this is the info I was missing, I had not found this info on the Net.
So I will be able to move forward now
Does it work with ESP8266? I suspect that it is.
Thank you very much for putting me on this path.
Regards

richardpub4136, 2021-10-11:
Arduino : 1.8.10 (Windows 10), Carte : "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), dtr (aka nodemcu), 26 MHz, 40MHz, DOUT (compatible), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (190703), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0X2341_0X0042 -ide-version=10810 -build-path C:\Users\JEANLO~1\AppData\Local\Temp\arduino_build_377669 -warnings=none -build-cache C:\Users\JEANLO~1\AppData\Local\Temp\arduino_cache_637387 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.python3.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose E:\Documents\Arduino\libraries\knx-master\examples\knx-demo\knx-demo.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\Documents\Arduino\libraries -fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=legacy,ssl=all,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=1M64,led=2,sdk=nonosdk_190703,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=0X2341_0X0042 -ide-version=10810 -build-path C:\Users\JEANLO~1\AppData\Local\Temp\arduino_build_377669 -warnings=none -build-cache C:\Users\JEANLO~1\AppData\Local\Temp\arduino_cache_637387 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.python3.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Jean Louis Richard\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose E:\Documents\Arduino\libraries\knx-master\examples\knx-demo\knx-demo.ino
Using board 'generic' from platform in folder: C:\Users\Jean
Using core 'esp8266' from platform in folder: C:\Users\Jean
Detecting libraries used...
"C:\\Users\\Jean Louis Richard\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Jean Louis Richard\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\Jean Louis Richard\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\Jean Louis Richard\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\JEANLO~1\\AppData\\Local\\Te [... cut ...]
Utilisation de la bibliothèque ESP8266WebServer version 1.0 dans le dossier: E:\Documents\Arduino\libraries\ESP8266WebServer 
Utilisation de la bibliothèque DNSServer version 1.1.1 dans le dossier: E:\Documents\Arduino\libraries\DNSServer 
Utilisation de la bibliothèque SPI version 1.0 dans le dossier: E:\Documents\Arduino\libraries\SPI 
Utilisation de la bibliothèque EEPROM version 1.0 dans le dossier: E:\Documents\Arduino\libraries\EEPROM 
exit status 1
Erreur de compilation pour la carte Generic ESP8266 Module

thelsing, 2021-10-11:
The error already says what to do: open src/knx/config.h and comment in what you need. In your case it should probably be https://github.com/thelsing/knx/blob/57b1950e38de1e26e17aec6e815a6b1b6043214a/src/knx/config.h#L21

richardpub4136, 2021-10-12:
Yes Yes Yes!!!! Great I was finally able to compile, thank you very much, I could not have guessed all this without your help ....
Before that I didn't think there was any of this to do ....
I still have a lot of knowledge to acquire.
Again a thousand thanks