thelsing/knx #152: SAMD21 lets to be programmed from ETS only once?

EugenElephant, 2021-11-02:
Dear thelsing, first of all, huge respect and a big thanx for your work.

I use the library with Siemens TPUART. The microcontroller is Seeeduino XIAO (SAMD21). 
Everything compiles fine. The knx-demo-tp works also just fine.
However there is an issue with programming the demo via ETS5:
1. when I configure it via ETS5, the very first time after flashing the board, the physical address and the application programs are transferred to the board just fine.
2. when I repeat the programming, the physical address is transferred, however the application program not. From the bus monitor it looks like the board just doesn't answer on certain requests.

In attachment you will find the recordings from the bus monitor for the 1st case when everything works just fine and for the second case when the programming of the application fails.

I haven't changed anything in the demo.
Any idea what can be wrong and how to fix it?

thank you.

best wishes
  Eugen
[bus_monitor_case1_everything_is_fine.txt](https://github.com/thelsing/knx/files/7460386/bus_monitor_case1_everything_is_fine.txt)
[serial_output_case1_everything_is_fine.log](https://github.com/thelsing/knx/files/7460388/serial_output_case1_everything_is_fine.log)
[image]
[bus_monitor_case2_application_programming_fails.txt](https://github.com/thelsing/knx/files/7460387/bus_monitor_case2_application_programming_fails.txt)
[serial_output_case2_application_programming_fails.log](https://github.com/thelsing/knx/files/7460400/serial_output_case2_application_programming_fails.log)

Domos-Snips, 2021-11-04:
Hello all and thanks Thelsing for your great work.
I am using the same componants (Xiao Seeeduino + Siemens) and I have the same behavior.
I can only do a "complete programmation" on the participant, no  partial programmation.
I test that with my own code and knxprog, but also with the demo code.
Regards

thelsing, 2021-11-04:
I think partial programming is a bit experimental. See #93. But you should be able to program only the application without the PA. 
Its a bit hard to give advice. I would add printHex Calls to the read/writeUart functions in the platform to get a better understanding where the difference between success and failure is.

etrinh, 2021-11-05:
Have you tried to regenerate the knxprod file with the latest  CreateKnxProd?

EugenElephant, 2021-11-05:
i have problems with both the knxprod from the samples as well as with the newly generated knxprod from the xml file (the XML file had to be adjusted first)

EugenElephant, 2021-11-15:
i have found the reason of the problem. Well the true reason is still unclear to me, but 
the problem was caused by platformio in VS Code.  I have defined the section:
[env:seeed_xiao]
platform = atmelsam
board = seeed_xiao
framework = arduino

which somehow resulted in including wrong libraries, so ARDUINO_ARCH_STM32 instead of ARDUINO_ARCH_SAMD was #defined

After switching the environments a couple of times and recompiling the project, the right library was included and ARDUINO_ARCH_SAMD was #defined

In order to make sure, that the right library is included, just have a look into the file knx_facade.cpp, the block after the code line:
#ifdef ARDUINO_ARCH_SAMD
should be shown enabled, the rest of the blocks should be disabled:

[image]

Domos-Snips, 2021-12-13:
Hello,
After enforcing the ARDUINO_ARCH_SAMD, I was not able to program the device. Did you face the issue ?
I can't see it as device in ETS as a programmable device. 
Before diving in a debug session any clue on what I did wrong ?
Regards

EugenElephant, 2021-12-14:
Hi 
It looks like you have 2 questions :

>  enforcing the ARDUINO_ARCH_SAMD, I was not able to program the device.
How exactly have you enforced ARDUINO_ARCH_SAMD

> I can't see it as device in ETS as a programmable device.
For this you need to create a knxprod file and load it into ETS

Best wishes
  Eugen

Domos-Snips, 2021-12-14:
Thanks Eugen for your answer. I found the issue I had. I not correctly set the MASK_VERSION, it is why i removed my question.
To enforce the ARDUINO_ARCH_SAMD, I just put -DARDUINO_ARCH_SAMD=1 as extra parameters in platformio.ini.
Now everything is working well.
Regaards

EugenElephant, 2021-12-14:
You do not need to do this
-DARDUINO_ARCH_SAMD=1

If you are using seeeduino Xiao, then just put:
platform = atmelsam
board = seeed_xiao
framework = arduino

Best wishes
  Eugen

thelsing, 2021-12-16:
This issue can be closed, can't it?

EugenElephant, 2021-12-16:
Yes

Best wishes
  Eugen

closed 2021-12-16 by thelsing