thelsing/knx #154: How to force a memory save on power interrupt?

Sonnengruesser, 2021-12-10:
Hi,
I'm using an NCN5120 based TPUART with the "save" pin wired to the SAMD21. Now I've defined an interrupt that triggers in case the KNX power is lost and I would like to force the SAMD to call the "save" callback (save_restore.h).
Saving/loading user data works using the callbacks, but I didn't find a function to manually trigger the save callback. Is this implemented somewhere in the framework?
Thanks!

thelsing, 2021-12-13:
You just need to add a method like save() to the knx-facade to call _bau.writeMemory(). Then you can just use knx.save() in the interrupt handler.

thelsing, 2021-12-13:
I don't think that the interupt registration should be added to the stack, as it depends on how the board is wired. But it could be added to an example like knx-demo.

Sonnengruesser, 2021-12-19:
Thank you for the hint, works perfectly as you suggested.

closed 2021-12-19 by Sonnengruesser