thelsing/knx #155: issues with Node MCU ESP8266 device and knx-demo

richardpub4136, 2021-12-16:
I am stuck doing the download in ETS5, from my Node MCU ESP8266 device, I have tried many methods to make "knx-demo.ino" work with this device.
Obviously, it is at the level of "if (knx.configured ()) '=" that it gets stuck. I debugged at this level, but the "if (! Knx.configured ()) '=" branch is activated.
Could you tell me in which part of the sources is this programmed, or what is the most obvious solution ???

thelsing, 2021-12-16:
The problem is usually the network. Enable the lines https://github.com/thelsing/knx/blob/87edd3dfe01398f56af3bf97633dc9fbd90f6d76/src/esp_platform.cpp#L71
and 
https://github.com/thelsing/knx/blob/87edd3dfe01398f56af3bf97633dc9fbd90f6d76/src/esp_platform.cpp#L91
to see if your nodemcu gets packets from ETS.

etrinh, 2021-12-16:
I have also issue to communicate on IP with esp8266 and esp32, both devices have same behaviour:
- It receives requests from ETS
- It answers requests, but no UDP packets are sent to the bus (not really none, sometimes some appears but not enough to finish a programming, just up to individual address programming...)
I suspect issue with esp frameworks... I have not seen issues with the stack so far, the failure is in a system call and all calls seem correct...
Strange that on both esp32 and esp8266 I have exactly the same behaviour...

thelsing, 2021-12-16:
I think that is a bit unlikely. Maybe it's just normal packet loss on wifi. UDP has no retransmissions. I think the standard has  a tcp mode in an application note, but I don't have access to them.

richardpub4136, 2021-12-16:
I am totally lost.
I activate the lines

knx / src / esp_platform.cpp

Line 71 in 87edd3d
// printHex ("<-", buffer, len);

and
knx / src / esp_platform.cpp

Line 91 in 87edd3d
  // printHex ("->", buffer, len);

but i have nothing different

etrinh, 2021-12-16:
In fact on ESP32, all work fine with just the exception of a waiting time of 5mins after each reset before it starts to send frames... After these 5 mins, it can be programmed (first individual address only, then application program) and it communicates normally...
On ESP8266, there is also this delay, but programming doesn't seem to finalize, I will try to figure it out when I will be able to progress with the delay...
@richardpub4136, if you start discovering device in programming mode in ETS, you should see succession of lines like these:
-> 06 10 05 30 00 11 29 00 B0 D0 11 00 00 00 01 01 00 
<- 06 10 05 30 00 11 29 00 B0 E0 13 02 00 00 01 01 40
First one means that the stack receives the request, and the second that it answers to the request.
But nothing is sent on the network, if you monitor with wireshark, during the first 5mins after reset.

thelsing, 2021-12-17:
Are you aware that you wifi monitoring (especially under windows) is not as easy as starting wireshark and be done? You need to set the wifi card into monitor mode too. Not all wifi cards support this. See https://blog.packet-foo.com/2019/04/wireless-capture-on-windows/comment-page-1/

etrinh, 2021-12-18:
I figure out my delay issue, it was due to 'IGMP Proxy' which was disabled on my router.
Now frames are working well.
@richardpub4136  can you double check you don't have this option disabled as well at your side?
You should see the device address in diagnostic tool.

I am now trying to investigate why knx-demo is still crashing during address programming with my ESP8266...

etrinh, 2021-12-18:
After cleaning up all my stuffs for delay investigation, it happens that the demo is now working correctly at my side :)
@richardpub4136 maybe you have the same issue as I had with my router.

richardpub4136, 2021-12-18:
Indeed the ESP8266 sends commands in IGMPV2.
I have to find out how to activate it in my freebox ...
I watch this step

richardpub4136, 2021-12-19:
Call for help with IGMPv2 concerns.
After several months of research and testing, I think the concerns I am having are related to the configuration of my Ethernet network.

I started with Sonoff Basic R2, flashed with Tasmota KNX. It worked fine with mqtt, but with ETS5 I could control my Sonoffs with diagnostic mode, to actuate them, but it didn't work when I wanted to use an ignition control button physically installed on my KNX bus.
I had already suspected that the problem could come from multicasting.
 Faced with this failure, I switched to the KNX Thelsing version, and I took a long time to figure out how to create my own .knxprod files. That resolved, I started flashing esp8266 and esp32, and that's when I got into a stalemate: could not program from ETS5 (no response when fully uploading to ETS5).
etrinh, having validated the operation of knx-demo.ino on esp8266 and esp32, I came to wonder about IGMPv2 multicast.
 So I still have to figure out how to make multicast work on my network. My installation has a Pop frebox as input on my network, configured in router mode, and behind I have a Cisco 28 managed port switch in which IGMPv2 is not activated.
Unfortunately I do not have the skills to manage my Cisco switch which has all the desired parameters, but which has so many that I get confused. And with the Pop freebox serving as a router, I get lost.
According to the sites I look at, it is the router that manages the IGMPv2, and on others it is the Switch.
Cordially.

jeff25, 2022-03-05:
I try it today with latest version  of stack with ESP32. Adress i can write, if i upload appl. Program i got the messagte that "Device is not responding". I try it with a KNX Stack Version from January without problems same config. Not sure what changes from January until today, Anyone any idea?

jeff25, 2022-03-06:
Update: I try to find the branch with the defect for ESP32/8266. This Brunch is working from 13.2.222. I try to find the latest working version...

Revision: d26771c432a9cb6b60786b4da90aa4def7e325ab
Author: thelsing <xxxx>
Date: 13.02.2022 21:56:03
Message:
Merge pull request #171 from mptei/tx_throttle_rx_wait_addr

thelsing, 2022-04-25:
This should be fixed with #196 and #197

closed 2022-04-25 by thelsing