thelsing/knx #160: IP router

Avenitos, 2021-12-24:
Hello!

is it possible to make a router between TP and IP?
are there any examples?

thelsing, 2021-12-25:
See the *coupler examples.

Avenitos, 2021-12-25:
there's almost no code there :) 

It is necessary to call the callback when passing telegrams and filter them when passing to another interface

nanosonde, 2021-12-25:
> there's almost no code there :)
> 
> It is necessary to call the callback when passing telegrams and filter them when passing to another interface

It is correct that the code examples do not contain any code related to routing.

The code related to routing is included in the stack itself.
I have implemented the routing as described in the KNX spec.
However, to really use it, you would have to create your ETS product database for a coupler. It could be done, but I did not have any further time to work on this.
As soon as you would add this product database to ETS, the idea is to use this coupler like an other coupler. ETS will handle writing the filter tables, etc. This should be part of this stack already.

nanosonde, 2021-12-25:
Concerning the coupler product database, I would recommend to look at a database of an already existing coupler. Should not be to hard to create our own one and use the CreateKNXProd tool.

Avenitos, 2021-12-25:
It's good that the stack can do everything by itself! But with non-standard tasks, it can be difficult.

In my task, it is necessary to make a decision on filtering the telegram from my project, as well as to create a cache for group addresses, the application, which will be accessed from IP KNX, generates a large number of group address value reads at the time of launch, and there is no way to do without a cache.

For this purpose, it is necessary to be able to integrate into the telegram routing process.

project based on ESP 32 + TPUART2 + Ethernet

thelsing, 2021-12-25:
I think it would be best if you get the default coupler working first. After that you can add caching.  You could also add a callback that can be registered for filtering. Or add some properties that influence forwarding to the knxprod like "only forward if value changed". More examples (even more complex ones) are always welcome as pull requests.

Avenitos, 2021-12-26:
OK, let's start with a simple one. I connected tpuart, how do I legalize all the telegrams in the log? Thanks!

nanosonde, 2021-12-26:
> OK, let's start with a simple one. I connected tpuart, how do I logging all the telegrams in the log? Thanks!

For group addresses look here:
https://github.com/thelsing/knx/blob/fb74931bec618c32db48f7b2732a558e083fab16/src/knx/network_layer_coupler.cpp#L79

Avenitos, 2021-12-26:
Oh… Do I need to edit the sdk to output telegrams to the log?

nanosonde, 2022-07-05:
Just for the records. I have added some more (german) comments here: https://knx-user-forum.de/forum/projektforen/openknx/1780408-fragen-zum-routing-im-knx-stack?p=1780447#post1780447

mgoeller, 2023-02-06:
> Oh… Do I need to edit the sdk to output telegrams to the log?

@Avenitos did you proceed any further on this? If yes I would appreciate an email to "mailgoe at gmail.com" and have an offer for you.
Thank you!

thelsing, 2024-09-13:
This should now work.

closed 2024-09-13 by thelsing