thelsing/knx #169: Programming download works, but not application download

ChriD, 2022-01-28:
Hi,
I can programm my device with all settings and KO's and it will work. But when i try to download partially or the application (no programming mode) the device seems to "crash" at the very and and it doesn't work. The last stuff its sends via bus when downloading the application is this:

[image]

The last serial output is this (eherby this messafe appeaers while programming too and it doesnt matter there):
```
got U_STATE_IND: RE
L_DATA_CON not received within expected time
```

afterwards the board needs a reset to work again. What may cause this issue?

ChriD, 2022-01-30:
I did some further tests and the problem appears when i am adding a KO in the CreateKNXProd
Parameters work and if i only have parameters i can download the application. But if i have one KO defined, it wont work
(Full download always works)

ChriD, 2022-01-30:
Well, i have broke it down to the settings of the "dataPointType" for a group object. I have to investigate further

thelsing, 2022-01-31:
If you add a KO in the knxprod you have to do a full download. The number of KO and parameters determine the size of these tables. If you change those, you should changed firmware version to reinitialize the device.

ChriD, 2022-01-31:
I initialy did not change the KO count in the knxprod, i did this only to find the problem and if it was related to the KO's

What i do is follwoing
I do a full download and afterwards (after the automatic reboot of the device) without changing anything, i do the application download. (i even do not change parameter values or KO links in the ETS). The Application download runs for a while and before the last reset, it stalls and the device hangs. 

So Full download works, and the application download directly after the full download stalls at the very end i think.
It does not stall on application downoad if i do not "add any specification of the dataPointTypes" in the application itself.
I will investigate further tomorrow.

ChriD, 2022-02-01:
Okay. it seems that i had a bunch of "dataPointType"  methods  called on GO's that did not exist (wrong index). 
But that did only lead to problems when downloading the application via ETS. The "running app" had no issues with this

closed 2022-02-01 by ChriD