thelsing/knx #178: problems with APDU > 56

SirSydom, 2022-02-21:
no respond to MemoryWrite on RP2040 plattform when APDU > 56

similar issues exists with SAMD.

Must be investigated.

ExtMemoryWrite with 128Byte worked RP2040 plattform with Core arduino-pico 1.9.1

SirSydom, 2022-02-21:
a MemoryWrite with 60byte payload is received within the stack.
if verifymode == false, the response is good.
if verifymode == true, the response also has 60byte payload and is not received.

It seems the MemoryResponse is added correctly to the tx queue in the stack.
but when sending it something goes wrong.
that is beyond my knowledge.

@mptei maybe you have an idea as tpuart specialist?

[trace.zip](https://github.com/thelsing/knx/files/8112104/trace.zip)

SirSydom, 2022-02-22:
from the Trace:
> F604E7C118F1011A0A16A23FA34EA4A5BCA68A7A80A9A0AAA1ABACA2ADA3AEA4AFB0A5B1A6B2A7B3B4A8B5A9B6AAB7B8ABB9ACBAADBBBCAEBDAFBEBFB0C0B1C1B2C2C3B3C4B4C5B5C6C7B6C8B7C9B8CACBB9CCBACDBBCECFBCD0BDD1BED2D3BFD4C0D5D6C1D7C2D8C3D9DAC4DBC55DTX_WAIT_ECHO
C6C7C8C9CACBCCCDCECFD0D1D2D3D4D5D6D7D8D9DADBBDRX_WAIT_START
17got U_STATE_IND: PE

from the datasheet:
pe (protocol error): ‘1’ if an incorrect sequence of commands sent by the host controller is detected

mptei, 2022-02-22:
I guess you are using an NCN5120? TPUART and TPUART2 have only 64 byte buffers for sending frames. The 60 byte transmission exceeds that limit.

My test system currently has a TPUART2. Therefore it doesn't support longer packets. I'll switch to an NCN5120, but this might take a while.

Yes, that is an protocol error. The trace output is not sufficient to say what's wrong. The bytes send to TPUART are not fully shown (only the L2 bytes, but not the protocol stuff).

But by looking at the code I think this patch may help:
[send_more_than_64.txt](https://github.com/thelsing/knx/files/8117482/send_more_than_64.txt)

To send up to 263 bytes a 9 bit index is needed. This is splitted in 3 bits via U_DataOffset.req and the lower 6 bits are part of U_L_DataCont.req. In the code the masking of the upper bits is missing when sending U_L_DataCont.req.

Can you try that please?

SirSydom, 2022-02-22:
Damit, debugged for an hour or so this morning and found that solution also ;)
Could have used that hour for something else, but the bug is solved, so it is fine now.

Yes I'm usig NCN and your Info regarding TP-Uart tells me we need a MAX_APDU_SIZE that is configureable!

closed 2022-02-24 by thelsing