thelsing/knx #181: Porting to H8I8O board

pavkriz, 2022-02-23:
Hello,
did anyone try this library on [H8I8O board](https://www.aliexpress.com/item/1005003124183354.html)?

It's a cheap KNX device having reasonable number of inputs and outputs. 
It has GD32F103CBT6 MCU (STM32 compatible) and TP-UART 2 ver. 5.3 on board. 
So it should be feasible to run this library on this board.

thelsing, 2022-02-24:
Is there any way to modify the firmware for this board? Is so it could work.

SirSydom, 2022-02-24:
looks like it is not meant to change the fw since:

"Bieten eine e-mail adresse zu erhalten die datenbank nach der bestellung"

If there is a SWD/SWCLK interface on the board one could change the fw.
the 4 pin D C G V could be such an interface.

pavkriz, 2022-02-25:
I confirm the 4 pins D C G V are SWDIO, SWDCLK, GND and +3V. Unfortunately it looks SWD is disabled in the running firmware since my ST-LINK is unable to connect to target MCU.

I am not sure if it's feasible to perform Chip Erase without having to access the NRST pin or BOOT0 (which is inconvenient since these are probably not routed to any header).

BTW by "Bieten eine e-mail adresse zu erhalten die datenbank nach der bestellung" they mean they send you an ETS project (from which you can copy the H8I8O device to another project) and a "User manual" PDF in chinese.

SirSydom, 2022-02-25:
could you upload the project and the manual? would be interesting.

Regarding programming ESP I have no idea...

pavkriz, 2022-02-25:
I got SWD working with J-Link: https://gist.github.com/pavkriz/aca0851f38bc79789e878c7aa3777ece
ST-Link have not worked for me this way but I'll try to find out some cheap solution; most people probably do not have Segger J-Link.
The files from the seller are attached there as well.

thelsing, 2022-04-25:
@pavkriz Did you get anywhere further with this board or do you plan to work on this? Otherwise I would close this issue.

pavkriz, 2022-04-25:
I got SWD working with a cheap ST-Link clone and through 3 debug pins only. I'll update the instructions ASAP. My colleague has prepared a "testbed" and is going to try the library on the board.

m-byte, 2022-06-04:
Thanks for your instructions, @pavkriz! I can't find my JLink at the moment, so I roughly followed them with my ST-Link. Essentially, it boils down to following steps 1-3 from your instructions and then running `~/STMicroelectronics/STM32Cube/STM32CubeProgrammer/bin$ ./STM32_Programmer.sh -c port=swd -rdu` while disconnecting the needle.
Also, I replaced the needle with the tip of my tweezers since they have a more comfortable handle. This then resets the read protection and erases the chip.

Anyone about to follow these instructions: The original software will be erased and gone! You will have to program in your own and there is currently no way back.

m-byte, 2022-06-04:
Btw: the following link might be interesting for anyone only wanting to try out running this stack on the board (either H8I8O or H8C09)
https://www.youtube.com/watch?v=L1Qcnc3T99c

m-byte, 2022-06-05:
After some testing, I now have a slightly modified version running on my H8C09 (GD32). Apart from changing the serial port and pins for LED and button, I also ran into an issue with the EEPROM library. It seems like the GD32 (same behavior on two boards that I tested) needs one extra call to `HAL_Flash_Unlock` before flushing the buffer. Only the second call succeeds on my devices and this way the second call is made inside `eeprom_buffer_flush`.

@thelsing If I want to create a pull request for this, should I rather create a new platform or create something around build flags? Changes necessary for the H8C09 (should be very similar to H8I8O, but I haven't yet checked the pinout on that) are a different serial port (Serial1 isn't enabled in stm32duino by default), the change in `commitToEeprom` and setting the correct pins for button and LED.

Regarding the serial port: What is the intended way of setting that? If I understand correctly, we are supposed to define `KNX_NO_DEFAULT_UART` and then call `knx.platform().knxUart()` to pass in the correct serial port? Wouldn't a `#define` for the serial port be more straight forward? I'm mainly asking because with a `#define` it would be very easy to transparently set the correct UART inside `platformio.ini` (same would be the case for LED and button)... And that way a lot of examples could easily be run on boards like the H8I8O/H8C09 or any others with a different pinout.

m-byte, 2022-06-06:
In case anyone needs them, here are the relevant parts of the pinout for both boards (button and led are actually on different pins):
*TPUART2 connections (identical for both boards)*
| TPUART2 | GD32 |
| --- | --- |
| SAVE | PB15 | 
| RESn | PA8 | 
| RxD | PA9 (USART0_TX) |
| TxD | PA10 (USART0_RX) |

*H8I8O*
| GD32 | Connection |
|---|---|
| PA11 | KEY |
| PA12 | LED |
| PA15 | DI1 |
| PB3 | DI2 |
| PB4 | DI3 |
| PB5 | DI4 |
| PB6 | DI5 |
| PB7 | DI6 |
| PB8 | DI7 |
| PB9 | DI8 |
| PC13 | DIAUX |
| PA0 | DO1 |
| PA1 | DO2 |
| PA2 | DO3 |
| PA3 | DO4 |
| PA4 | DO5 |
| PA5 | DO6 |
| PA6 | DO7 |
| PA7 | DO8 |
| PB0 | DOAUX |

*H8C09*
| GD32 | Connection |
| --- | --- |
| PB5 | LED |
| PB0 | KEY |
| PA3 (USART1_RX) | RX |
| PA2 (USART1_TX) | TX |

Please note that I'm using the GD32 UART names. For STM32duino, you'll have to add 1 to get the correct name (i.e. TPUART2 is connected to `Serial1` in the Arduino framework.

pavkriz, 2022-06-06:
I've updated the flashing guide and moved here: https://github.com/pavkriz/hacking-knx-h8i8o
You can now use cheap ST-Link clone to flash the device and avoid any hassles with the needle ;-)

thelsing, 2022-06-07:
@m-byte I think defines for the serial port would be the way to go. I the  additional Hal_flush_unlock call works for other boards too just add it with a comment.

m-byte, 2022-06-08:
I'm unable to test it on any STM32 boards at the moment. We are currently moving in and I only have the modules and parts that I knew I would need available. What I can do is look at the code and that indicates that there should not be any issue. If the flash is already unlocked, it will just return `HAL_OK` and not do anything. There is equivalent code in this function for all STM32 variants that are known to STM32duino.

m-byte, 2022-06-08:
@pavkriz I like your approach.
Given that PlatformIO uses OpenOCD by default, it's probably better to use that as you are very likely to have it installed when developing for this stack. Therefore, I reorganized it a bit so that it is now making use of OpenOCD. It can be either run as a build target or directly. (See the newest commit)

thelsing, 2022-06-08:
Closed by #209

closed 2022-06-08 by thelsing