thelsing/knx #193: Can only read 18 Bytes of parameter data from ETS

EugenElephant, 2022-03-26:
In knxprod I have specified several parameters resulting in 24 Bytes of data. 
All parameters are displayed in ETS and let set their values. 

Reading them out in the code results in correct data for some of the first parameters (within first 18 Bytes of data) and some garbage for the remaining parameters.
After some investigation I have added:
`
bool TableObject::allocTable(uint32_t size, bool doFill, uint8_t fillByte)
{
    Serial.printf("TableObject.allocTable (%d)\n", size);

`
in the table_object.cpp which handles the memory for the Parameters
this has generated the output:
`
TableObject.allocTable (18)
TableObject.allocTable (42)
TableObject.allocTable (2)
TableObject.allocTable (2)
`

the first allocation seem to correlate to my problem - only 18 Bytes of data is allocated resulting in correct data for the first parameters in the memory. The remaining parameters are probably read outside of the allocated memory.

Please assist with a direction where to investigate further.

thanks a lot

mumpf, 2022-03-27:
I would assume, the size information in knxprod is 18 bytes. Look at all Size attributes in your xml. The should always represent the size of your data.

Regards,
Waldemar

EugenElephant, 2022-03-27:
Hi Waldemar, thank you for your response.

here how it looks like in knxprod:

```
<Code>
              <RelativeSegment Id="M-00FA_A-0001-00-0000_RS-04-00000" Name="Parameters" Offset="0" Size="24" LoadStateMachine="4" />
</Code>
```

and exactly the last two parameters each of size:
```
<TypeNumber SizeInBit="32" Type="unsignedInt" minInclusive="0" maxInclusive="10000" />
```
are read out corrupted.

mumpf, 2022-03-27:
Yes, that is the Size of your Memory Segment for your parameter data. But how big is the Size of the Table you allocate?
That's why I wrote 

>  Look at **all** Size attributes in your xml.

Regards,
Waldemar

mumpf, 2022-03-27:
If you share here your xml, I can have a look...

EugenElephant, 2022-03-27:
I would appreciate if you can have a look. Here it is. The Parameters dummy2 and dummy3 come from ETS broken.
[Dimmer TW3-24V16A.v.08.xml.zip](https://github.com/thelsing/knx/files/8358019/Dimmer.TW3-24V16A.v.08.xml.zip)

mumpf, 2022-03-27:
Hi, as mentioned above, check ALL Size-Attributes in your xml:
Line 14:

    <RelativeSegment Id="M-00FA_A-0001-00-0000_RS-04-00000" Name="Parameters" Offset="0" Size="23" LoadStateMachine="4" />

Size is only 23, not 24 Bytes.
Line 153:

    <LdCtrlRelSegment AppliesTo="full" LsmIdx="4" Size="18" Mode="0" Fill="0" />

Size is just 18 Byte. Same in Line 156:

    <LdCtrlWriteRelMem ObjIdx="4" Offset="0" Size="18" Verify="true" />

Just correct this, and everything will be ok. Or you use my tool [multiply-channels](https://github.com/mumpf/multiply-channels/releases/tag/2.1.2), which calculates and inserts all sizes automatically. In addition, it contains many further checks, which prevents many common errors you can do in a knxprod-xml.

Regards, Waldemar

EugenElephant, 2022-03-28:
Thanks Waldemar, it has now worked fine. Also your tool is a big help, I have marked your repository in my favorites :)

closed 2022-03-28 by EugenElephant