thelsing/knx #212: Device programming via ETS

KreativeLabs, 2022-06-23:
I am learning this stack and it is very promising but I've ran into a problem I cannot solve.

I am fiddling with the BME680 demo project in Arduino IDE and I managed to configure it and compile it, then also the upload works too. I am using Seeeduino XIAO + Siemens 5WG1117. TPUART connection is okay also. I defined a prog button and a LED and they work too, I can see the serial debug info (progmode on).

I tried to use the sample knxprod file, but as it was configured as IP protocol, I couldn't use it on my TP network. So I tried to open the provided xml file with the CreateKNXProd application. It complained about an xml schema (http://knx.org/xml/project/11 was not good for it, and I had to rewrite http://knx.org/xml/project/20). After all I could generat a new .knxprod file but this time ETS could import it however I couldn't add it to my building structure. Error was: "The given key was not present in the dictionary". (What does that even mean?) So I gave it up for a while and I created another .knxprod which worked fine.

My problem was when I tried to program my device through ETS it asked for prog button push. I pushed and it started programming phisical address, which worked fine, then it rebooted the device and after that it timed out without programming. It was not the BME680.knxprod file, maybe that caused the problem that the XIAO wanted a specific device programming and got another knxprod parameter list? Can you point me in a direction so I can figure this out? Also about the knxprod creation and ETS error it would be good to know why it was problematic.

Thank you for your kind help! (I will post my serial output in a new comment.)

KreativeLabs, 2022-06-23:
Here is my serial output, if that helps:

<pre>
00:34:26.353 -> start
00:34:26.353 -> readMemory
00:34:26.353 -> RESTORED D8 2D 00 20 89 02 00 00 85 02 00 00 87 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 85 02 00 00 00 00 00 00 00 00 00 00 85 02 00 00 7D 02 00 00 85 02 00 00 85 02 00 00 85 02 00 00 85 02 00 00 85 02 00 00 
00:34:26.353 -> DataObject api changed, any data stored in flash is invalid.
00:34:26.353 -> expexted DataObject api version: 1, stored api version: D82D
00:34:26.353 -> ETS has to reprogram PA and application!
00:34:26.391 -> ownaddr FFFF
00:34:26.391 -> Timestamp [ms], raw temperature [°C], temperature[°C], pressure [hPa], raw relative humidity [%], humidity [%], gas [Ohm], IAQ, IAQ accuracy, CO2, CO2 Accuracy, breathVocEquivalent, breathVocAccuracy, compGasValue, compGas Accuracy, gasPercentage, gasPercentage Accuracy, staticIAQ, staticIAQ Accuracy, runInStatus, stabStatus
00:34:49.498 -> progmode on
00:34:54.978 -> Basic restart requested
00:34:54.978 -> save saveRestores 2
00:34:54.978 -> Writing state to EEPROM
00:34:54.978 -> 3
00:34:54.978 -> 7
00:34:54.978 -> 4
00:34:54.978 -> 1
00:34:54.978 -> 3D
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 73
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 2D
00:34:54.978 -> 0
00:34:54.978 -> 1
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> FF
00:34:54.978 -> FF
00:34:54.978 -> 2
00:34:54.978 -> 2
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 2
00:34:54.978 -> C
00:34:54.978 -> 0
00:34:54.978 -> 2
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> C8
00:34:54.978 -> 41
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> C8
00:34:54.978 -> 41
00:34:54.978 -> 10
00:34:54.978 -> 0
00:34:54.978 -> 3
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 16
00:34:54.978 -> 0
00:34:54.978 -> 5
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> C
00:34:54.978 -> 0
00:34:54.978 -> 9
00:34:54.978 -> 0
00:34:54.978 -> FF
00:34:54.978 -> FF
00:34:54.978 -> FF
00:34:54.978 -> FF
00:34:54.978 -> FF
00:34:54.978 -> FF
00:34:54.978 -> FF
00:34:54.978 -> 7F
00:34:54.978 -> 8
00:34:54.978 -> 0
00:34:54.978 -> A
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> 0
00:34:54.978 -> CA
00:34:54.978 -> CC
00:34:54.978 -> 0
00:34:54.978 -> 0
00:35:21.083 -> start
00:35:21.083 -> readMemory
00:35:21.083 -> RESTORED D8 2D 00 20 89 02 00 00 85 02 00 00 87 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 85 02 00 00 00 00 00 00 00 00 00 00 85 02 00 00 7D 02 00 00 85 02 00 00 85 02 00 00 85 02 00 00 85 02 00 00 85 02 00 00 
00:35:21.083 -> DataObject api changed, any data stored in flash is invalid.
00:35:21.083 -> expexted DataObject api version: 1, stored api version: D82D
00:35:21.083 -> ETS has to reprogram PA and application!
00:35:21.083 -> ownaddr FFFF
00:35:21.083 -> Timestamp [ms], raw temperature [°C], temperature[°C], pressure [hPa], raw relative humidity [%], humidity [%], gas [Ohm], IAQ, IAQ accuracy, CO2, CO2 Accuracy, breathVocEquivalent, breathVocAccuracy, compGasValue, compGas Accuracy, gasPercentage, gasPercentage Accuracy, staticIAQ, staticIAQ Accuracy, runInStatus, stabStatus

</pre>

thelsing, 2022-08-15:
You should be able the solve this issue if you reprogram the device com completely via ETS.

KreativeLabs, 2022-08-16:
Hi! Thank you for your answer. What do you mean exactly by "device com"? I have KNX of course and I tried reprogramming the device but maybe not in the best way. Thank you!

closed 2024-07-21 by thelsing