thelsing/knx #233: Issue on decoding DPT 10.001 values - "Index"

Sonnengruesser, 2023-03-13:
I'm trying to read a DPT 10.001 time value to feed the Arduino RTC. Now I face an issue on the decoding of the bus values.
The code here: [busValueToTime](https://github.com/thelsing/knx/blob/3bf7f5ce481f143269940140cbf5448b86a176a7/src/knx/dptconvert.cpp#L411) will always use "case 0", since the index is defaulted to 0 [here](https://github.com/thelsing/knx/blob/3bf7f5ce481f143269940140cbf5448b86a176a7/src/knx/dpt.h#L366).

I didn't find a function to set the Index (also for any other DPT where more than Index 0 exist). Am I missing something?
For DPT10.001 this means bytes 1 (minutes) and 2 (day&hours) will never be read out correctly.

Is the "case 0" function required anyways? Where does the Index come from?

mumpf, 2023-03-13:
As far as I know, index is used to read/write parts of combined DPT, like DPT3 or DPT19 (and obviously also DPT10). In most cases I have also difficulties to use them, so I tend to use RAW data and decode them on my own.

mumpf, 2023-03-13:
You found the answer on your own, I will close the issue.

closed 2023-03-13 by mumpf