thelsing/knx #239: Pass through KNX-bus to Serial with H8C09 board

mrWheel, 2023-04-05:
Hi,

Can I use a H8C09 board to simply pass through KNX to an ESP32-KNX device via Serial (Tx/Rx)?

Does anyone have experience with this board and/or this setup?

My ESP32 device now works via WiFi but that is not always reliable.

msfuture, 2024-01-07:
The H8C09 has a STM32 processor on board, that is between the KNX and Serial connection. The easiest way without soldering would be to write a sketch, that echoes the KNX serial from the BCU to the Serial and the other way around. As the STM32 does not consume that much energy I think that would be okay...

closed 2024-01-07 by mrWheel

Ing-Dom, 2024-01-07:
why don't you use a NanoBCU ? It's smaller and less expensive ?

mrWheel, 2024-01-07:
@Ing-Dom .. If only there was some documentation and a reseller ..

Ing-Dom, 2024-01-07:
https://github.com/OpenKNX/OpenKNX/wiki/NanoBCU

I sell them.
https://muster.ing-dom.de/
but currently DE only, you would have to use [LogiX ](https://www.logoix.com) or a similar service.

msfuture, 2024-01-07:
Was my first thought too. Best option in combination with ESPs and RP2040s... But I'm from Germany, so no shipping issues...