thelsing/knx #247: Debug message since commit 338aa1b of 2023-05-28 maybe inconsistent ?

Maggyver, 2023-06-12:
Debug message before commit 338aa1b from 2023-05-28:

`.
restored Tableobjects
Timeout: 0
Zykl. senden: 0
Min/Max senden: 0
Aenderung senden: 0
Abgleich: 0
ownaddr 1114
Start`

Debug message before commit 338aa1b from 2023-05-28:

`.
restored Tableobjects
Timeout: 0
Zykl. senden: 0
Min/Max senden: 0
Aenderung senden: 0
Abgleich: 0
ERROR, TPUART not responding
Start`

Ing-Dom, 2023-06-12:
yes this is true, i can confirm that.
The log is misleading.

should be reworked.
any ideas @thewhobox ?

thewhobox, 2023-06-13:
I can confirm that this happens to me as well.

I will have a look at it if i have some time^^

thewhobox, 2023-06-13:
https://github.com/OpenKNX/knx/tree/fix-tpuart-error
I think i fixed it.
It now blocks the core when DataLinkLayer gets enabled and tpuart is not _enabled.
I think this is a good compromis, since this will happen only at startup with max 100ms block time.

Would be great if someone can confirm its working.

Maggyver, 2023-06-13:
I tested it. The output behaves as expected.

Can anyone else confirm this?

Ing-Dom, 2023-06-14:
works for me

Maggyver, 2023-06-14:
The basic diagnostics should include monitoring the two hardware pins, SAVEB and RESETB.

A kind of active monitoring could be realized via the state service request ... this allows you to detect the "failure" of TP-UART and that of the NCN51xx.

If desired.

thewhobox, 2023-06-15:
I think this is too far for now.
I will create a pull request which will fix this issue.
Afterwards we can implement more.

commit 494c76c7 (2023-06-16): Merge pull request #249 from OpenKNX/fix-tpuart-error

fix #247 - TPUART Error Message
closed 2023-06-16 by thelsing