thelsing/knx #276: Basic documentation

nerdoc, 2024-01-20:
ATM it is not easy to find a working KNX library in the internet. KnxTpuart is outdated and dead, KONNECTING seems to be dead, OpenKNX seems to be badly maintained and has no documentation how to start with it, and all the other libs have last commits years ago - this here seems to be the only lib to be worked on.

But: As a newcomer, it is nearly impossible to start with it, as there is no documentation at all. The readme says: "Generated documentation can be found [here](https://knx.readthedocs.io/en/latest/)." - but the docs mainly is hte readme file itself and a link to the API, which is an array of stubs.

It would be much appreciated if there were ANY documentation for beginners on how to use this software properly:

* simple, working example in "blink" style. (All the examples in the `examples` folder have no documentation either - a simple `README.md` in each directory would suffice, indicating what this example does and how to use it, and a link the main README where to start with and how.
* A simple exmplanation how to fully create a dummy device: compile and upload to an arduino, create a knxprod file, etc. - I found nothing, and don't know where to start with.
* The link to [CreateKnxProd](https://github.com/thelsing/CreateKnxProd) seems outdated, as that repo says it is deprecated. Maybe use  [Kaenx-Creator](https://github.com/OpenKNX/Kaenx-Creator) as link?

I think development of **this** library could be supported better if there was simple documentation on how to start.

Ing-Dom, 2024-01-20:
> OpenKNX seems to be badly maintained and has no documentation how to start with it,
Oh, starting with random insults maybe not the best idea if you want to get some help, isn't it?

Where exactly did you get stuck by finding documentation to @OpenKNX

btw, OpenKNX uses this stack, it is only a technical fork where you did your PR.

doctorseus, 2024-01-21:
From my own experience (as I started to get serious with this in the last few weeks and had to start from 0): Before getting started with your own development you will have to have a good understanding of KXN itself. Buy some HW and the ETS to play with. Then most of these things and this very stack will make much more sense without actually having to have the documentation you are missing.

There is no explanation about the basics of KNX in any of the projects you have mentioned because the people working and developing them have no real need to write it down. These projects are no standalone thing as so many other projects found on the internet. They integrate in an existing eco-system with its own existing documents.

That said, from what I understood: `KONNEKTING` tries to let you build KNX devices which explicitly don't use the vendor made ETS toolkit for programming. This is because of legal reasons as `Kaenx-Creator` uses a specific key to sign the `Produktdatenbanken` which contain the device description / parameters for programming. I assume most users don't think it's worth to worry about that. `OpenKNX` is the newest and tries to collect some of the existing projects under one new umbrella (e.g. it uses this stack) and also develop "standardized" new HW and SW based on RP2040. Which I think is an interesting and promising approach although I don't agree with some of the aspects which IMHO just hinder adoption (most of all the license chosen).

Ing-Dom, 2024-01-21:
> I don't agree with some of the aspects which IMHO just hinder adoption (most of all the license chosen).

what do you mean with that @doctorseus ? The GPL license choosen for the firmware code? well, that wasn't a free choice, neverless I prefer copyleft..
And in what way does that hinder you from adoption?

nerdoc, 2024-01-23:
Sorry, no insult intended. It was just my POV from looking at the OpenKNX wiki, which seems to be very unmaintained. I can see the projects using the RP2040 - but in my search I came from various variations of the [arduino-tpuart-knx-user-forum](https://github.com/thorsten-gehrig/arduino-tpuart-knx-user-forum) lib from Thorsten Gehring (where all began IMHO), to Konnekting and OpenKNX, trying to  get some arduino devices to work with KNX.
I don't care in the first place if I can program them using the ETS, ATM i have a working setup with a dummy in the ETS and Thorsten Gehring's library on an "Arduino Nano Every" which finally works great for what I need.

However, I find the idea very good to separate config and application, and program custom devices using the ETS - but I got stuck how to use that library. And yes, I am a bloody beginner with embedded development, not even a professional programmer. But enthousiastic.
So, what I am missing is a more "user friendly" way of embedding this library in a simple project for a breadboard setup with e.g. a simple Arduino - just to get started with, for dummies.
Even if this library is  used mostly by people who know what they are doing, it could bet more attention if there was more documentation. E.g. in https://github.com/OpenKNX/Examples there are projects for the Neopixel and PiPico. I don't own those - is that really necessary to use just those "recommended" ones - except the maintenance burden you would have if you add more devices to the list...

It's just my simplified view of a person that is starting to read about OpenKNX - BTW, I am using Linux, and many of the tools (Kaenx-Creator) are Windows-only. But ETS too anyway... I just hoped any project would break that barrier sometimes.)

Ing-Dom, 2024-01-23:
well, you know, we all do that for fun. And most consider writing docu not fun.. thats the point.
When you look at the OpenKNX Wiki, there are plenty of pages, but I admit, it is not that easy to start.
This lib is not ideal for beginners of embedded programming.

You may stay with the konnekting lib, which is abandoned but should work. I got it working on rp2040 but then I left konnekting.. because tuxdeo didn't really wanted help nor had subtantial progress.
https://knx-user-forum.de/forum/projektforen/konnekting/1668626-knx-mit-rp2040-pi-pico#post1735776

In OpenKNX currently one guy is working on a so called "rapid prototyping" plattform, where you don't need to hav an own knxprod but can you a generic one (like the dummy device) together with an adaptable firmware. stay tuned.

nerdoc, 2024-01-23:
I read that post ;-) (and many others...) - no, the konnekting seems not to be the right for me. I like OpenKNX here better. And again, sorry, I had no intention of insulting here.

Thanks for the info, I'll drop by from time to time and have a look. And if there are little where I could help from my perspective, I'd like to do that.

closed 2024-07-21 by thelsing