thelsing/knx #278: Missing ARDUINO_ARCH_RP2040 in knx/bits.h

mobil750, 2024-02-10:
Hi,
today I started after a longer period again on my knx project (RP2040 based) and therefore pulled the current master branch of KNX.

After that compilation failed because "ntohs(x)" was not defined ...
I checked and found in bits.h the following

> #if defined(__linux__)
#include <arpa/inet.h>
#elif defined(ARDUINO_ARCH_SAMD) || defined(ARDUINO_ARCH_STM32) || defined (DeviceFamily_CC13X0)
#define getbyte(x,n) (*(((uint8_t*)&(x))+n))
#define htons(x)  ( (getbyte(x,0)<<8) | getbyte(x,1) ) 
#define htonl(x) ( (getbyte(x,0)<<24) | (getbyte(x,1)<<16) | (getbyte(x,2)<<8) | getbyte(x,3) )
#define ntohs(x) htons(x)
#define ntohl(x) htonl(x)
#endif

As I'm using RP2040 I added " || defined (ARDUINO_ARCH_RP2040)" to the above line and compilation is OK again.

Is there an intension to avoid "ARDUINO_ARCH_RP2040" here or is it just missing?

Regards
Helmut

Ing-Dom, 2024-02-10:
the byteorder macros for the rp2040 plattform are already defined by the arduino-pico core.
Did you define LWIP_DONT_PROVIDE_BYTEORDER_FUNCTIONS ?
You must not.

mobil750, 2024-02-11:
I didn't define LWIP_DONT_PROVIDE_BYTEORDER_FUNCTIONS - as far as I'm aware of. I checked my files and didn't find the name. Where should it be defined?

Ing-Dom, 2024-02-11:
> Where should it be defined?

It should NOT be defined. The rp2040 plattforms relies on the byteorder macros defined by the pico-core, which usually defines them, at least if you did not configure it otherwise.
You must find out why in your environment pico-core does not define them.

mobil750, 2024-02-11:
I don't think you are right. The problem is that the mentioned definitions are missing ("**ntohs(x), ntohl(x)**").
There is no hint regarding LWIP_DONT_PROVIDE_BYTEORDER_FUNCTIONS in bits.h
It is only an issue within the bits.h file and adding " **|| defined (ARDUINO_ARCH_RP2040)**" solves the compile issue. The binary works well. From my point of view it is obvious that the RP2040 part is missing. Compare the two parts of the bit.h file below.

> #pragma  once

#include <cstddef>
#include <cstdint>

#if defined(__linux__)
#include <arpa/inet.h>
#elif defined(ARDUINO_ARCH_SAMD) || defined(ARDUINO_ARCH_STM32) || defined (**DeviceFamily_CC13X0**)
#define getbyte(x,n) (*(((uint8_t*)&(x))+n))
#define htons(x)  ( (getbyte(x,0)<<8) | getbyte(x,1) ) 
#define htonl(x) ( (getbyte(x,0)<<24) | (getbyte(x,1)<<16) | (getbyte(x,2)<<8) | getbyte(x,3) )
#define ntohs(x) htons(x)
#define ntohl(x) htonl(x)
#endif

#ifndef MIN
#define MIN(a, b) ((a < b) ? (a) : (b))
#endif

#ifndef MAX
#define MAX(a, b) ((a > b) ? (a) : (b))
#endif

#ifndef ABS
#define ABS(x)    ((x > 0) ? (x) : (-x))
#endif

#if defined(ARDUINO_ARCH_SAMD) || defined(**ARDUINO_ARCH_RP2040**) || defined(ARDUINO_ARCH_STM32)
#include <Arduino.h>
#elif defined(ARDUINO_ARCH_ESP8266)
#include <Arduino.h>
#include <user_interface.h>
#elif defined(ARDUINO_ARCH_ESP32)
#include <Arduino.h>
#include <esp_wifi.h>
#else // Non-Arduino platforms
#define lowByte(val) ((val)&255)
#define highByte(val) (((val) >> ((sizeof(val) - 1) << 3)) & 255)
#define bitRead(val, bitno) (((val) >> (bitno)) & 1)

Maybe the "DeviceFamily_CC13X0" is wrong and should be "ARDUINO_ARCH_RP2040" as in the second part?

Can you agree?

Ing-Dom, 2024-02-11:
> Can you agree?
no.

You may know that I am the author of the rp2040 support in this stack and I use it every day, as it is used in the complete OpenKNX Team.
So - it just works. For all but you.

> The problem is that the mentioned definitions are missing ("ntohs(x), ntohl(x)").

I understood your problem. **And I told you**, that these macros - they are called byteorder macros - should already be defined by the arduino-pico core. There should not be more then one definition, this is the reason why this plattform, which is made and intended for the use with the arduino-pico core, **does not define the byteorder macros _intentionally_.**
You must find out why in you environment these macros are not defined.

[image]

Here you can see, that this macro is defined and the sketch builds.

mobil750, 2024-02-13:
OK understood and sorry for bothering.

I found the LWIP_DONT_PROVIDE_BYTEORDER_FUNCTIONS which I wasn't aware of.

Thanks

thelsing, 2024-02-13:
Thanks @Ing-Dom for the support.

closed 2024-02-13 by thelsing