thelsing/knx #285: compiling doesn´t work

radontec, 2024-07-21:
tryed with VisualStudio, I always run into this problem:
Any idea?

In file included from src\knx\bau57B0.cpp:4:0:
src\knx\bau57B0.h: In constructor 'Bau57B0::Bau57B0(Platform&)':
src\knx\bau57B0.h:26:21: error: 'Bau57B0::_dlLayer' will be initialized after [-Werror=reorder]
     IpDataLinkLayer _dlLayer;
                     ^
src\knx\bau57B0.cpp:15:30: error:   base 'DataLinkLayerCallbacks' [-Werror=reorder]
       DataLinkLayerCallbacks()
                              ^
src\knx\bau57B0.cpp:11:1: error:   when initialized here [-Werror=reorder]
 Bau57B0::Bau57B0(Platform& platform)
 ^
cc1plus.exe: some warnings being treated as errors

[image]

thelsing, 2024-07-21:
You can try changing:
https://github.com/thelsing/knx/blob/cc54c2c98c7c5da19dad7159082abe2784db4a17/src/knx/bau57B0.cpp#L15C7-L15C31

to 
`
Bau57B0::Bau57B0(Platform& platform)
    : BauSystemBDevice(platform), DataLinkLayerCallbacks(),
      _ipParameters(_deviceObj, platform),
      _dlLayer(_deviceObj, _ipParameters, _netLayer.getInterface(), _platform, (DataLinkLayerCallbacks*) this)
`

radontec, 2024-07-22:
perfect this works! thx a lot!

closed 2024-07-22 by radontec

Ing-Dom, 2024-07-22:
as I introduced the DataLinkLayerCallbacks I feel somehow responsible - but I do not know why this failed (and why it does not fail for me and in the CI tests).
why was this change needed?

Trexis5, 2024-07-26:
Hello everyone.
I have a question.
I have the example knx-demo.ino working in Platform IO.
knx-demo-ip.knxprod works.

I have already made my own project.
Not much changed. Just adapted the channels. But everything is still on a demo basis.

Via ETS: Programming the physical address only works very rarely. Mostly never. At some point, with luck, it works. 
Then I can transfer the application program over and over again. Always works perfectly. No matter how often, it works great.

I delete the ESP. Then I can no longer get the device programmed via ETS With a lot of patience and luck, it works at some point. Can't say what the problem is.

Can anyone tell me where I could look? What information do you need? Possibly my project?

PR #286 (2024-07-22): Update bau57B0.cpp
fix #285